MarlinFirmware/Marlin #8415: Z Axis Not Homing Correctly

Deneteus, 2017-11-14:
I have 2 HICTOP Creality CR10s and with their stock firmware it homes X & Y and then Z moves up instead of down and stops after 5-10mm. So then I built and compiled a new firmware using Marlin Bugfix using the CR10 profile and it does the exact same thing. I would think if the motor wiring was reversed that it would keep moving Z up till it failed. Homing again gets the same result where Z is 5-10mm higher than where it stopped before.

Deneteus, 2017-11-14:
Not even 10 seconds after I recorded a video I figured out it was the end stop switch. The JST plug was plugged in but the connector had slipped out. As soon as I locked the pin in the JST connector and reconnected it then it went down.

thinkyhead, 2017-11-14:
Thanks for the update!

closed 2017-11-14 by thinkyhead