MarlinFirmware/Marlin #8418: M32 or M23/M24 ???

Sauerland1973, 2017-11-14:
I apologize for my bad english ...
I would like to start another program from the GCODE.
The command M32 or also M23 / M24 does not work.
Alternatively, a jump label could help me. But I did not find that.
Does anyone have an idea how I could get the problem under control?

thinkyhead, 2017-11-14:
The correct command for this is `M32` and it does have some limitations due to a shoddy implementation.

- The initial file must be in the root folder if it wants to do chaining (unless started from the LCD controller).
- Chaining requires full paths and must use the obfuscated DOS 8.3 names. To figure out which DOS 8.3 filename corresponds to the file you want, you need to use `M33 /PATH/TO/DOSNAME.GCO`.

After all that preliminary work, you end up with something like this:

#### /StartChain.gcode
```gcode
;
; G-code Chaining Test
;
; M23 /CHAIN/STARTC~1.GCO
; M24
;
; M33 CHAIN/TEST~1.GCO      ; Get long name (Chain/Test3.gcode)
; M33 CHAIN/TEST~~2.GCO     ; Get long name (Chain/Test2.gcode)
;
; M32 !CHAIN/TEST~~2.GCO# ; Test2.gcode
; M32 !CHAIN/TEST~1.GCO# ; Test3.gcode
; 

M117 G-code Chaining Test
M118 G-code Chaining Test

M300 S440 P200

M118 Chain TEST2.GCO

M32 P !CHAIN/TEST~~2.GCO#

M118 Chain TEST3.GCO

M32 P !CHAIN/TEST~1.GCO#

M118 Done!
```
#### /Chain/Test2.gcode
```gcode
;
; G-code Chaining Test (2)
;

M117 Test 2!
M118 Running Test2.gcode...

M300 S220 P200
M300 S260 P200

M118 Chain TEST3.GCO

M32 P !CHAIN/TEST~1.GCO#

M118 "Test2.gcode" done.
```
#### /Chain/Test3.gcode
```gcode
;
; G-code Chaining Test (3)
;

M117 Test 3!
M118 Running Test3.gcode...

M300 S440 P200
M300 S520 P200

M118 "Test3.gcode" done.
```

I'm looking at the code now to see if there's a better way to get the directory of the active file rather than relying on the LCD menu changing directories, which is the only way it works currently.

thinkyhead, 2017-11-14:
I'm looking at the code now to see if this can be made more versatile, so it will work in sub-folders in all cases, and so it allows use of files in any folder, not just subfolders of the main file.

thinkyhead, 2017-11-15:
I've patched `G32` so it should work more reliably. In fact, until I patched it I could invoke a sub-procedure without any problem, but when the sub-procedure file was done it wouldn't return to the previous file. So… please give `G32` some more testing and see if it works well for you.

As I was working on it, I noticed that it uses a lot of SRAM to store the path and name of the caller. If I can get it to work with the INODE instead, that would make it much less wasteful. So, that's on my list for a future patch.

closed 2017-11-16 by thinkyhead

Celino-Lin, 2020-05-12:
Sorry, I encountered a problem while debugging marlin2.0.5.3.
I want to use M32 instruction to call other subroutines, but found that the order of execution is different.
Can you tell me, am I missing any steps? What should I do?

**Steps to Reproduce**
Step1: modify macro  SD_PROCEDURE_DEPTH   1 →2
Step2: Define some marco below
`
  /**
 * cardreader.h  line 244
 */
  #ifndef SD_PROCEDURE_DEPTH
    #define SD_PROCEDURE_DEPTH 2
  #endif

/**
 * Configuration_adv.h  line 1106
 */
 #define SDCARD_SORT_ALPHA
 #define LONG_FILENAME_HOST_SUPPORT
 #define SCROLL_LONG_FILENAMES
`

Celino-Lin, 2020-06-01:
I found the reason why the printing order of SD card using M32 instruction is disordered. This is because the load instruction was read ahead of the command queue.

[image]

When the M32 instruction is executed, the queue already contains the next three queues in the parent program. Only when the execution of the parent program instruction in the queue is completed can the execution of the child program instruction begin.

问题处理方法：（在 Configuration_adv.h 中将 BUFSIZE 设置为 1）
https://github.com/MarlinFirmware/Marlin/blob/e4d8336175ee09b28d23220f446d07451ca3de43/Marlin/Configuration_adv.h#L1728
[image]

I'm not sure if this modification is appropriate? No queuing will not affect the speed of the machine?

@thinkyhead  Please help me, is there a better solution, thank you!

thinkyhead, 2020-07-08:
@Celino-Lin — 

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).