MarlinFirmware/Marlin #8429: Z_Safe_Homing do not home in the middle of the bed

Gorgoton, 2017-11-15:
Hi everyone,

I know that there is another subject (issue #5873) who is kind of the same one but when I red the comments, those could not help me.

I am new in the world of marlin and 3D printing and I have a lot of problem to make my printer works. I have a tronxy P802ma with an arduino mega 2560 with a ramps 1.4.

------------------------------------------------------------
### Bug Report
- Description: Z_Safe_Homing not working well
- Expected behaviour: Z_Safe_Homing home in the middle of the bed
- Actual behaviour: Z_Safe_Homing home in the back right corner

When I print a calibration cube with cura (I made the configuration of the bed dimension in cura), even if the cube is place in the middle of the plate, the cube print in the back right corner. I think it is because of the Z_Safe_Homing problem.

Here is my actual `Configuration.h` and `Configuration_adv.h`
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/1473375/Archive.zip)

------------------------------------------------------------

Roxy-3D, 2017-11-15:
Ah.....   What Slic3r are you using?   If it is Slice3r or Cura, you can check a box and tell it to render the object in the center of the bed.   How about you post your Gcode file and we look to see where it is telling things to print?

Gorgoton, 2017-11-15:
I don't know what is Slic3r.

**And here is my Starting Gcode:** 

G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z15.0 F9000 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F9000
;Put printing message on LCD screen
M117 Printing...

**And My ending Gcode:**

M104 S0 ;extruder heater off
M140 S0 ;heated bed heater off (if you have it)
G91 ;relative positioning
G1 E-1 F300  ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
G28 X0 Y0 ;move X/Y to min endstops, so the head is out of the way
M84 ;steppers off
G90 ;absolute positioning

Roxy-3D, 2017-11-15:
Starting Gcode in what tool?   It will be some flavor of a Slicer...

Gorgoton, 2017-11-15:
I don't know if it will respond to your question.

When I start my print, I upload a .stl file on cura (witch is a slicer?) and I start the print with my usb port connection to the arduino.

thinkyhead, 2017-11-15:
Whenever there are homing or leveling issues, we ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark.

- Download Marlin `bugfix-1.1.x` to test with the latest "nightly" code.
- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Copy the text from the Host Console and paste it into a new message or a text file that can be dropped directly onto your reply.

From this we can see exactly what the machine is doing.

I look forward to your report.

AnHardt, 2017-11-15:
>  G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops

That bypsses save homing. First home x and y. Then home z where you are.

'G28 X Y Z' should perform a save homing.

The idea of **moving** to somewhere with a G28 is wrong! It's ought to home one or more axes. The point where it stops depends on too many options to trust in a special point. When you want to move to a special point use G0 Xx Yy Zz.

Roxy-3D, 2017-11-15:
Yes.  Cura is the answer to the question.

thinkyhead, 2017-11-15:
> The idea of moving to somewhere with a G28 is wrong!

Truly. When I first started using Marlin this had me confused. I thought that surely, if you did `G28 X10 Y10` it would first home XY and then set my position to `X10`, `Y10`. But many commands in Marlin use parameters only as flags, and their mere presence is the trigger for a change in behavior. Such is the case with `G28 X Y` or `G28 Z`. If you want to invoke Z Safe Homing, you must use `G28` or `G28 X Y Z`.

closed 2017-11-15 by thinkyhead

Gorgoton, 2017-11-16:
Perfect, I will look at that and I will give you news tonight. Thank you

Gorgoton, 2017-11-20:
Sorry for the delay, I was busy last week.

I wasn't sure about what AHardt said. But I change the G28 X0 Y0;  G28 Z0 to G28 X Y Z but nothing change )as far as I can see). So I decided to follow the step of thinkyhead. I use Ponterface and this is the response for the M111 S247 code:

>>> M111 S247
SENDING:M111 S247
echo:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
echo:M105
>>> G28
SENDING:G28
echo:G28
>>> gcode_G28
Machine Type: Cartesian
Probe: NONE
Mesh Bed Leveling
  current_position=(110.00, 110.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
Raise Z (before homing) to 5.00
>>> do_blocking_move_to(110.00, 110.00, 5.00)
<<< do_blocking_move_to
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
>>> do_blocking_move_to(-397.50, -337.50, 5.00)
<<< do_blocking_move_to
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, -397.50, 0.00)
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, 5.00, 0.00)
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, -10.00, 25.00)
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
<<< do_homing_move(X)
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = -45.00
 soft_endstop_max = 220.00
> home_offset[X] = 0.00
  current_position=(-45.00, 0.00, 5.00) :
<<< set_axis_is_at_home(X)
  current_position=(-45.00, 0.00, 5.00) : sync_plan_position
  current_position=(-45.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
  current_position=(-45.00, 0.00, 5.00) : > homeX
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move(Y, -337.50, 0.00)
  current_position=(-45.00, 0.00, 5.00) : sync_plan_position
<<< do_homing_move(Y)
Move Away:
>>> do_homing_move(Y, 5.00, 0.00)
  current_position=(-45.00, 0.00, 5.00) : sync_plan_position
<<< do_homing_move(Y)
Home 2 Slow:
>>> do_homing_move(Y, -10.00, 25.00)
  current_position=(-45.00, 0.00, 5.00) : sync_plan_position
<<< do_homing_move(Y)
>>> set_axis_is_at_home(Y)
For Y axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = -5.00
 soft_endstop_max = 220.00
> home_offset[Y] = 0.00
  current_position=(-45.00, -5.00, 5.00) :
<<< set_axis_is_at_home(Y)
  current_position=(-45.00, -5.00, 5.00) : sync_plan_position
  current_position=(-45.00, -5.00, 5.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
  current_position=(-45.00, -5.00, 5.00) : > homeY
Z_SAFE_HOMING >>>
  current_position=(-45.00, -5.00, 5.00) : sync_plan_position
  destination=(110.00, 110.00, 5.00) : Z_SAFE_HOMING
>>> do_blocking_move_to(110.00, 110.00, 5.00)
<<< do_blocking_move_to
>>> homeaxis(Z)
Home 1 Fast:
>>> do_homing_move(Z, -345.00, 0.00)
  current_position=(110.00, 110.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, 2.00, 0.00)
  current_position=(110.00, 110.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, -4.00, 1.00)
  current_position=(110.00, 110.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
>>> set_axis_is_at_home(Z)
For Z axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 230.00
> home_offset[Z] = 0.00
  current_position=(110.00, 110.00, 0.00) :
<<< set_axis_is_at_home(Z)
  current_position=(110.00, 110.00, 0.00) : sync_plan_position
  current_position=(110.00, 110.00, 0.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Z)
<<< Z_SAFE_HOMING
  current_position=(110.00, 110.00, 0.00) : > (home_all_axis || homeZ) > final
  current_position=(110.00, 110.00, 0.00) : sync_plan_position
  current_position=(110.00, 110.00, 0.00) : clean_up_after_endstop_or_probe_move
<<< gcode_G28
echo:M105
echo:M105
echo:M105
echo:M105

Hope you can help me.

thinkyhead, 2017-11-25:
Double-check your configuration and make sure `Z_SAFE_HOMING_X_POINT` and `Z_SAFE_HOMING_Y_POINT` are defined the way you want in `Configuration.h`. For the middle of the bed, the default values (i.e., `((X_BED_SIZE) / 2)`) should work. Since your `X_MIN_POS` is -45 I assume that the printable corner of your bed starts at X = 0 and that the far corner is at X = 220, making 110,100 the middle.

I don't know what you're trying to do with `#define BED_CENTER_AT_110_110`. 😕  Just leave it disabled (commented out) with the original name `#define BED_CENTER_AT_0_0`.

cliftar1, 2019-02-22:
I am having the same issue.  I just updated to Marlin 1.1.9 and love it.  But now for some reason its homing like your were describing.   How did you solve the issue.  I am wondering if having Z_safe_homing is causing the problem.

AnHardt, 2019-02-22:
Most likely not.
More likely `#define BED_CENTER_AT_0_0`.

Almost always a beginners MeTo has completely different reasons.

For a real answer open a new issue and follow the advises.
Describe the problem in your words.
Don't make assumptions where the error comes from. This usually distracts from the real sources.
In this case i'd like to see an additional serial log from booting until the first printing moves.