MarlinFirmware/Marlin #8430: Endstop pin on Rumba board is not interrupt-capable?

heming3501, 2017-11-15:
The default pin for X-MIN is 37, and when I upload the firmwire, an error message showed up... Should I change the endstop pin definition, or disable the endstop interrupts feature instead?

In file included from sketch\Marlin_main.cpp:320:0:

sketch\endstop_interrupts.h: In function 'void setup_endstop_interrupts()':

endstop_interrupts.h:128: error: static assertion failed: X_MIN_PIN is not interrupt-capable

       static_assert(digitalPinToPCICR(X_MIN_PIN) != NULL, "X_MIN_PIN is not interrupt-capable");

       ^

endstop_interrupts.h:148: error: static assertion failed: Y_MIN_PIN is not interrupt-capable

       static_assert(digitalPinToPCICR(Y_MIN_PIN) != NULL, "Y_MIN_PIN is not interrupt-capable");

       ^

endstop_interrupts.h:168: error: static assertion failed: Z_MIN_PIN is not interrupt-capable

       static_assert(digitalPinToPCICR(Z_MIN_PIN) != NULL, "Z_MIN_PIN is not interrupt-capable");

       ^

exit status 1
static assertion failed: X_MIN_PIN is not interrupt-capable

Roxy-3D, 2017-11-15:
I've spent zero time digging in and looking up facts.   But only some of the GPIO pins on the AVR can be setup to provide interrupts.   My guess is, your X_MIN_PIN was arbitrarily put on a uninterruptible pin by the board designer.    (I don't know that for a fact....  But that is my starting position.)

thinkyhead, 2017-11-15:
You can enable `PINS_DEBUGGING` and then send `M43` to get a full list of all pins and what their capabilities are. If you have enough unused interrupt-capable pins you may be able to move your endstops to those.

heming3501, 2017-12-22:
Sorry for the late reply, the board I am using is MKS Rumba+ board, and they states it has protection circuits for the endstop pins.
`PIN:   0   Port: E0        RXD                         protected
PIN:   1   Port: E1        TXD                         protected
PIN:   2   Port: E4        HEATER_0_PIN                protected
PIN:   3   Port: E5        <unused/unknown>            Input  = 0    TIMER3C   PWM:     0    WGM: 1    COM3C: 0    CS: 3    TCCR3A: 1    TCCR3B: 3    TIMSK3: 0
PIN:   4   Port: G5        SPINDLE_LASER_PWM_PIN       Input  = 0    TIMER0B   PWM:   128    WGM: 3    COM0B: 0    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   compare interrupt enabled   overflow interrupt enabled
PIN:   5   Port: E3        SERVO0_PIN                  Input  = 0    TIMER3A   PWM:     0    WGM: 1    COM3A: 0    CS: 3    TCCR3A: 1    TCCR3B: 3    TIMSK3: 0
PIN:   6   Port: H3        HEATER_BED_PIN              protected
PIN:   7   Port: H4        FAN_PIN                     protected
PIN:   8   Port: H5        E0_AUTO_FAN_PIN             protected
.                          FAN1_PIN                    protected
.                          HEATER_3_PIN                protected
PIN:   9   Port: H6        HEATER_2_PIN                Output = 0    TIMER2B   PWM:     0    WGM: 1    COM2B: 0    CS: 4    TCCR2A: 1    TCCR2B: 4    TIMSK2: 0
PIN:  10   Port: B4        <unused/unknown>            Input  = 0    TIMER2A   PWM:     0    WGM: 1    COM2A: 0    CS: 4    TCCR2A: 1    TCCR2B: 4    TIMSK2: 0
PIN:  11   Port: B5        BTN_EN1                     Input  = 1    TIMER1A   PWM:  2000    WGM: 4    COM1A: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode   compare interrupt enabled
PIN:  12   Port: B6        BTN_EN2                     Input  = 1    TIMER1B   PWM:     0    WGM: 4    COM1B: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
PIN:  13   Port: B7        LED_PIN                     Input  = 0    TIMER0A   PWM:     0    WGM: 3    COM0A: 0    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   overflow interrupt enabled
 .                  TIMER1C is also tied to this pin                  TIMER1C   PWM:     0    WGM: 4    COM1C: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
PIN:  14   Port: J1        SPINDLE_LASER_ENABLE_PIN    Input  = 0
PIN:  15   Port: J0        SPINDLE_DIR_PIN             Input  = 0
PIN:  16   Port: H1        X_DIR_PIN                   protected
PIN:  17   Port: H0        X_STEP_PIN                  protected
PIN:  18   Port: D3        LCD_PINS_D4                 Output = 1
PIN:  19   Port: D2        LCD_PINS_RS                 Output = 0
PIN:  20   Port: D1        SDA                         Input  = 1
PIN:  21   Port: D0        SCL                         Input  = 1
PIN:  22   Port: A0        E0_DIR_PIN                  protected
PIN:  23   Port: A1        E0_STEP_PIN                 protected
PIN:  24   Port: A2        E0_ENABLE_PIN               protected
PIN:  25   Port: A3        E1_DIR_PIN                  Output = 0
PIN:  26   Port: A4        E1_STEP_PIN                 Output = 0
PIN:  27   Port: A5        E1_ENABLE_PIN               Output = 1
PIN:  28   Port: A6        E2_DIR_PIN                  Output = 0
PIN:  29   Port: A7        E2_STEP_PIN                 Output = 0
PIN:  30   Port: C7        <unused/unknown>            Input  = 0
PIN:  31   Port: C6        <unused/unknown>            Input  = 0
PIN:  32   Port: C5        <unused/unknown>            Input  = 1
PIN:  33   Port: C4        Z_MIN_PIN                   protected
PIN:  34   Port: C3        <unused/unknown>            Input  = 1
PIN:  35   Port: C2        Y_MIN_PIN                   protected
PIN:  36   Port: C1        <unused/unknown>            Input  = 1
PIN:  37   Port: C0        X_MIN_PIN                   protected
PIN:  38   Port: D7        LCD_PINS_D5                 Input  = 0
PIN:  39   Port: G2        E2_ENABLE_PIN               Output = 1
PIN:  40   Port: G1        LCD_PINS_D7                 Input  = 0
PIN:  41   Port: G0        LCD_PINS_D6                 Input  = 0
PIN:  42   Port: L7        LCD_PINS_ENABLE             Output = 0
PIN:  43   Port: L6        BTN_ENC                     Input  = 1
PIN:  44   Port: L5        BEEPER_PIN                  Output = 0    TIMER5C   PWM:     0    WGM: 1    COM5C: 0    CS: 3    TCCR5A: 1    TCCR5B: 3    TIMSK5: 0
PIN:  45   Port: L4        CASE_LIGHT_PIN              protected
.                          PS_ON_PIN                   protected
PIN:  46   Port: L3        KILL_PIN                    Input  = 1    TIMER5A   PWM:     0    WGM: 1    COM5A: 0    CS: 3    TCCR5A: 1    TCCR5B: 3    TIMSK5: 0
PIN:  47   Port: L2        Y_DIR_PIN                   protected
PIN:  48   Port: L1        X_ENABLE_PIN                protected
PIN:  49   Port: L0        SD_DETECT_PIN               Input  = 0
PIN:  50   Port: B3        AVR_MISO_PIN                Input  = 1
.                          MISO_PIN                    Input  = 1
PIN:  51   Port: B2        AVR_MOSI_PIN                Output = 1
.                          MOSI_PIN                    Output = 1
PIN:  52   Port: B1        AVR_SCK_PIN                 Output = 0
.                          SCK_PIN                     Output = 0
PIN:  53   Port: B0        AVR_SS_PIN                  Output = 1
.                          SDSS                        Output = 1
.                          SS_PIN                      Output = 1
[... cut ...]
PIN:  63   Port: K1 (A 9)  <unused/unknown>   Analog in =   556   Input  = 1
PIN:  64   Port: K2 (A10)  <unused/unknown>   Analog in =   745   Input  = 1
PIN:  65   Port: K3 (A11)  TEMP_BED_PIN                protected
PIN:  66   Port: K4 (A12)  <unused/unknown>   Analog in =  1023   Input  = 1
PIN:  67   Port: K5 (A13)  TEMP_2_PIN                  Analog in =  1023
PIN:  68   Port: K6 (A14)  TEMP_1_PIN                  Analog in =  1023
PIN:  69   Port: K7 (A15)  TEMP_0_PIN                  protected`

thinkyhead, 2018-02-11:
In `M43` the "protected" label means the pin is being protected from change by Marlin.

closed 2018-02-11 by thinkyhead