MarlinFirmware/Marlin #8436: Board resetting while unloading the second extruder

vanvanven, 2017-11-15:
I am using dual extruder with different Esteps. While trying to unload in one of the extruder ( 10 mm or more )which's Estep is 2560, the board resets. 
I tried to change the Estep with different values like 400 ,there is no problem. I swapped the motors and drives the issue remains.
For me its happening only when one of the extruder Estep is 2560.

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1473869/Configuration_adv.zip)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1473870/Configuration.zip)

studiodyne, 2017-11-15:
Try to test this , but unconnect the motors , to see if it's not a power outage or a bad motor driving , and to see if the board reset because of software crash , or power supply failure

thinkyhead, 2017-11-15:
What version of Marlin are you using?

vanvanven, 2017-11-16:
Hi i am using marlin 1.1.6.  I tried to work on it with different stepper drivers and motors,
 I can only retract the extruder at 1/32 micro stepping ( driver drv8825),here my estep was 5120. The retraction is not working in 1/16 micro stepping in the same driver and the motor making some noise too..
when i used A4988 as the driver the retraction of extruder is not working in both 1/8  & 1/16 .
I used these testing for the single extruder setup also, there also its causing the same problem with the Estep step value of 2560 at 1/16 micro stepping.
Thank you

thinkyhead, 2017-11-17:
The AVR processor can't handle 32x microstepping at high speeds and if you try to push the processor past its limits (which prevents the sensors from being read) it will cause Marlin to reboot as a safety precaution. For extrusion there's no good reason to use 32x microstepping. You should go back to 8x or 32x.

closed 2017-11-17 by thinkyhead

vanvanven, 2017-11-18:
Hi, the resetting problem stopped. I used to 32x microstepping to test the board is faulty or not, because only in 32x both of my extrusion and retraction working well. In 16x and 8x only the extrusion is working. while trying to retract the material, the motor makes some noise and it's not moving at all.
I am using 16x  for all my three axes. 
I am not using extruder set up for filament extrusion. I am working chocolate extruder using threaded rod mechanism that's how I came in 2560 Estep value.
My past setup was with printrboard. I used same Estep(2560) for extruder there. In that board it worked well without any problem.,  thank you