MarlinFirmware/Marlin #8440: Compiling for STM32F7, problem...

Spawn32, 2017-11-15:
Hi, 
By making a new board in STM32GENERIC-master for arduino and adapting https://github.com/Aus3D/Marlin/tree/bugfix-2.0.x-F446-PicoPrint for the F7 i can now compile that version without error's even with the older TMC2660 drivers.

So the next step was to adapt it to the current release of bugfix-2.0 (the release for Picoprint is about a month old), most went well, but the i end up with some errors that i think has something to do with what was done when debug was moved to serial.h / cpp by @thinkyhead about the same time.

Is it some debug stuff i can force off or is there something i need to modify ?
have included the modded bugfix and also the STMGeneric..
Error:
Build options changed, rebuilding all
WARNING: library U8glib claims to run on (avr, sam) architecture(s) and may be incompatible with your current board which runs on (STM32) architecture(s).

In file included from sketch\src/inc/MarlinConfig.h:47:0,

                 from C:\Users\Familien\Desktop\PicoPrint-master\Marlin-bugfix-2.0.x\Marlin\Marlin.ino:23:

c:\users\familien\appdata\local\temp\arduino_build_782223\sketch\src\core\serial.h:37:30: error: '_BV' was not declared in this scope

   DEBUG_ECHO          = _BV(0), ///< Echo commands in order as they are processed

                              ^

c:\users\familien\appdata\local\temp\arduino_build_782223\sketch\src\core\serial.h:38:30: error: '_BV' was not declared in this scope

   DEBUG_INFO          = _BV(1), ///< Print messages for code that has debug output

                              ^

c:\users\familien\appdata\local\temp\arduino_build_782223\sketch\src\core\serial.h:39:30: error: '_BV' was not declared in this scope

   DEBUG_ERRORS        = _BV(2), ///< Not implemented

                              ^

c:\users\familien\appdata\local\temp\arduino_build_782223\sketch\src\core\serial.h:40:30: error: '_BV' was not declared in this scope

   DEBUG_DRYRUN        = _BV(3), ///< Ignore temperature setting and E movement commands

                              ^

c:\users\familien\appdata\local\temp\arduino_build_782223\sketch\src\core\serial.h:41:30: error: '_BV' was not declared in this scope

   DEBUG_COMMUNICATION = _BV(4), ///< Not implemented

                              ^

c:\users\familien\appdata\local\temp\arduino_build_782223\sketch\src\core\serial.h:42:30: error: '_BV' was not declared in this scope

   DEBUG_LEVELING      = _BV(5), ///< Print detailed output for homing and leveling

                              ^

c:\users\familien\appdata\local\temp\arduino_build_782223\sketch\src\core\serial.h:43:30: error: '_BV' was not declared in this scope

   DEBUG_MESH_ADJUST   = _BV(6), ///< UBL bed leveling

                              ^

exit status 1
Error compiling for board The Borg.

[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/1475148/Marlin-bugfix-2.0.x.zip)

[STM32GENERIC-master.zip](https://github.com/MarlinFirmware/Marlin/files/1475176/STM32GENERIC-master.zip)

Spawn32, 2017-11-15:
Would really appreciate if one of the dev's could help me out with this one 😄

AnHardt, 2017-11-15:
For now you can add
```
  #define _BV(bit) (1UL << (bit))
```
in `serial.h`
Usually it's defined in 'arduino.h',  'fastio_AVR.h', `HAL_Due.h`, `HAL_Stm32f1.h` and `LPC176x.h`

Spawn32, 2017-11-15:
i actually had that in the HAL_Stm32f7.h, found the problem, i really messed up while copy pasting the path to the f7 hal files in the main hal directory, sometime it just helps with some input to get your eyes open again, thanks @AnHardt  back on track for now :)

closed 2017-11-15 by Spawn32