MarlinFirmware/Marlin #8441: EPCOS B57560G104F - which thermistor type in Marlin

tekkydave, 2017-11-15:
Not an issue just a question. I have some EPCOS B57560G104F thermistors purchased from Ooznest. They say it should be set up as #1 but I think it should be #3. Can anyone shed any light on this?
Thanks.

AnHardt, 2017-11-15:
1 and 3 are very similar. Likely the difference gets lost in the tolerance of the 4.7k pullup.
Take whatever you want. Or make your own table for the exact hardware you have.

thinkyhead, 2017-11-15:
To add to what Andreas said, the best tool for the job is ultimately an accurate laser thermometer that you can check against the temperature reading. If you find a thermistor or a batch of them that are unique, there's a script included with Marlin to generate a new thermistor table. But our current selection is surprisingly resilient. We haven't added or changed thermistor tables for a long, long while.

closed 2017-11-15 by thinkyhead

tekkydave, 2017-11-16:
Thanks for the comments. I have an IR thermometer but it wont read the heat block temperature due it being metal and low emissivity. Any tips on how to get round this? Thanks.

thinkyhead, 2017-11-17:
> Thanks for the comments. I have a laser thermometer but it wont read the heat block temperature due it being metal and low emissivity. Any tips on how to get round this? Thanks.

How annoying! Well, you could try some other kind of well-calibrated thermometer. Otherwise… paint it black with india ink … ? … apply a single layer of Kapton or PTFE plumber's tape … ? …

No idea if any of that would work.