MarlinFirmware/Marlin #8443: [FR] Calibrating against a standard

rszasz, 2017-11-15:
Would it be feasible to add a calibration step that touches off from not only the bed, but something like a 1x2x3 block? For Delta's especially, actually sensing how many steps it takes to move a given distance would be a big step towards a fully self contained calibration.

AnHardt, 2017-11-15:
Except for the extruder every measured value for steps/mm is less exact than a calculated one.
https://www.prusaprinters.org/calculator/
In x and y steps/mm depends on where you are (DELTA).

rszasz, 2017-11-15:
Perhaps not so much calibration as sanity check of given values then? You can zero out the bed height, and level it, while your absolute measurements are totally out of whack. For Deltas probing a point raised off the bed in the center and near each pillar would catch such things, right? I'm still trying to figure out why my printed objects Z axis is squashed, even after using the hole depth pin on my cheap micrometer as a travel gauge (and learning I'm getting exactly the right amount of travel). That's what made me think it would be nice to verify/calibrate as much as possible so that the troubleshooting of inevitable print problems is easier.

AnHardt, 2017-11-15:
Lets calculate:
height 500mm -> belt-length 1m -> GT2, 500 teeth -> 20 teeth per pulley, 200 steps/rotation * 16 microsteps/step ->   80 µsteps/mm
If we give a bit more tension, lets say 2mm, height will be 502mm -> belt length 1004mm, still 500 teeth -> 80steps/1.004 mm -> 79,681274900398406374501992031873µsteps/mm

AnHardt, 2017-11-15:
A (DELTA) move in z is done with stepping a three towers steppers the same amount of steps. Measuring in the center and at the towers is pointless.

rszasz, 2017-11-15:
(T.T) STL files are a pain to measure.. turns out the Z dimension given in the calibration description were for a different version.... getting close to .1MM over 60mm accuracy now after a few iterations.

rszasz, 2017-11-15:
This sort of secondary testing might help find something weird like out of round pulleys, but I'm closing this as a bad idea unless someone can come up with a better use case than I did.

closed 2017-11-15 by rszasz

thinkyhead, 2017-11-15:
The most tricky things to calibrate are Delta geometry and extrusion steps. For the Delta tuning issue we now have `G33` to zero in on the best settings. And for extrusion, it depends on so many factors, even down to the accuracy of the filament width in a specific roll, and even the density and expansion of the material. So, we don't have any straightforward exercises for that.