MarlinFirmware/Marlin #8444: problem with Z0 and home

ruggb, 2017-11-15:
don't know where to start or how to explain because things change so bear with me.
I have tried the latest 1.1.x-bf back to ca97348 and I never noticed this b4.
I do a home all and everything seems fine. It stops on endstop signal.
I then run scripts in Repetier to test the bed level - I have been using these for a long time and not noticed this. However, I did change from G28 Z to G0 Z0. It just goes to each corner and tests Z=0.
When it executes G0 Z0 it stops in random places and is not even looking at the endstop. I guess it is suppose to do that from memory. However, it is not stopping in the same place as G28 Z. Sometimes it is lower, sometimes higher.

With the latest build 9cf5b96 there are more issues - it is totally getting lost, changing speed and studdering at the end of XY travel.

Printing has been fine with no apparent related issue on 1b09733. Following that I have not tried to print because of other strange behavior related to movement.
What to look for????????

thinkyhead, 2017-11-15:
Thanks for providing more details on this. I applied an overhaul to the coordinate system recently and apparently missed some loose ends that relied on `current_position` being explicitly offset. Today I plan to do a bunch of testing to see where the glitches are, and I will try some of those linked commits to compare the results.

ruggb, 2017-11-15:
steve - not sure of that. this issue may have been from long ago. I am running ca... now because I really wanted to print some things. But the issue may be an old one. It seems to print OK
BTW I have a coreXY
the newest builds seem to have additional issues. I have not used them because when I loaded and started to run them weird things happened.

thinkyhead, 2017-11-15:
> the newest builds seem to have additional issues. I have not used them because when I loaded and started to run them weird things happened.

Ah… Do you mean, newer than 1.1.6 but not as new as today's code pile?

ruggb, 2017-11-15:
no i think it is today's 9cf5b96

thinkyhead, 2017-11-15:
I'm away from my machine right now, but I will start digging into these problems in a couple of hours. The "native machine space" patch was long overdue, so there's no going back. Glitches were definitely expected, and we were not disappointed. These last few subtle ones are possibly going to be the hardest to track down. While I'm unable to use a full machine right now, I can do some testing on a spare RAMPS. I can also do a full read-through on the code changes since 1.1.6 to see if anything seems obviously wrong. It's the changes that _didn't_ get made that _should've_ that will be the hardest to find.

ruggb, 2017-11-16:
glad to help if I can. I'm not a coder, but am an EE with lots of experience and a corexy that I built.

thinkyhead, 2017-11-16:
> I then run scripts in Repetier to test the bed level
>  G0 Z0 … stops in random places and is not even looking at the endstop
> However, it is not stopping in the same place as G28 Z.

Of course G0 Z0 simply moves Z down to whatever 0 is at each point. It shouldn't do the same thing as `G28 Z`, which will keep going down until the endstop/probe is triggered. If bed leveling is enabled, it should be more-or-less at the "leveled distance" to the bed. Depending on which type of bed leveling you use there may be differences in accuracy.

After the `G0 Z0` I presume you are then doing a `G0 Z10` or something to move back up, before going to the next corner…?

I suggest trying some different forms of bed leveling — BILINEAR and UBL — and compare results. If you see the problem with one type and not the other, that will help us get closer to the answer.

ruggb, 2017-11-17:
I am not using bed leveling - maybe there is some issue there, like it is affecting the positioning but not enabled.
each script raises Z5, moves to XY location and lowers Z0
So I would expect that Z0 is the same point as G28 Z0.
since my print bombed, maybe it will run a few more tests.

thinkyhead, 2017-11-17:
Maybe your Z axis is physically slipping. Check the grub screws on your couplers.

ruggb, 2017-11-17:
if it were i would have some pretty ugly prints.

ruggb, 2017-11-17:
I did more test - I found an issue with my bed that might explain the variations.
I also reloaded yesterday's build and it did not do the weird things it did the first time - yet.
That would not be a function of the bed problem. So I have no idea what is going on at this point.
I went thru the configs again b4 I reloaded it and adjusted a few things but none should have effected that strange behavior. I will keep running it and see if anything reoccurs.
thanks

Tannoo, 2017-11-17:
Can you post the scripts you are running?

ruggb, 2017-11-17:
more info - 
the first time I ran after reinstalling the new build it did the same thing.
it started home at maybe 9000 then stopped and restarted at maybe 2000
when it got near X0 it studdered about 3 times and proceeded to home all.
after another total reset it seemed to be ok.
In running the leveling test again, it is still using something besides home 0 for Z0.
I do a home all
run the tests
readjust the endstop
do a home Z
and it is maybe 1mm off the bed.
script for test at center - others are the same except for the position.
G0 Z5 F9000
G0 X150 Y165 F9000
G0 Z0 F9000

I do not see the studdering on previous versions.
The offset from home does not seem to be a function of the bed problem as I monitored that during the test, but I am fixing that and will report.

ruggb, 2017-11-17:
I was previously - a long time ago - using a home to test so I never noticed this. For whatever reason I changed it to just a G0 - probably because the home bounced.

boelle, 2019-02-19:
@ruggb have you tried bugfix 2.0?

ruggb, 2019-02-23:
another forgotten issue

closed 2019-02-23 by ruggb