MarlinFirmware/Marlin #8448: Marlin 1.1.6 Delta

prahjister, 2017-11-16:
### Bug Report
- Description: --- UI sluggish unresponsive
- Expected behaviour: --- not sluggish unresponsive
- Actual behaviour: --- very hard to use menu timeout or blank screen
- Steps to reproduce: print anything 

Also have to power cycle to recognize SDcard

I will upload config in the morning. Thank you.

Using mks 1.4 with 2560 and 4 line screen with sdcard

He3d k280

I thought slowing down print would help but doesn't. I have not slowed all the way down.

thinkyhead, 2017-11-16:
Do you see the same issue with `bugfix-1.1.x` branch? It's possible the issue you are seeing is fixed in the latest code.

prahjister, 2017-11-16:
I will test. Thank you for your efforts. I have had really good success and only this minor issue. Auto calibration rocks. I was printing in about 30 minutes coming from repetier.

thinkyhead, 2017-11-16:
I'm gratified to hear that auto-calibration works well! @LVD-AC puts a lot of heart and soul into it.

prahjister, 2017-11-16:
UI is still sluggish.  I did slow the print down more with the knob and returns to normal. I think that this is an acceptable solution. I guess the next question is there anything that I can do to reduce the load on the processor?
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1479038/Configuration.zip)

LVD-AC, 2017-11-16:
diminish DELTA_SEGMENTS_PER_SECOND 200 to e.g. 160

thinkyhead, 2017-11-17:
We put a huge amount of effort last year into optimizing graphical display drawing and concocting a method to throttle back display updates when the planner buffer is in danger of starving. If your display is being unresponsive, it is likely due to too-small segments being sent to the planner. Reducing the `DELTA_SEGMENTS_PER_SECOND` setting will produce slightly longer segments without slowing down the print.

Also try playing with the following settings to see if you can get the screen to draw more efficiently:
```cpp
/**
 * Additional options for Graphical Displays
 *
 * Use the optimizations here to improve printing performance,
 * which can be adversely affected by graphical display drawing,
 * especially when doing several short moves, and when printing
 * on DELTA and SCARA machines.
 *
 * Some of these options may result in the display lagging behind
 * controller events, as there is a trade-off between reliable
 * printing performance versus fast display updates.
 */
#if ENABLED(DOGLCD)
  // Enable to save many cycles by drawing a hollow frame on the Info Screen
  #define XYZ_HOLLOW_FRAME

  // Enable to save many cycles by drawing a hollow frame on Menu Screens
  #define MENU_HOLLOW_FRAME

  // A bigger font is available for edit items. Costs 3120 bytes of PROGMEM.
  // Western only. Not available for Cyrillic, Kana, Turkish, Greek, or Chinese.
  //#define USE_BIG_EDIT_FONT

  // A smaller font may be used on the Info Screen. Costs 2300 bytes of PROGMEM.
  // Western only. Not available for Cyrillic, Kana, Turkish, Greek, or Chinese.
  //#define USE_SMALL_INFOFONT

  // Enable this option and reduce the value to optimize screen updates.
  // The normal delay is 10µs. Use the lowest value that still gives a reliable display.
  //#define DOGM_SPI_DELAY_US 5
#endif // DOGLCD
```

prahjister, 2017-11-17:
I adjusted the DELTA_SEGMENTS_PER_SECOND to 180 and uncommented #define DOGM_SPI_DELAY_US 5. After clearing eeprom and starting over.

Seems to have cleared the issue. Sometimes when I press the button to change setting midprint it goes blank but if i turn the wheel it appears. 

Might be worthwhile to change the title to include sluggish UI or slow updates

The sdcard randomly doesn't work until a power cycle when removed and reinsterted. Will search issues to try to find solution

Thanks again for all your effort.

prahjister, 2017-11-17:
It appears to be a poorly made SD Card. Tried couple other SD cards and they work fine. Used proper sd card formatter and still no luck. Problem solved

novio8, 2017-11-26:
Hi prahjister,
you wrote that you came from Repetier FW. 
What are your expiriences with Marlin VS Repetier FW on Delta.
I installed Marlin first on Mega + Discount LCD. Now I"m using Repetier
printing is not "smooth" as I would like.

prahjister, 2017-11-26:
I did not print anything big I could never get it calibrated well enough without gouging the plate in repetier. I only printed small items.  Marlin seems like it is working very well for me.

closed 2018-02-09 by thinkyhead