MarlinFirmware/Marlin #8449: [FR]  Follow z-hight with extruder or camera

Hywelmartin, 2017-11-16:
Feature Request

I have a delta and a spare stepper E1 output that that could be used for a new axis to follow print hight. my plan was to mount the my camera and a swivelling extruder on this axis.. but I need help to code this

thinkyhead, 2017-11-17:
This is not a straightforward thing to do at all. Changes would need to be made all the way down to the Stepper::ISR. My suggestion would be to enable `Z_DUAL_STEPPER_DRIVERS` which will automatically assign the extra Z stepper to your next unused E stepper driver. Then build your camera rig to use exactly the same Z raise mechanism as your machine, or otherwise gear it so that the same number of steps results in the same amount of movement.

Hywelmartin, 2017-11-17:
If it was an cartesian printer that seems like an solution, but on an delta printer? I tough that this feature [Z_DUAL_STEPPER_DRIVERS] sends the same information to driver module X and E1..

AnHardt, 2017-11-17:
Marlin is made for 4 independent axes (XYZE or ABCE). 
Wanting more means a more or less complete rewrite.
So - no way.

Hywelmartin, 2017-11-17:
@AnHardt That was already clarified , thank you for your time spent..

But what happens if I enable the Z_DUAL_STEPPER_DRIVERS feature when running delta kinematics, as I mentioned before, Will the A(X) stepper and the E1 move the same?

I can try that myself, but if anybody knows it's less of an hassle ....

thinkyhead, 2017-11-18:
Yeah, sorry. `Z_DUAL_STEPPER_DRIVERS` is not possible on DELTA. Here's another concept:

Implement your camera rig as E1. Add to your G-code something akin to this:
```gcode
T1 ; Select camera rig (E0)
M83
E0.2 ; Raise by 1 layer or so
M82
T0 ; Back to E0
```gcode
This won't work completely without a code tweak to preserve the E position for T0. But it will make it possible to move the camera axis between layers. You just need to enable `DISTINCT_E_FACTORS` and supply the correct `STEPS_PER_UNIT` for E1.

Hywelmartin, 2017-11-18:
I have to look in to that next week. My main use for this is not camera but extruder( sort of flying to reduce bowden tube length) but the camera will tag along

Hywelmartin, 2017-11-26:
Closing this one

closed 2017-11-26 by Hywelmartin