MarlinFirmware/Marlin #8451: baby z not working during skirt?

ir-fuel, 2017-11-16:
### Bug Report
- Description: 
Modifying the baby z steps during skirt doesn't do anything

- Expected behaviour: 
The Z axis should move

- Actual behaviour:
Nothing happens

- Steps to reproduce:
Start a print on my CR-10S, both with 1.1.6 and 1.1.x-bugfix branches after bed levelling. Baby Z is at 0, z axis endstop is not engaged. It starts printing the skirt and I see the nozzle is way too high (it's like dropping toothpaste), so I start changing the baby Z. No matter what I do (positive or negative values), nothing happens, I keep getting the same toothpaste until the printer starts on the actual print, and there the z offset changes and I can hear when turning the knob that the stepper motors actually adjust the z distance to the bed.

ir-fuel, 2017-11-16:
Hmm, it seems more complicated than that. It's like there is some kind of delay. I will investigate further and try to do some measurements and come back to this.

tcm0116, 2017-11-16:
What bed leveling method are you using?

ir-fuel, 2017-11-16:
9 point bed leveling

tcm0116, 2017-11-16:
Linear, Bilinear, UBL?

thinkyhead, 2017-11-16:
Please include your `Configuration.h` and `Configuration_adv.h` when reporting. You can rename them with "`.txt`" file extensions or ZIP them up, and then drop them onto your reply.

ir-fuel, 2017-11-17:
here you go

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1482239/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1482240/Configuration.txt)

Rigid3D, 2017-11-17:
This is not meaningful. During the print Marlin does not know whether it is a skirt, bottom or infill.

ir-fuel, 2017-11-17:
I agree.

thinkyhead, 2017-11-18:
Might as well drop us the G-code file you're using too…

thinkyhead, 2017-11-22:
> During the print Marlin does not know whether it is a skirt, bottom or infill.

True. The only thing I can see that would stop babystepping is if the Z motor was allowed to turn off. Try this setting and see if it helps…
```cpp
#define DISABLE_INACTIVE_Z false  // set to false if the nozzle will fall down on your printed part when print has finished.
```

ir-fuel, 2017-11-22:
I had it happen again when testing. I was printing a test square that drew long lines along the print bed, and while turning the knob to adjust baby z it was like nothing happened and then all of the sudden the nozzle went up way too high and filament came out like toothpaste on my bed instead of being squished.

Bit short on time atm, but I will try to make a video of what's going on and also put the gcode file here, as I can't make any sense out of all this while it should be straightforward.

AnHardt, 2017-11-22:
Just a suspicion.
Babysteps are sent from the temp_intettupt to the stepper interrupt.
It needs a working temp_interrupt to reset the watchdog timer.
So likely we have a problem with the temp_interrupt.
Do we have an exit where in_temp_isr is not set to false?
Can the temp_interrupt be interrupted in a way that not returns until a move has finished?
Can a xyz-printer use so much time in any interrupt, so no time is left for the temp_interrupt?
Something in that direction.

thinkyhead, 2017-11-26:
> Do we have an exit where `in_temp_isr` is not set to false?

The temp ISR always goes to the end, where `in_temp_isr` is cleared.

> Can a xyz-printer use so much time in any interrupt, so no time is left for the temp_interrupt?

Possibly, but this would also result in a watchdog timeout.

> Can the temp_interrupt be interrupted in a way that not returns until a move has finished?

This shouldn't be possible.

> Bit short on time atm, but I will try to make a video of what's going on and also put the gcode file here, as I can't make any sense out of all this while it should be straightforward.

I'd like to see that g-code. Marlin doesn't differentiate for any reason between the skirt and the rest of the first layer, or the first layer and the rest of the print. To Marlin it's all just a bunch of meaningless movement commands.

AnHardt, 2017-11-28:
Is the 'skirt' made from many short moves or only a few painting a large rectangle or hexagon (something simple with long lines)?

closed 2018-02-09 by thinkyhead