MarlinFirmware/Marlin #8453: Filament width sensor settings in 1.1.6

Erlandsson, 2017-11-16:
Not really a bug of feature request, just info how it is supposed to work.

In marlin config i have it set to 3.0mm as it is default.
In slicer (s3d) i set (just for testing) 2.8mm diameter.
Then insert a filament that IS 2.8.. 
The display (reprap full graphic controller) says "diameter2.8" and on the right "***125%**".. should it not be **100%** as slicers is set to 2.8mm and filamnent is 2.8mm?
Even if i set slicer to 3.0mm it shows about the same. 

Is it **the setting in marlin config that set the rule** and that can **not** be overridden by slicer?
I really can not see why a diameter needs to be set in firmware.

thinkyhead, 2017-11-16:
For Marlin to be aware of the filament diameter you've set in your slicer, the slicer needs to sends `M404 W[width]` at start. Marlin gets the measured filament diameter from the filament width sensor or you can set it directly with `M200 D[width]`.

So let's run the numbers and see what the value on screen is supposed to be…

The volumetric multiplier is calculated as `1.0 / (π * (diam ÷ 2) * (diam ÷ 2))`. For a filament diameter of `2.8` the cross-sectional area is `(π * 1.4 * 1.4)` or `6.157 sq.mm`. So the volumetric multiplier will be calculated as `1.0 / 6.157` which is `0.162`.

The `volumetric_multiplier` is used to convert an input value in terms of volume back into a mm length for the E axis, so in this case if you were to request 6.157 cubic mm of filament in volumetric extrusion mode, Marlin will advance the extruder by 1mm.

As reference, the cross-sectional area for 3.0mm diameter filament is `(π * 1.5 * 1.5)` or `7.068` sq.mm, giving a volumetric multiplier of `0.141`. `(7.068 * 0.141 ≠ 1.0)`

I'm not sure why the display is showing 125% in your case. The value that the code is using for display is incorrect if it's supposed to be a percentage relative to 3.0mm filament. The LCD code just displays `100.0 * volumetric_multiplier`. This means your `volumetric_multiplier` is close to `1.25`. Thus…
- `1.0 / (π * r * r) = 1.25`
- `(π * r * r) = 0.8`
- `r * r = 0.8 / π`
- `r = sqrt(0.8 / π)`
- `r = 0.504`
- `d = 1.008`

So, your filament width sensor is apparently reporting a value of 1.0 as the filament diameter. You can try giving different values to `M200 D` and see from the screen whether the math is accurate.

----
You've certainly identified a bug in the way the volumetric multiplier is being displayed. Whoever implemented this feature didn't understand the meaning of the `volumetric_multiplier`, and assumed it was a percentage rather than a conversion factor.

To fix the issue, the filament display should instead show the ratio between the nominal filament width (`M404 W`) and the cross-sectional area (or the volumetric multiplier) and the corresponding value for the measured (or `M200 D`) diameter.

In my opinion showing a percentage value is of limited value. If it's just the ratio of filament widths, you pretty much get the same usefulness from the (`2.8`) filament width measurement value.

Anyway, to fix the percentage there are two options. Either show the ratio between the widths, (which is simplest)…
```cpp
strcpy(mstring, itostr3(100.0 * filament_width_meas / filament_width_nominal));
```
For `2.8` measured with `3.0` nominal this will show `93%`.

Or, show the ratio between the cross-sectional areas (which would best be added as a new global to avoid the cost of constant re-calculation):
```cpp
strcpy(mstring, itostr3(100.0 * (1.0 / pi * sq(filament_width_nominal * 0.5))
                              / planner.volumetric_multiplier[FILAMENT_SENSOR_EXTRUDER_NUM]));
```
For `2.8` measured with `3.0` nominal this will show `87%`.

Erlandsson, 2017-11-17:
hmm not much above makes sense to me ;) I thought it should show the extrusion multiplier.. 
As it is now i don't know if it extrudes to much or too little.. (i guess way too much).

Is it a bug that recently got into the software or has it been from the beginning?

Erlandsson, 2017-11-17:
You wrote above:

"For 2.8 measured with 3.0 nominal this will show 93%."

I thought it should/would show a little more than 100% as it has to push out more filament for it to fill 100%.

thinkyhead, 2017-11-18:
> I thought it should/would show a little more than 100% as it has to push out more filament for it to fill 100%.

That makes sense, seen from that perspective. So we need the reciprocal, and the volume relationship. That is, "how far must the extruder move to push out 7.068 cu.mm when we know that 1mm will produce 6.157 cu.mm?" 1.14mm — which is, 114%.

I'll make the correction shortly.

Erlandsson, 2017-11-18:
thanks.. 
is there somehow possible to implement a "blob" filter in the firmware?
I have noticed that sometimes the sensor freaks out and send 250-350% filament which makes some really heavy blobs just for 1-2cm or so. One of the sensor projects on thingiverse had a "anti blob" routine as future fix, but does not seem to be implemented yet, and i am not using his sensor anyway. I am using the "tinywise" version.  A function that filters away ridicilous measurments, or just limits to max setting in firmware. 
The glitches is that the sensor thinks suddenly the filament is 0.5mm diameter or something.

empakoso, 2017-11-18:
The anti-blob idea is a good one! How do we go about getting that in place for our sensors?

In machine sensing we use a method where values are added only if we get x number of sequential samples within a range (say for example the min/max values already set in the fw).  This way sensor sampling’s are muted when they go crazy from outside interferences.

Great term “anti-blob”.  I’ve been experiencing that as well and have stopped using it for that reason.

Erlandsson, 2017-11-19:
Rellay nice to hear someone else using or experimenting with it. I think it is some kind of interference, be it power fluctuations or shadow errors i dont know. I have tried filtering power on the board but it does not help. As they are sudden it should be easy to compare read value with a previous value, and if it is way to big value, use the prevous ones. There is already a big buffer for samples, so it would not be too hard? 
I don't know programming, but guess its a piece of cake for you guys programming the fw.

Erlandsson, 2017-11-19:
I am getting really good results on filamenst varying between 1.8-2,6 (forced bad extrusion on my extruder), can not see a thing on the print, until that crazy misreading.

empakoso, 2017-11-19:
I have overcome a lot of noisy signals with this post.
https://github.com/MarlinFirmware/Marlin/issues/7297

Erlandsson, 2017-11-19:
wow. thanks i will try those things.. and do another fw update to 1.1.5 bugfix.. sigh...

Erlandsson, 2017-11-19:
wait a minute.. i am already on 1.1.6.. hmm

closed 2017-11-25 by thinkyhead

Erlandsson, 2017-11-25:
Why close this? I still have problems. The adding of shielded wires and doodling around i still get blobs in 1.1.6

thinkyhead, 2017-11-26:
The original issue is resolved. Pushing the scope of resolved issues bothers some maintainers, so we ask for a new issue if you want to track a "blob filter" feature request.

Erlandsson, 2017-11-26:
aha right ofcourse.. sorry.