MarlinFirmware/Marlin #8457: Bug: Change delta_height in refresh_zprobe_zoffset?

thinkyhead, 2017-11-16:
Check out this section of `refresh_zprobe_zoffset`:
```cpp
#if ENABLED(DELTA) // correct the delta_height
  delta_height += diff;
#endif
```
This makes no sense, as `delta_height` has nothing to do with the `zprobe_zoffset` unless you're homing Z with the probe. Homing with a probe is unsupported on `DELTA`.

LVD-AC, 2017-11-17:
Pretty sure is needed on delta's but I'll look into it.

BTW where did you get that `+= diff`; it should be `-= diff`

thinkyhead, 2017-11-17:
Look carefully. The sign of the `diff` changed.

thinkyhead, 2017-11-17:
> Pretty sure is needed on delta's but I'll look into it.

I doubt it. The `zprobe_zoffset` isn't used in determining the current Z position, as it is when homing with a probe.

LVD-AC, 2017-11-17:
Deltas do not home at the bed but al the way up at .... Z=delta_height; so delta_height is used to determine the current Z position on deltas.

LVD-AC, 2017-11-17:
notes to myself:

in   static float run_z_probe(const bool short_move=true) {
```
    return RAW_CURRENT_POSITION(Z) + zprobe_zoffset
// probes do not report back any zprobe_zoffset changes
```
in   void refresh_zprobe_zoffset(const bool no_babystep/*=false*/) {
```
      #if ENABLED(AUTO_BED_LEVELING_BILINEAR)
        // Correct bilinear grid for new probe offset
        ...
              z_values[x][y] -= diff;
// bilinear grid is corrected for the zprobe_zoffset change
```

Conclusion: next time G29 is run those diff changes to bilinear grid are undone no longer taking into account the zprobe_zoffset change. 

Interesting....

thinkyhead, 2017-11-17:
See also #8548 / #8456

tcm0116, 2017-11-17:
I think it's there because the delta height is computed as follows:
```
delta_height = z_max_position - z_min_position + zprobe_zoffset
```
The `zprobe_zoffset` is included in the computation because when the `delta_height` is measured using the probe, `z_min_position` is actually the Z position at which the probe is triggered, which means the gap between the bed and the probe at that point needs to be accounted for.

If the user then subsequently adjusts the `zprobe_zoffset`, then the `delta_height` needs to be adjusted accordingly.

thinkyhead, 2017-11-17:
I've already pointed out that you're presuming everyone will be using `G33` to set `delta_height`. You can set `delta_height` with `M665`, with `G33` completely disabled. You've introduced a side-effect that only applies in the narrow case.

The probe can be used for setting the delta height, but the probe is not purposed to do that. The probe is is simply for probing. The `zprobe_zoffset` is meant to be set **_ahead of probing_** to change the result of _**the next probe**_. We can't go retroactively altering everything that might have been set with a probe when someone invokes `M851 Z`.

Downvote. 👎

LVD-AC, 2017-11-17:
Without `delta_height += diff;` the M851 command does nothing for Delta printers, it does not change the print height. The zprobe_zoffset change only takes effect after one recalibrates the printer (at least a recalibration of  the delta_height is required). If people calibrate the height using G33 P1 or do it manually with a paper test and make the changes with M665 Hx.xx or in Configuration.h makes no difference. This applies to all Delta users. But imposing having to recalibrate a Delta after each zprobe_zoffset change is a bit too much of that infamous stepmother treatment Marlin reserves for Delta users. Hence the little tweak : do that recalibration (which only requires a change to delta_height by diff + the necessary postprossing done in G28) to make it work.

On a Delta the probe (or the nozzle in a papertest) IS meant to set the delta_height; this is in every manual "setting up a Delta 101"; first thing to do after you tested the steppers are running in the right direction and the end-stops are working: home the printer and probe or do a papertest in the centre in order to set the delta_height. It is a shame M851 was set up in such a way that it messes up the basics of every delta printer; this is not a narrow case at all. But one extra statement resolves the problem.

This still leaves the requirement to home the printer after each zprobe_zoffset change for the change to take effect; what @tcm0116 mentioned is done in that homing process. It would be nice if the requirement to home would no longer be necessary after a change of one of the delta parameters; but @thinkyhead was going to set up a project for that to test if we can make that happen. (that can wait till after the 2.0.0 release).

After a good night sleep:

- Cartesians have XYZ home_offsets that can altered with M206 (disabled for Deltas)
- Delta only have an additional Z_offset and IMO the correlated command to alter that is M851 (should be disabled for Cartesians) but this is actually not implemented as such in Marlin; that offset needs to be changed by using the probe Z-offset. (which works; but it would leed to less confusion if a proper variable is set up for that)

And like all Cartesians can have home_offsets (probe or no probe) to center and set the correct print height all deltas (probe or no probe) can have an additional Z_offset (they are centered by definition but still need to set the correct print height).

Probe offsets are something completely different; they are fixed in Marlin, but if one wants to implement they can be altered the Gcode for that is M565; of course those offsets only apply to printers that have a probe (Cartesian or Delta). Very few FW have implemented this and most take the probe offsets as fixed, only to be altered in Configuration.h

What I propose is the following:

For Cartesians: 
- allow M206, disable M851
- write an LCD-menu item to change at least the Z home_offset

For Deltas:
- disable M206, enable M851
- create a new variable "z_offset' that can be changed with a new M851 and LCD-menu (just change the value and call recalc_delta_settings), and can be stored and saved to EEPROM
- treat this variable as one of the delta_settings and do the maintenance on it in recalc_delta_settings, properly so people do not have to recalibrate the printer in order for Z-offset changes to take effect, if possible even without the need to home the printer.
- abandon refresh_zprobe_zoffset all together since the probe offsets are fixed

In the code:
- replace all references to `zprobe_zoffset` with `Z_PROBE_OFFSET_FROM_EXTRUDER`
- replace (4-5 if I counted right) instances of `delta_height` with `delta_height - z_offset`

ALTERNATIVE if you want to keep M851:
- leave zprobe_zoffset and M851 unaltered
- make a light tough M206 for deltas that only has an additional Z offset towards the set delta_height

cdedwards, 2017-11-17:
since we are talking about probes here, with 2.0.x there are 2 problems with G33. One of these I have to assume is also in bugfix-1.1.x

probing is failing with G33 P1 on my delta. I've included the log from a G28/G33 P1 run. This leads to the second problem. G33 should not ignore the probe failed and set the height of the delta. there's an error, it shouldn't be ignored. I've confirmed that G33 P1 works with bugfix-1.1.x. it detects the bed and set's the height.

Marlin bugfix-2.0.x
READ: 
READ: echo: Last Updated: 2017-10-19 12:00 | Author: (Colten Edwards)
Last Updated: 2017-10-19 12:00 | Author: (Colten Edwards)
READ: echo:Compiled: Nov 17 2017
Compiled: Nov 17 2017
READ: echo: Free Memory: 2640  PlannerBufferBytes: 1232
Free Memory: 2640  PlannerBufferBytes: 1232
READ: echo:V45 stored settings retrieved (598 bytes; crc 40808)
V45 stored settings retrieved (598 bytes; crc 40808)
READ: Unified Bed Leveling System v1.01 inactive.
SENT: T0
READ: 
READ: Unified Bed Leveling initialized.
READ: 
READ: UBL System reset()
READ: echo:  G21    ; Units in mm
G21    ; Units in mm
READ: echo:  M149 C ; Units in Celsius
M149 C ; Units in Celsius
READ: 
READ: echo:Filament settings: Disabled
Filament settings: Disabled
READ: echo:  M200 D1.75
M200 D1.75
READ: echo:  M200 T1 D1.75
M200 T1 D1.75
READ: echo:  M200 D0
M200 D0
READ: echo:Steps per unit:
Steps per unit:
READ: echo:  M92 X160.00 Y160.00 Z160.00 E400.00
M92 X160.00 Y160.00 Z160.00 E400.00
READ: echo:Maximum feedrates (units/s):
Maximum feedrates (units/s):
READ: echo:  M203 X800.00 Y800.00 Z800.00 E60.00
M203 X800.00 Y800.00 Z800.00 E60.00
READ: echo:Maximum Acceleration (units/s2):
Maximum Acceleration (units/s2):
READ: echo:  M201 X9000 Y9000 Z9000 E10000
M201 X9000 Y9000 Z9000 E10000
READ: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
READ: echo:  M204 P2000.00 R3000.00 T3000.00
M204 P2000.00 R3000.00 T3000.00
READ: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
READ: echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z20.00 E20.00
M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z20.00 E20.00
READ: echo:Unified Bed Leveling:
Unified Bed Leveling:
READ: echo:  M420 S0
M420 S0
READ: 
READ: Unified Bed Leveling System v1.01 inactive.
READ: 
READ: Active Mesh Slot: -1
READ: EEPROM can hold 9 meshes.
READ: 
READ: echo:Endstop adjustment:
Endstop adjustment:
READ: echo:  M666 X0.00 Y0.00 Z0.00
M666 X0.00 Y0.00 Z0.00
READ: echo:Delta settings: L<diagonal_rod> R<radius> H<height> S<segments_per_s> B<calibration radius> XYZ<tower angle corrections>
Delta settings: L<diagonal_rod> R<radius> H<height> S<segments_per_s> B<calibration radius> XYZ<tower angle corrections>
READ: echo:  M665 L360.32 R178.91 H354.44 S160.00 B121.50 X0.00 Y0.00 Z0.00
[... 372 lines cut ...]
M92 X160.00 Y160.00 Z160.00 E400.00
Maximum feedrates (units/s):
M203 X800.00 Y800.00 Z800.00 E60.00
Maximum Acceleration (units/s2):
M201 X9000 Y9000 Z9000 E10000
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
M204 P2000.00 R3000.00 T3000.00
Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z20.00 E20.00
Unified Bed Leveling:
M420 S0
Endstop adjustment:
M666 X0.00 Y0.00 Z0.00
Delta settings: L<diagonal_rod> R<radius> H<height> S<segments_per_s> B<calibration radius> XYZ<tower angle corrections>
M665 L360.32 R178.91 H354.44 S160.00 B121.50 X0.00 Y0.00 Z0.00
Material heatup parameters:
M145 S0 H180 B70 F255
M145 S1 H240 B100 F255
PID settings:
M301 P14.42 I1.07 D48.74
M304 P151.24 I20.75 D275.00
Z-Probe Offset (mm):
M851 Z-2.18
Active Extruder: 0
SD init fail

thinkyhead, 2017-11-25:
G33 is adding error handling with #8556 and #8558.

closed 2019-05-16 by thinkyhead