MarlinFirmware/Marlin #8462: G27 moves out of bounds when T1 selected

bjarchi, 2017-11-17:
[marlin-116-bugreport-g27.zip](https://github.com/MarlinFirmware/Marlin/files/1480826/marlin-116-bugreport-g27.zip)

------------------------------------------------------------
### Bug Report
- Description: using NOZZLE_PARK_FEATURE and G27, the extruder will try to move outside of defined motion bounds if T1 is active when G27 is issued and the G27 park position is less than _toolhead_offset_ mm from the outer motion bounds. This occurs with [min,max]_SOFTWARE_ENDSTOPS enabled and active. 

E.G. - in the attached files, my max_x and max_y values are 249 and 186, respectively, and x_park and y_park are set to (max_x - 5) and (max_y - 5); my HOTEND_OFFSET_(x,y) is (-34, 0). If I have issue a G27 command to park the head when T1 is active - either in a dual extrusion print or when printing T1 only - marlin tries to move T1 to (244, 181), which puts T0 outside the bounds of MAX_X. 

I have not investigated whether this occurs with normal moves as well, or whether the MAX_foo bounds are respected there. I realize this may be 'as coded' behavior but it does create the potential for mechanical damage since endstops are not nominally polled during the non-homing moves.

The simple workaround is to make sure T0 is selected before issuing G27. I could also pull in my max_x value by 34 mm to respect the toolhead offset, but this seems overly conservative. I searched and did not find any other reports describing this issue.

- Expected behaviour: When G27 is issued to park the head, if T1 is active (or other non-T0 tool) then either the final position should be dereferenced to T0 coordinates, or the T1 (Tn) offsets taken into account when validating the move against _SOFTWARE_ENDSTOPS.

- Actual behaviour: When G27 is issued to park the head when T1 is active, the carriage crashes into the max_x endstop (and tries to keep going) because the machine is trying to move T1 to x=244 - which is outside the machine bounds.

- Steps to reproduce:
  Assuming Marlin (current / 1.1.6) has been built using the configuration.h and configuration_adv.h attached, the minimal (gcode) steps to reproduce should be:
  - T1
  - G27
Resulting in the head crashing into the x_max side. I have only observed this behavior after a print; as soon as I saw it happening I added T0 to my final gcode script before G27 and it has not recurred. I'm hesitant to intentionally trigger this issue with a minimal example due to the potential for mechanical damage or misalignment. 

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------

thinkyhead, 2017-11-17:
Does this also occur in the `bugfix-1.1.x` branch? We might already have fixed it.

bjarchi, 2017-11-17:
That is an excellent question; I will test ASAP. I guess I could pull x_max in enough to see whether the behavior persists w/o actually smashing into the endstop. I will report back as soon as I can build and test. 

Are configuration and configuration_adv compatible between 1.1.x and bugfix-1.1.x?

thinkyhead, 2017-11-17:
> Are configuration and configuration_adv compatible between 1.1.x and bugfix-1.1.x?

If there are any changes needed, the compiler should throw an error. You will need to change these to…
```cpp
#define CONFIGURATION_H_VERSION 010107
. . .
#define CONFIGURATION_ADV_H_VERSION 010107
```

bjarchi, 2017-11-17:
I have confirmed that this issue still occurs on bugfix-1.1.x (pulled directly from current git). I realized I could use M120 to enable the endstops, avoiding mechanical stresses while still preserving all the other conditions that initially exposed the bug. 

Behavior (same on 1.1.x and bugfix-1.1.x)
Startup
G28          ; home all (required for G27)
G27          ; correctly positions T0 at park position
G1 X180   ; move off a bit
M120
G27          ; Correctly parks - carriage just short of endstop switches
G1 X180
T1             ; Positions T1 @ selected position
G27          ; FAIL - hits endstop

Again, I would expect that the G27 with T1 active would either position T0 at the park position, or at least stop before taking T0 outside the bounds of the soft endstops. You agree that this is the desired behavior?

I haven't explored the code enough yet to contribute usefully to a fix, but I'm more than happy to help with testing and until I am up to speed. Is it reasonable to keep running this bugfix build on my machine for now, or is that branch considered unstable?

Thanks for the super fast followup!

bjarchi, 2017-11-17:
Possibly related - still on bugfix-1.1.6 / 3dee1063... - I just noticed that issuing a G29 after T1 also results in a coordinate issue. It appears that the probe position was offset by the hotend_offset distance, as if the probe offset were also being applied in the current tool coordinate space despite being defined in T0 space (== machine space?)

I've browsed through some of the other recent tickets, and I see that there's been recent work on how the coordinate spaces are defined. This is probably related, as may be the initial issue I reported. 

Not sure if this should be a separate issue, belongs here, or is best attached to a different issue. I don't want to pollute the issue stream unnecessarily so I'll put it here for now.

studiodyne, 2017-11-17:
Stupid question :  Do you use the right extruder as ' zero tool ' with negative offset for the left tool ?
Because it works at my home , but i have left tool as primary ' zero ' , and the right tool as secondary with 34mm ' positive ' offset

bjarchi, 2017-11-17:
@studiodyne I'm aware that there are workarounds - as I said in the original report, the simplest one (IMO) is to just make sure to issue T0 before G27 in any ending / pause scripts if you want to G27 to park the head. That doesn't change the fact that it is probably a bug for the printhead to try to move out of bounds under the conditions described, nor that there is a coordinate reference issue when initiating auto-probing (auto_level_bilinear in my case) with T1 active.

studiodyne, 2017-11-17:
It's not a tool change issue.
Each class or function that make a movement is free to go everywhere.  And each author of each feature have to make a CLAMP TO SOFTWARE ENDSTOP but need to actualise these for each  selected tools.
G27 use Nozzle CLASS and I'm not sure if smart clamping inside.

Try M125  .... same as G27 and tell us if clamping is ok

bjarchi, 2017-11-18:
I will try M125 when I'm near my machine again, later tonight, and report whether the results are the same. Interesting that each feature has to implement its own soft endstop clamping; thanks for pointing that out.

bjarchi, 2017-11-18:
@studiodyne FWIW, M125 does not work equivalently to G27 - it requires different #defines to be enabled, which are not true in my build, so I cannot verify its operation right now and it's not a functional replacement for G27 in the sense of just enabling 'nozzle park feature'

@thinkyhead I think this has been confirmed as a bug that still exists in bugfix branch (and see also my comment on probe position offset w/ T1 active, which may be a separate or different issue). Please let me know how I can help further to address this or test things out when someone has the time to look into it. 👍

bjarchi, 2017-11-19:
@thinkyhead Bumping, only because I think the specific bug reported has been confirmed (and not because I'm impatient about seeing it fixed).

thinkyhead, 2018-02-09:
Please test the `bugfix-1.1.x` branch and close this issue if the problem has been addressed.