MarlinFirmware/Marlin #8465: LEFT_PROBE_BED_POSITION can't be reached by the Z probe. (15 >= 28) ??

knobik, 2017-11-17:
im trying to configure new marlin version (1.1.6) for my corexy. While compiling i get this strange bug, where the compiler asserts bad values.

basicly what i have configured:
```
X_MIN_POS 0
X_PROBE_OFFSET_FROM_EXTRUDER 28
LEFT_PROBE_BED_POSITION 15

MIN_PROBE_X (max(X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER, 0)) = 28

assert LEFT_PROBE_BED_POSITION >= X_MIN_POS

result = 15 >= 28 (true ???)
```

end compilation error:
```
SanityCheck.h:741: error: static assertion failed: LEFT_PROBE_BED_POSITION can't be reached by the Z probe.

       static_assert(LEFT_PROBE_BED_POSITION >= MIN_PROBE_X, "LEFT_PROBE_BED_POSITION can't be reached by the Z probe.");
```

full configuration:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1481380/Marlin.zip)

bjarchi, 2017-11-17:
Having gone through a similar configuration recently... 

I think you probably have the sign flipped on your probe offset - is your probe to the left or the right of your head? 

You specify the probe points in the same coordinate space as everything else - the offsets are applied automatically. In this case your configuration is saying that you have a minimum X position of 0, and your probe offset is +28 - so 28 mm to the right of your extruder. Then you're asking the machine to probe at x=15mm, but that would require moving the extruder to 15 - 28 = -13 mm. -13mm <= 0 so the sanity check fails. 

The static asserts can be a little tricky to follow, but the documentation in Configuration.h is quite helpful with respect to the coordinate space.

knobik, 2017-11-17:
You are right, probe is +28 to the right of the extruder. setting X_MIN_POS to -15 should fix the problem? i have some space between the real bed x 0 and the endstop.

bjarchi, 2017-11-17:
No problem, glad I could help. Should be closed as not a bug I guess :) 

For what it's worth either solution you suggested would work, but I've found the behavior to be a little unexpected when using negative _MIN_POS values without bed center at 0,0 so I think moving the probe position as you did is the better choice.

closed 2017-11-17 by knobik