MarlinFirmware/Marlin #8466: No example files

android444, 2017-11-17:
The files are missing in bugfix 1.1.x and bugfix 2.0.x.

Tannoo, 2017-11-17:
You mean these?

https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x/Marlin/example_configurations

https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/Marlin/src/config/examples

android444, 2017-11-17:
Where is RAMPS and more boards?

Tannoo, 2017-11-17:
Ramps is not a controller.

android444, 2017-11-17:
There was always RAMPS board.

Tannoo, 2017-11-17:
Not in examples. These are printer make/model example configurations.

Tannoo, 2017-11-17:
You select a supported controller board from the boards.h file.

Ramps is named there. Although, it's still not a controller, but that's what most ATMega setups use and are known as RAMPS.

Tannoo, 2017-11-17:
Mortherboard, rather..

Tannoo, 2017-11-17:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/boards.h

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/core/boards.h

android444, 2017-11-17:
I understand. thanks.

closed 2017-11-17 by android444