MarlinFirmware/Marlin #8467: REQ: Prevent axis damages

mikko70, 2017-11-17:
I hope new option or via Configuration.h

10mm axis movement disabled before you are homing first. It is too fast for moving, if you are wrong place.

code ultralcd.cpp (row about 3070: 
Before: 
`
   MENU_BACK(MSG_MOVE_AXIS);
	if ((axis_known_position[X_AXIS] && axis_known_position[Y_AXIS] && axis_known_position[Z_AXIS]))
        MENU_ITEM(submenu, MSG_MOVE_10MM, lcd_move_menu_10mm);
	MENU_ITEM(submenu, MSG_MOVE_1MM, lcd_move_menu_1mm);
    MENU_ITEM(submenu, MSG_MOVE_01MM, lcd_move_menu_01mm);
    END_MENU();
`

After:
`
   MENU_BACK(MSG_MOVE_AXIS);
	if ((axis_known_position[X_AXIS] && axis_known_position[Y_AXIS] && axis_known_position[Z_AXIS]))
        MENU_ITEM(submenu, MSG_MOVE_10MM, lcd_move_menu_10mm);
	MENU_ITEM(submenu, MSG_MOVE_1MM, lcd_move_menu_1mm);
    MENU_ITEM(submenu, MSG_MOVE_01MM, lcd_move_menu_01mm);
    END_MENU();
`

[ultralcd.zip](https://github.com/MarlinFirmware/Marlin/files/1481698/ultralcd.zip)

Tannoo, 2017-11-17:
`NO_MOTION_BEFORE_HOMING` won't work?

mikko70, 2017-11-17:
I think then no any motion?

Tannoo, 2017-11-17:
Correct.

mikko70, 2017-11-17:
I think this can hardcoded  somewhere prevent all user.

Tannoo, 2017-11-17:
In ` Configuration.h`:

//#define NO_MOTION_BEFORE_HOMING  // Inhibit movement until all axes have been homed

It's not enabled by default.

boelle, 2019-06-22:
wrong label, should be feature request (Hint: REQ)

closed 2019-07-21 by thinkyhead