MarlinFirmware/Marlin #8468: DISCUSSION: 2.0.x LPC1768 - Repetier HOST & specifying pins for M42, M43 & M226

Bob-the-Kuhn, 2017-11-17:
This is a continuation of a discussion in PR #8446.

Currently we do not have an elegant way of naming pin numbers for the LPC1768 that works both in Marlin and in Repetier Host.

The current system uses P1_21 within Marlin but that isn't valid in Repetier Host.  Entering  "M42 P1_21" in Repetier Host results in "M42 P0" being sent to Marlin.

The current work around is enter "M42 P1.21".  Unfortunately that doesn't work with P1_20.  "M42 P1.20" in Repetier Host results in "M42 P1.2" being sent to Marlin, the same as if "M42 P1.2"  were entered.

What do we want for a long term method/system?

NOTE - Pronterface will pass P1_2, P1.2, P1.20 & P1.02 as is.  I don't know what the other hosts do.

----

There is a small code change that could be done to Marlin that would result in:

- "M42 P1.02" results in "M42 P1.02" being sent & Marlin interpreting it as P1_2
- "M42 P1.20" results in "M42 P1.2" being sent & Marlin interpreting it as P1_20
- "M42 P1.2" results in "M42 P1.2" being sent & Marlin interpreting it as P1_20

PRO:
Could be done quickly.
Will probably work with all hosts.
Probably OK for a short term solution

CON:
Confusing for the user 

----

Another idea is to switch from starting the pin name with "P" when entering a command into the host.  For example entering "M42 R1_20" into Repetier Host will result in  "M42 R1_20" being sent to Marlin.

This gets messy with the M43 command which uses P, S & E as pin name parameters for various tests (Pxxx - pin, Sxx - start pin, Exx - end pin).

The semi-good news is that we have a lot of letters to choose from if we go away from "P".  Only E, F, G, P, S, T, X, Y & Z can't be used.  We'd need to pick three letters from the remaining ones.

PRO:
Could be done quickly
Works with Repetier Host and Pronterface

CON:
Confusing to the user.
Uncertain about other hosts

----

Bob-the-Kuhn, 2017-11-17:
@repetier - we'll need your input on this.

Bob-the-Kuhn, 2017-11-17:
Variation on the "small code change" idea

In addition to the code change, change the Marlin internal pin numbers from P1_2 to P1_02.  That would be a little less confusing to the user.

----

There is a small code change that could be done to Marlin that would result in:

- "M42 P1.02" results in "M42 P1.02" being sent & Marlin interpreting it as P1_02
- "M42 P1.20" results in "M42 P1.2" being sent & Marlin interpreting it as P1_20
- "M42 P1.2" results in "M42 P1.2" being sent & Marlin interpreting it as P1_20

PRO:
Could be done quickly.
Will probably work with all hosts.
Probably OK for a short term solution
Not as confusing for the user

CON:
Still confusing for the user

repetier, 2017-11-17:
Ok, I see P1_02 is not a number so that is not working on hosts interpreting gcode is no big wonder.
P and S are treated as integers in host/server so using 1.02 is also not working here. 

No idea why you did not use the simple solution and multiply by 100. So 1_02 becomes 102 and everything would work and M42 is still only using integers as before. 

Not sure why you want 1_20, 1_02 but I guess it is because 32 pins are aggregated into one 32 bit register? Makes also parsing easy since sub pin is p % 100 and bank is (int)(p/100).

thinkyhead, 2017-11-17:
I downvote 👎 both `1_20` and `1.20` formats for pin numbers to `M42` and `M43`. I still favor two separate arguments if we are going to be specifying pins by port and sub-pin. One way to do this would be to use `R` for the port and `P` for the pin. We could apply this to AVR also by making `R` refer to the port letter, where 1 refers to Port A, 2 is Port B, etc.

Another option is to use the form `M42.1 P20` where `42` is the main code, the `subcode` is picked up as `1` (and then used as the port number) and the usual numeric `P` value is used for the sub-pin. Again, this could work for AVR in the same manner as above.

Even though `M42` is rare and the speed of `M43` is negligible, I do think it's a smarter implementation to spend some SRAM on lookup table rather than looping, but of course this isn't a pressing issue, and it only matters for `M42`. On that note, we don't necessarily have to maintain the fast lookup table in the source code, which adds more tedious maintenance and opportunities for error. Instead, we would add an initialization routine that pre-scans the pin translation table as it now exists and generates an index for use within `M42` (and `M43`).

Roxy-3D, 2017-11-17:
> M42 P1.2" results in "M42 P1.2" being sent & Marlin interpreting it as P1_20

I don't understand why M42 P1.2 should be interpreted as Port 1 Pin 20.   Why can't we make it parse correctly to be Port 1 Pin 2 ?

Bob-the-Kuhn, 2017-11-18:
The "P1.2" vs. "P1.20 vs. "P1.02" is all because Repetier Host truncates trailing zeros.  The result is Marlin sees "P1.2" if either "P1.2" or "P1.20" is entered into Repetier Host.  

Because Marlin can see "P1.2" if "P1.20" is entered then we can't use "P1.2" to mean "port 1 pin 2". 

We need a way of uniquely identifying/differentiating between port 1, pin 2 and port 1, pin 20 when entering commands into the host..  There's a total of 7 LPC1768 pin pairs that have this issue. 

The methods/systems I'm aware of that will uniquely identify all pins are (port1, pins 2 & 20 used as examples):
- **P102 & P120** (as suggested by the Repetier folks) - I personally like this one the best because it can be used with all hosts, is easy for people to correlate with pin & port and the Marlin internal LPC1768 pin numbers can be made to match exactly so the pin numbers in the pins_YOUR_BOARD.h files are exactly the same as is entered into the host.  It'll cost some effort to modify the existing pinmapping files/routines and FASTIO but that's not a big effort.  There probably will be a small speed penalty for digitalRead, digitalWrite & analogWrite because it takes more than a shift and an AND operation to convert from human readable format to pin & port.
- **P1.02 & P1.20** - I like this one almost as much as **P102 & P120**.  The only downside is the internal pin naming system can't use decimal points so the pin numbers in the pins_YOUR_BOARD.h files are just similar to that entered into the host.  
- **Separate arguments** (as suggested by Scott - I like this one the least because it requires the most typing and requires slightly more thinking on the part of the user when entering commands into the host.  Definitely not major concerns.

Bob-the-Kuhn, 2017-11-18:
I'm starting to think I may be overly concerned about this.

I was just over to the Smoothie site.  I didn't find anything on commands that required inputting pin numbers.  I also saw in a config file that they do NOT use #define to associate 2.1 with alpha_step_pin.   

There's nothing on the RepRap g-code WIKI about pin numbers except for the Pxxx format.   It also indicates that M2, M43 & M226 are not supported by Smoothieware.

How about we just go with the **P1.02 & P1.20** system.  It's easy to understand and easy to implement and is as close to the ARM system as we're going to get in the near future.

It's not what Scott likes but I think it's the best compromise we have.

I'm going to put together a PR for that system so at least people can see exactly what I'm talking about.  If it gets shot down then at least we'll have a clearer path to the final solution.

repetier, 2017-11-18:
@Bob-the-Kuhn If you go that way please internally fill up the right side of x.y with 0 so it has 2 digits. That way you can overcome hosts knowing that 1.20 is 1.2 and 1.2 is just shorter and faster to send. You would then again fill up the removed 0 here and everything is also fine. Then it is as good as multiplying the value by 100.

tcm0116, 2017-11-18:
I like the P1.02 P1.20 system. For consistency, I think we should consider renaming the pin definitions in pinmapping.h and the corresponding pins files to use the same format (P%d_%02d).

Bob-the-Kuhn, 2017-11-18:
Yes, changing P1_2 to P1_02 is what I intend to do when implementing the P1.02 P1.20 system.

Please see PR #8484.  It implements the two digit pin name.

thinkyhead, 2017-11-24:
With latest changes, the integer format `M42 P120` is used to specify P1.20 on LPC176x. For pin ranges, `M43 S102 L208` specifies the range P1.02…P2.08. Etc.

closed 2018-05-20 by Bob-the-Kuhn