MarlinFirmware/Marlin #8469: Corner levelling on CR-10S goes outside bed bounds

ir-fuel, 2017-11-17:
On my CR-10S, when trying to level the corners the ones in the front are fine, but as soon as the bed slides all the way forward to do the back ones the nozzle is past the end of the bed.
[image]

Bed size is set to 300x300 in the configuration

config files attached
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1482251/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1482252/Configuration.txt)

AnHardt, 2017-11-17:
Don't worry. While **probing** all is well as long as the **probe** can touch the bed.

ir-fuel, 2017-11-17:
There is no probe. I am doing the "sheet of paper" way of measuring

ir-fuel, 2017-11-17:
Also, from what I've seen (ordered the parts) the probe will be on the right hand side of the nozzle, so if the nozzle goes out of bounds, so will the probe as it will be next to it. This picture is taken from the side of the printer.

ir-fuel, 2017-11-25:
I have no prove. I'm manually leveling

thinkyhead, 2017-11-26:
Are you using `bugfix-1.1.x`? Please test with that, and also post your `Configuration.h` file (rename with `.txt` extension to drop on github reply box).

For corner leveling the far Y position is `Y_MAX_BED - 10`, as set by…
```cpp
#define Y_CENTER ((Y_BED_SIZE) / 2)
#define Y_MAX_BED (Y_CENTER + (Y_BED_SIZE) / 2)
```
So corner leveling should be moving to `Y_BED_SIZE - 10`. What is your `Y_BED_SIZE` setting?

ir-fuel, 2017-11-26:
@thinkyhead  I am running the bugfix build and the files you asked for are attached to my first post ;)

thinkyhead, 2018-02-09:
Presuming this is solved.

closed 2018-02-09 by thinkyhead