MarlinFirmware/Marlin #8470: Babyz not persisted between prints?

ir-fuel, 2017-11-17:
Even if I leave my printer on, it seems that the babyz setting is back to 0.000 at the start of each print.

Is this normal?

AnHardt, 2017-11-17:
Yes.
The idea of babystepping was to just move some steps, bypassing all registration.

ir-fuel, 2017-11-17:
Ok. So the best thing to do is to get the value and put it in my slicer for the next prints then.

cdedwards, 2017-11-17:
or edit your height/z offset

KevMags, 2017-11-17:
Line #604 (in my Version) //#define BABYSTEP_ZPROBE_OFFSET // Enable to combine M851 and Babystepping.

If I understand this correctly your babystepping will be applied to your z offset permanently.

ir-fuel, 2017-11-17:
But it is not persisted.

KevMags, 2017-11-17:
I haven't used it but a M500 may save it to EEPROM, where it should persist.

thinkyhead, 2017-11-18:
In the current `bugfix-1.1.x` branch:
- Babystepping with `BABYSTEP_Z_OFFSET` will allow babystepping which also adjusts the Probe Z Offset. The same value as set by `M851`.
- If you use `M500` it will save the Probe Z Offset to EEPROM.
- You must re-probe the bed to benefit from the new Probe Z Offset. It is not applied to your previous bed leveling probe data.

ir-fuel, 2017-11-18:
I'm all pretty new to this, so I might need a bit more explaining, but:

I am currently running the bugfix branch on my CR-10S. How do I reprobe the bed so it uses the new probe Z offset? The only things I currently use is 9 point bed levelling and then using the baby z to finetune the distance (which for some reason is always too high after my bed levelling using a sheet of paper, but that's a different discussion) while printing.

How exactly would I persist this baby Z then? Right now I just copied the value in my slicer software in the Z offset field.

bjarchi, 2017-11-18:
@ir-fuel Hopefully I can help

With `BABYSTEP_Z_OFFSET` enabled, you would do as @thinkyhead said - adjust your offset using babysteps to get the correct height, use `M500` to save the offset to EEPROM, and then (re-)issue `G29`to (re-)probe the bed. 

Without `BABYSTEP_Z_OFFSET` enabled, you can still fine-tune your z-probe offset to avoid having to finetune each time. `M851` will print the current probe offset to the terminal; `M851 Zx.xx` allows you to change it. `M500` persists the new value to EEPROM when you're happy with it. 

AntCLabs has a pretty good document for BLTouch installation that includes z-offset calibration [here](https://docs.wixstatic.com/ugd/f5a1c8_20aa9a75b70d46d3849140fb4d6b6e31.pdf) - look for the section titled 'calibration.' One note - if you have `_SOFTWARE_ENDSTOPS` enabled and you need to move outside the z-bounds to calibrate your offset you *can* do this. You need to issue `M211 S0` to disable software endstops before making the necessary Z moves, and then `M211 S1` to re-enable the software endstops once you've found your correct z-offset value.

ir-fuel, 2017-11-20:
I did another test today.
It seems baby z isn't remembered at all?
Test: 
while printing, double click the button and adjust baby z
Click again to quit that screen
double click again
baby z value = 0.000

Is that normal?

Config files included.

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1487867/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1487868/Configuration.txt)

AnHardt, 2017-11-20:
> Yes.
The idea of babystepping was to just move some steps, bypassing all registration.

**All !**

If you are talking about `BABYSTEP_Z_OFFSET` - that could be different.

ir-fuel, 2017-11-20:
Also, is baby Z applied at the next command to be executed, or is it real time?
If for instance I send a command to extrude a line over the entire width of the bed, will baby Z influence that line while I change its value during that extrusion or will it only be applied once the next command starts?

thinkyhead, 2017-11-22:
> Also, is baby Z applied at the next command to be executed, or is it real time?

Babystepping is accumulated and applied immediately.

closed 2017-11-22 by thinkyhead