MarlinFirmware/Marlin #8474: pause during SD print has no pause position

ruggb, 2017-11-17:
I paused during an SD print and it just stops where it is at.
Is that a function of Repetier where I sliced it or Marlin?

stefi01, 2017-11-18:
that happens with me to, can ruin a print i know, i dont think the slicer matters much as is more firmware controlled but using cura or slic3r and then print from sd and pause during the print makes no diference, will stop in place, i think thats what advanced pause is for but i dont use it as it is incompatable with the feature that auto extrudes 5mm every 30 seconds when nozzle is hot, very anoying

thinkyhead, 2017-11-18:
Is this feature broken, @ruggb?
```cpp
/**
 * Advanced Pause
 * Experimental feature for filament change support and for parking the nozzle when paused.
 * Adds the GCode M600 for initiating filament change.
 * If PARK_HEAD_ON_PAUSE enabled, adds the GCode M125 to pause printing and park the nozzle.
 *
 * Requires an LCD display.
 * This feature is required for the default FILAMENT_RUNOUT_SCRIPT.
 */
#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define PAUSE_PARK_X_POS 3                  // X position of hotend
  #define PAUSE_PARK_Y_POS 3                  // Y position of hotend
  #define PAUSE_PARK_Z_ADD 10                 // Z addition of hotend (lift)
  #define PAUSE_PARK_XY_FEEDRATE 100          // X and Y axes feedrate in mm/s (also used for delta printers Z axis)
  #define PAUSE_PARK_Z_FEEDRATE 5             // Z axis feedrate in mm/s (not used for delta printers)
  #define PAUSE_PARK_RETRACT_FEEDRATE 60      // Initial retract feedrate in mm/s
  #define PAUSE_PARK_RETRACT_LENGTH 2         // Initial retract in mm
                                              // It is a short retract used immediately after print interrupt before move to filament exchange position
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE 10  // Unload filament feedrate in mm/s - filament unloading can be fast
  #define FILAMENT_CHANGE_UNLOAD_LENGTH 100   // Unload filament length from hotend in mm
                                              // Longer length for bowden printers to unload filament from whole bowden tube,
                                              // shorter length for printers without bowden to unload filament from extruder only,
                                              // 0 to disable unloading for manual unloading
  #define FILAMENT_CHANGE_LOAD_FEEDRATE 6     // Load filament feedrate in mm/s - filament loading into the bowden tube can be fast
  #define FILAMENT_CHANGE_LOAD_LENGTH 0       // Load filament length over hotend in mm
                                              // Longer length for bowden printers to fast load filament into whole bowden tube over the hotend,
                                              // Short or zero length for printers without bowden where loading is not used
  #define ADVANCED_PAUSE_EXTRUDE_FEEDRATE 3   // Extrude filament feedrate in mm/s - must be slower than load feedrate
  #define ADVANCED_PAUSE_EXTRUDE_LENGTH 50    // Extrude filament length in mm after filament is loaded over the hotend,
                                              // 0 to disable for manual extrusion
                                              // Filament can be extruded repeatedly from the filament exchange menu to fill the hotend,
                                              // or until outcoming filament color is not clear for filament color change
  #define PAUSE_PARK_NOZZLE_TIMEOUT 45        // Turn off nozzle if user doesn't change filament within this time limit in seconds
  #define FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS 5 // Number of alert beeps before printer goes quiet
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT       // Enable to have stepper motors hold position during filament change
                                              // even if it takes longer than DEFAULT_STEPPER_DEACTIVE_TIME.
  //#define PARK_HEAD_ON_PAUSE                // Go to filament change position on pause, return to print position on resume
  //#define HOME_BEFORE_FILAMENT_CHANGE       // Ensure homing has been completed prior to parking for filament change
#endif
```

thinkyhead, 2017-11-18:
```cpp
/**
 * Advanced Pause
 * Experimental feature for filament change support and for parking the nozzle when paused.
 * Adds the GCode M600 for initiating filament change.
 * If PARK_HEAD_ON_PAUSE enabled, adds the GCode M125 to pause printing and park the nozzle.
 *
 * Requires an LCD display.
 * This feature is required for the default FILAMENT_RUNOUT_SCRIPT.
 */
#define ADVANCED_PAUSE_FEATURE
. . .
```

thinkyhead, 2017-11-18:
> incompatable with the feature that auto extrudes 5mm every 30 seconds when nozzle is hot

@stefi01 Incompatible in the sense that we disallow the two features to be used together? It seems to me that we should allow both.

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed? Sorry, I got bored.

InsanityAutomation, 2019-06-22:
Actually id tag this as FR as during a pause with a purge setup at the end, we have flags to look at to tell autoextrude to stop. Maybe a 2.1 milestone here as well.

thinkyhead, 2019-07-21:
Seems to me that having more "onSomeEvent" G-code options will cover these kinds of cases, and that feature may be more useful to spend time on in the long run.

qwewer0, 2021-01-01:
I think this can be closed, as `NOZZLE_PARK_FEATURE` and `EVENT_GCODE_SD_ABORT` can be used to move the nozzle away on pause/abort.

sjasonsmith, 2021-01-01:
I agree.

closed 2021-01-01 by sjasonsmith