MarlinFirmware/Marlin #8475: Auto Home Procedure

yeochoony, 2017-11-18:
Hi.
I'm trying to use my SN04 inductive proximity sensor for Auto Calibration.
The problem occurs when the G28 runs, the Y axis goes too far from the sensor that when It runs Z home, It goes down Infinitely.
I want to make G28 procedure as these; X home -> Y home -> Y move 50 (set home offset) -> Z home -> Y home.
Thanks.

bjarchi, 2017-11-18:
I'm not sure I understand correctly, but if I do it seems like you're trying to use your inductive sensor for Z-homing as well as leveling? If this is correct you probably want to look at the define Z_SAFE_HOMING in configuration.h. This allows you to specify that X and Y home first, then move to a specific position on the bed before homing Z. This allows you to ensure that the probe is over the bed before trying to home the Z axis. 

Is that the behavior you're looking for?

thinkyhead, 2017-11-18:
I concur. `Z_SAFE_HOMING`, plus `G1 Y0` after your `G28`.

yeochoony, 2017-11-19:
@thinkyhead I tried to enable `Z_SAFE_HOMING`, but it doesn't returns to the original point (0,0,0) and this makes UBL get outside of the bed.

bjarchi, 2017-11-19:
@yeochoony Did you check/set all of the options for z_safe_homing appropriate to your machine? Here's a snippet:

```
#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axis (G28).
#endif
``` 

The defaults for the Z_SAFE_HOMING points are sane for most setups but not all - you want to make sure they correspond to a point well inside your bed edges (in case your probe offset is a little bit off). 

Also the first time I set this up I missed the initial `#define Z_SAFE_HOMING`, as it's commented by default at the end of a long block comment.

yeochoony, 2017-11-19:
@bjarchi Yup. I set the options like below because the real middle point is 40mm away from the board Y axis.
`#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2 + 40)    // Y point for Z homing when homing all axis (G28).
#endif`

The problem is that UBL doesn't calibrate the front side after Y point is moved to `(Y_BED_SIZE) / 2 + 40`.
Plus, the sensor goes outside of the bed which makes Z axis almost unlimited downward movement.

benlye, 2017-11-19:
Sounds like your Y endstop position is not configured correctly and you need to set `#define Y_MIN_POS` to reflect the nozzle position relative to the bed when stop is triggered.  Or to put it differently, where is the nozzle when you move to Y=0?

Either that, or your probe offsets are not correct.  

If you posted your configuration (attach the configuration.* files) along with some information about the physical setup of your printer (actual location of probe relative to nozzle) you would get better advice.

bjarchi, 2017-11-19:
Agree with @benlye - we need to see Configuration.h / Configuration_adv.h and have a more thorough description of the mechanical setup to really help with troubleshooting. It could also be that the UBL probe points aren't configured correctly, since it sounds like *maybe* safe homing is working correctly and the problem occurs when you start to probe the UBL mesh? 

I use z safe homing on my machine with no issues, so I think this is a configuration issue rather than a bug (which we can still help to diagnose). I'm bringing up UBL on my machine today to help with documenting it, so my setup will be pretty close to yours soon.

closed 2018-02-09 by thinkyhead