MarlinFirmware/Marlin #8476: Homing,M851 and G33 P1  bugfix-2.0.x

cdedwards, 2017-11-18:
the value of M851 is being subtracted from the correct delta height twice. run G33 P1, and the delta height goes from a correct value to a value which is Delta height - M851 * 2. My second delta is running a Re-Arm board and doesn't have a sensor. I built one which attaches to the hotend therefore the switch is lower than the nozzle by 21.31mm. My correct height is 207.35mm. M851 is set to 21.31 (yes I fixed the configuration.h z_max/min). After G33 P1, my new delta height is now 164.77. which is 2 x the value in M851.  I also get a homing failed error.

thinkyhead, 2017-11-18:
I'm favoring the removal of any modification to `delta_height` (and any other things) in response to `M851 Z`. I think this is an attempt to be too clever, and it only opens us up more issues of this kind. The very specific intent of the z probe offset is to determine the behavior of the probe. If people want to raise or lower all the values in the mesh, we can certainly add or augment a different G-code to achieve that.

thinkyhead, 2017-11-18:
The PRs #8456 #8458 now include a total removal of all cleverness tied to `M851` and `M290`.

thinkyhead, 2017-11-18:
Give `bugfix-2.0.x` another try and see if it works better now. I got the idea that you were doing an `M851` somewhere in your process. Be sure to use `M851 Z-21.31`. That minus sign is very important.

LVD-AC, 2017-11-18:
Z_probe_offset should be negative if the probe is below the nozzle.

`#define Z_PROBE_OFFSET_FROM_EXTRUDER 0.10  // Z offset: -below +above  [the nozzle]
`

closed 2017-11-19 by cdedwards