MarlinFirmware/Marlin #8480: TMC2130 not working sensorless mode

nightdemonx, 2017-11-18:
- Description: Step motors do not going to home and i think not save HOMING_SENSITIVITY, because if i send command M914 or M914 X19 Y19, terminal say only:
 0:21:56.098: X driver homing sensitivity set to 
 0:21:56.101: Y driver homing sensitivity set to 
For command M119 output always:
 0:33:46.060: x_min: open
 0:33:46.060: y_max: open

- Steps to reproduce:
Install TMC2130 step drivers, wires to SPI and diag1 to S endstop pin on RAMPS.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1484870/Configuration.zip)

teemuatlut, 2017-11-18:
Does any of the other TMC related commands give you an output?
Can you move an axis that is controlled by an TMC2130 driver?

nightdemonx, 2017-11-18:
Yes, i successfully printed some models with it. Working autocurrent control, setting steps all is ok.

teemuatlut, 2017-11-18:
Could you try my [development branch](https://github.com/teemuatlut/Marlin/tree/master)? I've changed the code related to this a while back but I haven't made a pull request to upstream for an even longer while. It's quite possible that this is just a cosmetic bug with the serial communication.

Btw, I've removed automatic current control in the form that it is here. It didn't work the way I wanted it to.

nightdemonx, 2017-11-19:
I tryed yours fork. I changed a minimum settings. Now i can change and save the value of sensivity, but sensors not trigered when i try stop the motor spindel. Terminal log and configuration in attach
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1485932/Configuration.zip)
:

> 14:22:10.655: start
14:22:10.656: echo:Marlin bugfix-1.1.x
14:22:10.656: echo: Last Updated: 2017-05-04 12:00 | Author: (none, default config)
14:22:10.657: Compiled: Nov 20 2017
14:22:10.657: echo: Free Memory: 5187 PlannerBufferBytes: 1232
14:22:10.658: echo:Hardcoded Default Settings Loaded
14:22:10.658: echo: G21 ; Units in mm
14:22:10.659: echo:Filament settings: Disabled
14:22:10.659: echo: M200 D3.00
14:22:10.660: echo: M200 D0
14:22:10.660: echo:Steps per unit:
14:22:10.661: echo: M92 X80.00 Y80.00 Z200.00 E500.00
14:22:10.661: echo:Maximum feedrates (units/s):
14:22:10.662: echo: M203 X300.00 Y300.00 Z5.00 E25.00
14:22:10.662: echo:Maximum Acceleration (units/s2):
14:22:10.663: echo: M201 X3000 Y3000 Z100 E10000
14:22:10.664: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
14:22:10.664: echo: M204 P3000.00 R3000.00 T3000.00
14:22:10.665: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
14:22:10.666: echo: M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
14:22:10.666: echo:Home offset:
14:22:10.667: echo: M206 X0.00 Y0.00 Z0.00
14:22:10.667: echo:PID settings:
14:22:10.668: echo: M301 P22.20 I1.08 D114.00
14:22:10.668: echo:Stepper driver current:
14:22:10.669: echo: M906 X 800 Y 800
14:22:10.669: echo:Sensorless homing threshold:
14:22:10.670: echo: M914 X19 Y19
14:22:10.881: FIRMWARE_NAME:Marlin bugfix-1.1.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
14:22:10.882: Cap:EEPROM:0
14:22:10.883: Cap:AUTOREPORT_TEMP:1
14:22:10.883: Cap:PROGRESS:0
14:22:10.884: Cap:PRINT_JOB:1
14:22:10.884: Cap:AUTOLEVEL:0
14:22:10.885: Cap:Z_PROBE:0
14:22:10.885: Cap:LEVELING_DATA:0
14:22:10.886: Cap:SOFTWARE_POWER:0
14:22:10.887: Cap:TOGGLE_LIGHTS:0
14:22:10.888: Cap:CASE_LIGHT_BRIGHTNESS:0
14:22:10.889: Cap:EMERGENCY_PARSER:0
14:22:10.892: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
14:22:36.958: X driver homing sensitivity set to 5
14:22:36.962: Y driver homing sensitivity set to 5
14:23:04.992: X:0.00 Y:0.00 Z:5.00 E:0.00 Count X:0 Y:0 Z:1000
14:24:35.212: X:0.00 Y:200.00 Z:5.00 E:0.00 Count X:0 Y:16000 Z:1000 (2)
14:24:57.350: X driver homing sensitivity set to 50
14:24:57.353: Y driver homing sensitivity set to 50
14:25:18.220: X:0.00 Y:200.00 Z:5.00 E:0.00 Count X:0 Y:16000 Z:1000
14:25:51.585: Reporting endstop status
14:25:51.585: x_min: open
14:25:51.586: y_max: open
14:25:51.586: z_min: open

teemuatlut, 2017-11-19:
Try going lower, the range is from -64 to 63 with _lower_ values meaning better sensitivity. So with a setting of 50, you'd likely not trigger at all.

nightdemonx, 2017-11-20:
When i try set -19 value of sensity, output M914 is

> 21:08:13.514 : X driver homing sensitivity set to 109
21:08:13.517 : Y driver homing sensitivity set to 109

M914 X-60 Y-60 is out:

> 21:09:40.066 : X driver homing sensitivity set to 68
21:09:40.068 : Y driver homing sensitivity set to 68

But still not trigered. Also i try you sample code "StallGuard" from Arduino, result is 1 - motor lock and not rotate, 0 - motor unlock, on terminal output only "0 0"

teemuatlut, 2017-11-22:
I'll try to find time to take a look at the example sketch. I think the best course of action is to get the StallGuard example working first so that we can verify the feature is working properly.

nightdemonx, 2017-11-22:
I trying to understand how its work )) 
In StallGuard example i think author forgot string with enable/disable STEP pin. I add this

> boolean toggle1 = 0;
  ISR(TIMER1_COMPA_vect){
  PORTF |= 1 << 0;
  PORTF &= ~(1 << 0);
  if (toggle1){
    digitalWrite(STEP_PIN, HIGH);
    toggle1 = 0;
  }
  else{
    digitalWrite(STEP_PIN, LOW);
    toggle1 = 1;
  }
}

and motor start rotate, also i set terminal speed to 115200, but sg_result() still put 0.

teemuatlut, 2017-11-22:
I'm the author. The step pin is driven with direct port manipulation from an interrupt routine.
`digitalWrite()` is really slow in comparison.

thinkyhead, 2017-11-24:
```cpp
byte toggle1 = LOW;
ISR(TIMER1_COMPA_vect) {
  SBI(PORTF, 0);
  CBI(PORTF, 0);
  WRITE(STEP_PIN, toggle1);
  toggle1 ^= HIGH;
}
```
😁

nightdemonx, 2017-11-26:
I tried exactly as you described. I pulled the Ramps out of the printer and connected only the engine. The engine does not rotate and remains locked, in the terminal only zeros (as I understand it means the maximum load). My drivers are made in china, but they have the original German chip. Maybe all-such is some kind of problem in my electronics. Ordered MKS Gen-L, I'll try how these drivers will work on it.
[image]

teemuatlut, 2017-11-26:
The Stallguide example sketch should output two values on every line. Like 0 342 for example. The zero is because if you want to use the Arduino IDE plotter, the zero value will prevent the IDE from adjusting the minimum scale.
It's possible that the china drivers are the problem as they'd be the only difference.
Btw, I quite like your cable setup and I really should re-do mine.

nightdemonx, 2017-12-03:
I compare scheme and saw that in my driver i need cfg4 pin to ground.
[image]
After that i can enabled HYBRID_THRESHOLD and it work, but stallguard still not working.

Bob-the-Kuhn, 2017-12-03:
I'm probably missing something obvious but ... please point me to the documentation on the SilentStick.

I just picked up some kits via DigiKey for only $8 each.  The headers need to be soldered on and you need to buy the heat sinks separately.

nightdemonx, 2017-12-03:
Documentation you can find on https://github.com/watterott/SilentStepStick

Bob-the-Kuhn, 2017-12-03:
Missing the obvious is my specialty!

Thanks

teemuatlut, 2017-12-03:
Do not solder the jumper bridge on the Watterott drivers. This will put them in standalone mode that disables SPI communication. I have no information on the clone drivers.

Bob-the-Kuhn, 2017-12-03:
I'd be REALLY surprised if DigiKey was selling something that wasn't 100% compatible.

I do know that the soldermask color is different.

I really won't know for sure until I get my printer back up & running.

teemuatlut, 2017-12-03:
Digikey sells the drivers that are made by Watterott. It should say "www.watterott.com" on the top side silkscreen.
TMC2130 has green solder mask.

Bob-the-Kuhn, 2017-12-03:
The silkscreen lettering is so small that it can't be read entirely.    The letters that I can make out fit properly with "www.watterott.com".

nightdemonx, 2017-12-04:
@teemuatlut, i mean other jumper bridge:
[image]
Blue is SPI, must be open, 
Red is CFG5, 
Green is CFG4 (i solder it)
@teemuatlut please saw in what condition this bridges on yours driver.

teemuatlut, 2017-12-04:
Mine are configured the same way and I believe they come that way from Watterott. There's no way I could solder an SMD resistor that small.

nightdemonx, 2017-12-04:
Little victory! I did that yours sketch from examples worked. Numbers in terminal change if i try to stup spindle. But if i try enable SENSORLESS_HOMING in your fork of Marlin, it still not working. This output from M122
`01:01:22.249 : X	Y
01:01:22.249 : Enabled		false	false
01:01:22.252 : Set current	800	800
01:01:22.254 : RMS current	795	795
01:01:22.254 : MAX current	1121	1121
01:01:22.254 : Run current	25/31	25/31
01:01:22.256 : Hold current	12/31	12/31
01:01:22.258 : CS actual		12/31	12/31
01:01:22.258 : PWM scale	37	1
01:01:22.258 : vsense		1=.18	1=.18
01:01:22.258 : stealthChop	true	true
01:01:22.261 : msteps		16	16
01:01:22.261 : tstep		1048575	1048575
01:01:22.263 : pwm
01:01:22.263 : threshold		0	0
01:01:22.263 : [mm/s]		-	-
01:01:22.263 : OT prewarn	false	false
01:01:22.265 : OT prewarn has
01:01:22.265 : been triggered	false	false
01:01:22.268 : off time		5	5
01:01:22.268 : blank time	24	24
01:01:22.268 : hysterisis
01:01:22.268 : -end		2	2
01:01:22.268 : -start		3	3
01:01:22.270 : Stallguard thrs	-19	-19
01:01:22.270 : DRVSTATUS	X	Y
01:01:22.270 : stallguard		X
01:01:22.273 : sg_result		0	0
01:01:22.273 : fsactive
01:01:22.273 : stst		X	X
01:01:22.273 : olb
01:01:22.275 : ola
01:01:22.275 : s2gb
01:01:22.275 : s2ga
01:01:22.275 : otpw
01:01:22.275 : ot
01:01:22.275 : Driver registers:
01:01:22.277 : X = 0x81:0C:00:00
01:01:22.278 : Y = 0x80:0C:00:00
`

govido, 2017-12-11:
Hi, could you give me some hint what you changed to make the chinese drivers work? I desoldered the spi 'bridge' but couldn't get stallguard to work. The stallguard demo just outputs 0 0 and the motor is completely fix in position. The spi config seems to work (drivers get recognized in Marlin). Stealthchop works super smooth. But I can't sensorless home.
I would appreciate any hints :)

nightdemonx, 2017-12-11:
@govido , Hi! Yes, you right, you must doing little trick: you must solder two bridges as i show on the picture.
[image]

ss89, 2017-12-25:
i would love to try that with mine, but neither my hands are good enough for that nor seems my soldering iron to be. Should there be anyone near or in hamburg, germany, please message me.
Other than that: i will stick with my watterott chips for the moment. :-)

frahlg, 2018-01-04:
Thanks for this. I have managed to solder my Fystec TMC2130 V1.0 drivers according to @nightdemonx figure above, and it seems to work! As an absolute beginner in soldering it was probably more luck getting it to work, really small!

oakdesign, 2018-01-04:
May just an explanaiton from reading the TMC2130-LA datasheet.
https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2130_datasheet.pdf
The CFG4 and 5 depend in their fucntionaliy on the SPI bridge. So if the driver is used in standalone mode SPI bridge closed. CFG4 sets the CFG4: SETS CHOPPER HYSTERESIS  and CFG5: SETS CHOPPER BLANK TIME in tristate so either open, GNC or VCC and are just the same as like on a TMC2100 driver.
If  SPI=1 so SPI bridge is open, the CFG bridges sets the function of dcStep and has to by tied to GND dcStep disable see Datasheet page 11.  From the Datasheet

18.3 dcStep with STEP/DIR Interface
The TMC2130 provides two ways to use dcStep when interfaced to an external motion controller. The
first way gives direct control of the dcStep step execution to the external motion controller, which
must react to motor overload and is allowed to override a blocked motor situation. The second way
assumes that the external motion controller cannot directly react to dcStep signals

So in my understanding we are using "the second" way are therefore the CFG4 bridge has to tied to GND which is already the case on watterott drivers but not on the Chinese ones

lednicazar, 2018-01-13:
So removing the bridgeof the SPI + making the bridge red & green in the image above let us the chinnese drivers working as it should??

In the wattertot ones in the red & green bridges, are bridges or resistors??

Any additional differences between wattertot and the Fystec?

simon-jouet, 2018-01-14:
Hi, 

I've followed @nightdemonx instructions/picture by removing the SPI bridge and bridging CFG4 for and CFG5 on chinese drivers.

stealthChop is working very well, the motors are super quiet (I actually can't believe the difference between TMC2130 and DRV8825) but the sensorless homing is still not working, the X axis rams into the end skip a lot of steps and after some time (I guess because of the soft endstop) does the same in Y.

@lednicazar I used solder bridges, but that might be the issues that i'm having.

The config I'm using is (I tried changing the sensitivity but without success)

```
  #define SENSORLESS_HOMING // TMC2130 only

  #if ENABLED(SENSORLESS_HOMING)
    #define X_HOMING_SENSITIVITY  8
    #define Y_HOMING_SENSITIVITY  8
    #define X_HOME_BUMP_MM 0
    #define Y_HOME_BUMP_MM 0
  #endif
```

nightdemonx, 2018-01-14:
> but the sensorless homing is still not working
@simon-jouet, do you solder diag1 to endstop signal pin? at the beginning I wrong with diag pin and sensorless mode also not work. So please check it to.

simon-jouet, 2018-01-14:
Thanks, 

Diag1 if I'm not mistaken is the centre-most pin (the one closest to the pot) and I connected both X and Y to the endstops. 

 Anything else to change in the configuration for this to work? The pullup or inverting? Would you be able to upload both configuration files please so I can double check?

Thanks!

lednicazar, 2018-01-14:
I try to check in the documentation but i didn't found any info about if it's a resistor or only a bridge.

@simon-jouet Here you can find the scheme: [https://community.ultimaker.com/topic/11200-new-sss-driver-tmc2130/?page=2](url)

If you achieve good result just tell me, I doubt about take the ones of wattertot or the fytec ones.

nightdemonx, 2018-01-14:
My config files:
[Marlin1.7.zip](https://github.com/MarlinFirmware/Marlin/files/1629731/Marlin1.7.zip)
[image]

lednicazar, 2018-01-14:
@nightdemonx Great! So the chineese should be compatible 100%
Additional question, did you install also the flyback diodes (protector), or without it? It's needed?

nightdemonx, 2018-01-14:
@lednicazar, yes, you right, chines driver it is 100% clones of watterott driver.
I not need in flyback diodes, because (from datasheet) "The diodes will have less effect when working with stealthChop due to lower times of diode conduction in the chopper cycle. At current levels below 1.2A coil current, the effect of the diodes is negligible. "

simon-jouet, 2018-01-14:
Okay so got it figured out and it's a bit weird. When I plug my motor + driver on 12V and do the test it's fine and stallguard works perfect. When I connect the proper 24V supply of my printer then stallguard doesn't work ... I might try to lower the current or increase the sensitivity (i'm down to 8 already) anyway glad it's somehow working, just unfortunate that it's not working with my main power supply :)

@lednicazar Yes remove the SPI bridge and do a solder bridge as show in @nightdemonx picture

simon-jouet, 2018-01-15:
Okay it's all working now, thanks @nightdemonx. I'm amazed by the silence of the printer, with my DRV8825 it was very noisy, and now I barely hear anything (well I hear a lot the 30mm fan)

So I increased the sensitivity to -30 just to see if stallguard was working then I decreased slowly until I found a good setting with my motors and a 24V PSU the good value is ~4. I was expecting for it to work but skip more steps with a lower sensitivity but at 8 it was never triggering at 24V.

Anyway, hope it's helpful to someone, and I highly recommend the TMC2130, I got the FYSETC ones from aliexpress did the solder points @nightdemonx suggested and it's good.

teemuatlut, 2018-01-17:
I'm writing documentation for the TMC drivers used with Marlin and I have a dedicated section for the FYSETC drivers. What should I write? If I refer to [this image](https://ae01.alicdn.com/kf/HTB1AZ6mhnnI8KJjSszgq6A8ApXaE/4pcs-Stepper-Motor-Driver-Module-TMC2130-Stepper-Motor-Driver-Module-With-Heat-Sink-3D-Printer-Part.jpg), the user should desolder the bridge near the chip corner, as well as bridge the two smaller pad selectors right from the sense resistor? Top bridges to right, bottom to the left.

nightdemonx, 2018-01-17:
@teemuatlut yes, you right! Also need connect 4 pins to SPI. People it do different ways, some solder bridges on RAMS, but most remove pins and solder it on the other side.

simon-jouet, 2018-01-17:
@teemuatlut might be worth noting that on the image you sent it's a solder bridge while on the ones I got it's a 0ohm resistor, not that it changes anything, just that it looks slightly different.

VIPer2322, 2018-01-20:
@teemuatlut

Hi, please i really need your help again.

I have the original Watterott TMC2130 running since the step loss fix from you is implemented since Marlin 1.1.7. Have now 1.1.8 installed.

But when i activate sensorless homing it moves just a few mm everytime when i press Home X or Home Y. or Home. (wiring changed too from the diag to the ensop pin)

I  changed the endstop inverting to false which lets the axis move in homing direction but just a few mm each time i press homing... (with true it does not move in the direction a all)

SPI is working fine, tried to change current or stealth chop and it accepts the changes.
It gives me alo the values back if i use M906.

Wiring is correct. The Sensorless homing feature worked for me in Marlin 1.1.6 but then it had step loss like in this https://github.com/MarlinFirmware/Marlin/issues/8478 which i commented and got great help from you :)
Since your development version and 1.1.7/8 it does not work anymore.

If i change the endstop wires from the X to the Y axis it starts homing put crashes into the endstop.

So i guess it must be something with the stop signal comming to early or so.

I tried already to raise the sensitivity  to 190 in 10 increment steps.
After that i disassempled the belts to watch just the motors, they do the same, Turning just a few degrees everytime when i push homing.....

I use a MKS Gen board if that helps. 

I will upload later also a video from the beavior.

I attach my config files for you.
Hope you have an idea why it's acting so strange. 

Thanks for your help in advance....

[changed config files.zip](https://github.com/MarlinFirmware/Marlin/files/1648655/changed.config.files.zip)

teemuatlut, 2018-01-20:
The sensitivity range is -64..63.
In standstill, what does M119 report to you?
Can you try the stallGuard example sketch to monitor the stallGuard values? You will need to figure out the correct pins as it is setup for standard ATMEGA. Remember to take your belts off because it will not stop on a hard limit!

VIPer2322, 2018-01-20:
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
filament: TRIGGERED

And additional for you.

It moves only in stealthchop in direction from the end stop, about 5mm every time i press home x

In Spreadcycle it stands still all the time

VIPer2322, 2018-01-20:
Video will be uploaded in about 18 min

VIPer2322, 2018-01-20:
Ill try to figure out how to use the StallGuard sketch. Never did something like that before.
Can try it on X (easy to disassemble the belts)

teemuatlut, 2018-01-20:
StallGuard, which is used for sensorless homing, only works when in spreadCycle mode. When you have stealthChop enabled, Marlin will switch to spreadCycle for the homing procedure but not for normal operation.

VIPer2322, 2018-01-20:
Sounds interesting, because the beavior is different. With Stealthchop activated it moves every time a bit and in Spread Cycle it stands still when i press X Homing. So there must be something different.

VIPer2322, 2018-01-20:
I configured now the pins, how do i use the Stall guard example?

VIPer2322, 2018-01-20:
https://youtu.be/r8NFXYjxaWw

here is the video for you :)

teemuatlut, 2018-01-20:
The example when uploaded will just start running the motor and reporting the stallGuard values back to the user.

Your video looks like a too low sensitivity setting. Which you already tried increasing.
Could you post your M122 output?

VIPer2322, 2018-01-20:
echo: External Reset
Marlin 1.1.8
echo: Last Updated: 2017-12-25 12:00 | Author: (none, default config)
echo:Compiled: Jan 20 2018
echo: Free Memory: 2747  PlannerBufferBytes: 1232
echo:V47 stored settings retrieved (614 bytes; crc 35769)
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X100.38 Y80.38 Z399.92 E412.50
echo:Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X600 Y700 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1500.00 R2500.00 T1500.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X10.00 Y12.00 Z0.40 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Auto Bed Leveling:
echo:  M420 S0 Z10.00
echo:Material heatup parameters:
echo:  M145 S0 H150 B60 F0
echo:  M145 S1 H220 B97 F0
echo:PID settings:
echo:  M301 P26.22 I2.69 D63.97
echo:  M304 P534.98 I95.48 D749.37
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.26
echo:Stepper driver current:
echo:  M906 X 800 Y 800
echo:Sensorless homing threshold:
echo:  M914 X0 Y0
echo:SD card ok
echo:SD card ok
>>> M122
SENDING:M122

VIPer2322, 2018-01-20:
looks like it does not handover the stall guard threshold.

is saw that there is in the stepper_indirection.cpp the following part missing:

#if ENABLED(SENSORLESS_HOMING)
        stepperX.sg_stall_value(X_HOMING_SENSITIVITY);
      #endif

same for Y is not in the file anymore. Maybe moved somewhere else...

VIPer2322, 2018-01-20:
Here is the video from the Stall Guard Example.
Value goes to 0 when i put force on the motor shaft.

https://youtu.be/LGAua5tCR1c

VIPer2322, 2018-01-20:
Tried now to set the sensitivity with M914.
Then it works but 19 is way to high.
Seems to be working with 4

So conclusion for me it that no value is given to the Board on init....

teemuatlut, 2018-01-20:
Your sgt measurement values are quite low compared to what I've typically seen. I'm usually getting somewhere around 300-700.

SGT values are set in `configuration_store.cpp` but they actually are not re-sent with M80 if the user has no 12V power on when booting.

VIPer2322, 2018-01-20:
I added now following lines which where present in Marlin 1.1.6
Marked with a * at the beginning of the lines.
Now it's working....

#define _TMC2130_INIT(ST, SPMM) tmc2130_init(stepper##ST, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

  void tmc2130_init() {
    #if ENABLED(X_IS_TMC2130)
      _TMC2130_INIT( X, planner.axis_steps_per_mm[X_AXIS]);
*      #if ENABLED(SENSORLESS_HOMING)
*        stepperX.sg_stall_value(X_HOMING_SENSITIVITY);
*      #endif
    #endif
    #if ENABLED(X2_IS_TMC2130)
      _TMC2130_INIT(X2, planner.axis_steps_per_mm[X_AXIS]);
    #endif
    #if ENABLED(Y_IS_TMC2130)
      _TMC2130_INIT( Y, planner.axis_steps_per_mm[Y_AXIS]);
*      #if ENABLED(SENSORLESS_HOMING)
*        stepperY.sg_stall_value(Y_HOMING_SENSITIVITY);
*      #endif
    #endif
    #if ENABLED(Y2_IS_TMC2130)
      _TMC2130_INIT(Y2, planner.axis_steps_per_mm[Y_AXIS]);
    #endif
    #if ENABLED(Z_IS_TMC2130)
      _TMC2130_INIT( Z, planner.axis_steps_per_mm[Z_AXIS]);

teemuatlut, 2018-01-20:
The effect of that is that on every boot you will override the stored EEPROM setting with the default value from configuration_adv.
I'm working an a push command to the library that will push the library held shadow registers to the driver. The problem is that if there is no high voltage power to the driver, the commands will not stick. However, the commands will always have effect on the shadow registers, so when we turn on power with M80 we also need to push the intended values to the driver.

VIPer2322, 2018-01-20:
Okay i got it. 
So the 24V Step UPs are making the mess, because they do not power on the high voltage side quick enough. But i think all the other values are transferred right?

Do You have any idea how to fix the low sgt measurement readings.
The area in which the Stall Guard sensitivity works for me is really tight (2-5)
guess thats because of those low readings.

Thanks man!

What could be the reason for those low sgt measurement readings?

teemuatlut, 2018-01-20:
My understanding is that stallGaurd can be very hardware dependent and different setups will have different thresholds. I'm just surprised that the gap is that big.
Things that affect the SG reading are your motors, current setting, motor voltage, resistance in the axis.  Possibly and probably even more. This is why a gcode command for live tuning the threshold value was needed as every machine can have a slightly different setting.

[2 more replies]