MarlinFirmware/Marlin #8482: MINTEMP printer halted please reset

MuhammadAlKadii, 2017-11-18:
every time i power on my printer the thermistors reads 14c and don't write the right room temp or the actually temp and when i load a file to print and try to begin heating the extruder it gives me error: MINTEMP
printer halted 
please reset.
i tried to wire the thermistors to the ramps many time but nothing changed

psavva, 2017-11-18:
It means you have the wrong thermistor selected in your config file.  Common thermistors are no.1 and no.5. try those first and see if your get the real room temperature.  If you know what thermistor you have, then just set it up in your config file, recompile and upload...

bjarchi, 2017-11-18:
In addition to what @psavva said - which is the first thing to fix - note that you want to have HEATER_*x*_MINTEMP (*x* is number for each hotend) and BED_MINTEMP set to values well below the minimum room temperature you ever expect to measure. Having MINTEMP trigger at 14º suggests it may be set too high. **But fix your thermistor definition first**

The foo_MINTEMP values are used to set the trigger point for the MINTEMP bailout, which attempts to detect a broken thermistor circuit. The only way the FW knows whether a temperature is 'too low' is by comparing against these values, which you need to set for your environment.

closed 2018-02-09 by thinkyhead