MarlinFirmware/Marlin #8492: [2.0] Changing extruder move Z while moving X

GMagician, 2017-11-18:
This is more a question but I really think that can be considered a bug...

When I change extruder from lcd menu I see a Z raise, extruder change (by a servo) and then when it is moving X to positionate new extruder it lowers Z in the same movement.

I think that correct sequence should be:
raise Z
move X/Y
lowers Z

am I wrong?
If recognized as a bug I can try to address it

GMagician, 2017-11-18:
There is also somethnig not clear....
tool_change.cpp
  #if ENABLED(SWITCHING_NOZZLE)
  [cut]
  z_raise = 0.3 + (z_diff > 0.0 ? z_diff : 0.0);
  [cut]
  #endif
[cut]
#if ENABLED(SWITCHING_NOZZLE)
     if (z_raise != z_diff) {
    [cut]

if is alwasy true, maybe some mem bytes may be saved

GMagician, 2017-11-19:
There is another strange "potential" issue in tool_change routine. When no home is done is set "no_move" to skip axes movements but when SWITCHING_NOZZLE is enabled then X&Y axis always move and this can be dangerous..only Z should move if no home (to avoid collision, even if this could be dangerous to if Z is too high)

GMagician, 2017-11-19:
Wow... and also
if (!no_move && IsRunning()) {
     #if ENABLED(DEBUG_LEVELING_FEATURE)
         if (DEBUGGING(LEVELING)) DEBUG_POS("Move back", destination);
     #endif
     prepare_move_to_destination();
 }

will move to a **very bad** position if nozzle have offset.. this will move too low

thinkyhead, 2017-11-19:
> I think that correct sequence should be:
raise Z
move X/Y
lowers Z

In fact, if we were to use any of the `do_blocking_move_to` functions, they are built to handle this exact issue.

thinkyhead, 2017-11-19:
> if is always true, maybe some mem bytes may be saved

Possibly… I'll look more closely to see. It's possible this logic was pertinent due to signs of arguments being different, and something changed without being accounted-for in the assignment of `z_raise`.

GMagician, 2017-11-19:
> The newly-selected nozzle must be able to move to the previous position. The newly-selected nozzle cannot be higher than the non-selected nozzle when this happens. When there are two nozzles, either they are at the same exact height, or the active one will be lower.

> 
Yes that ok but it is done with 2 movements. First move X, Y, Z to original destination (and Z position may be not good, then it does the correct move

GMagician, 2017-11-19:
> n fact, if we were to use any of the do_blocking_move_to functions, they are built to handle this exact issue.

> 
Code uses 'prepare_move_to_destination'

GMagician, 2017-11-19:
maybe instead of 'do_blocking_move' is better to call
  planner.buffer_line(destx, desty, curZ)
Code doesn't need to move to current

thinkyhead, 2017-11-19:
The `do_blocking_move_to*` functions simply require a sightly different setup where `current_position` is set properly before use. We wouldn't want to skip kinematics by calling the planner directly.

thinkyhead, 2017-11-19:
So…
```cpp

#if ENABLED(SWITCHING_NOZZLE)
  // Move to the "old position" and move back down (move the extruder into place)
  destination[Z_AXIS] += z_diff;
  feedrate_mm_s = planner.max_feedrate_mm_s[Z_AXIS];
  do_blocking_move_to(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS] + z_diff);
#else
  // Move to the "old position" (move the extruder into place)
  if (!no_move && IsRunning()) {
    #if ENABLED(DEBUG_LEVELING_FEATURE)
      if (DEBUGGING(LEVELING)) DEBUG_POS("Move back", destination);
    #endif
    do_blocking_move_to(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS]);
    prepare_move_to_destination();
  }
#endif
```

thinkyhead, 2017-11-20:
Looks like possibly all that is needed is this…
```diff
-          if (z_raise != z_diff) {
+          if (!no_move && z_raise != z_diff) {
```

GMagician, 2017-11-20:
Well as I said before z_raise is always != z_diff because it is 0,3 <=0 or 0,3+itself > 0.
no_move only is wrong because Z has to move to put new nozzle to same position of old one.
only X 6Y must stay in pos.
Since x&y are different move from Z then x&y may be under if(no_move) and Z always

GMagician, 2017-11-20:
> The do_blocking_move_to* functions simply require a sightly different setup where current_position is set properly before use. We wouldn't want to skip kinematics by calling the planner directly.

> 
planner si already directly called just before (when lifting Z)

GMagician, 2017-11-20:
I close this, eventually discussion will be posted on PR

closed 2017-11-20 by GMagician