MarlinFirmware/Marlin #8493: 1.1.6 babystepping with M851 (Z-offset) not working as expected

JupaCreations, 2017-11-18:
In the TEVO BLACK WIDOW facebook group we use a 1.1.0 RC8 V4 version.
This version was made/reworked by one of our members.
He made the option of "Live Z adjust" available in the BLT vesion.
This function works fine and when Z-offset is changed during, printing the value is stored as new Z-offset when you exit the function. 

Memebers are always looking for latest firmware so I took the honors to start and rework the 1.1.6 from the github here and started to change to adapt to the Black Widow settings.
In configuration_adv.h the correct babystep functions are activated.
Regular printing is fine.

What is happening?
When I have a Z-offset of -1.4 mm and start a print with a large brim I check the layer heigth.
1.4 mm is the correct layer heigth for a first print after a G28 and G29.
When I lower the heigth lets say to -1.5 mm and more I would expect as with the previous version I used that the layer heigth is reduced. It does but not in the same way/behavior. 
When I turn the knob I feel and hear the Z motor making large steps and it goes down slowly and not directly. But when I stop lowering it looks as if the heigth is returning to a previous setting and the layer is getting thicker again..... For a test I turned the Z-offset to for example -3.4 mm while printing the brim which should put the nozzle under the bed very quickly and make the filament flow blocked:-)
But it does not. As if the new Z-offset steps are not added or substracted and not RAM stored.
The value of Z-offset is stored when I exit the function in the EEprom.
I tried the same on an other machine with the same result, so it is not board related.

Do I do something wrong? Am I expecting wrongly that the Live Z adjust is behaving the same as the babystepping with M851
 
Thanks in advance, Paco

BTW I googled before placing this, but could not find anything related to my problem.

Tannoo, 2017-11-18:
Try bugfix-1.1.x

LVD-AC, 2017-11-20:
Maybe this is related to #8479

JupaCreations, 2017-11-21:
I tried the suggested bug fix version and adapted it to the cartasian Black Widow.
Did not changed anything else.
Enabled M851 and Z-offset function and it works the same as with the previous version.
I have to get some beta testers in the group to confirm my findings.
I let you know.

Paco

JupaCreations, 2017-11-21:
Wrong button clicked....

thinkyhead, 2017-11-24:
Background: We tried having `M851`, and the Z Probe Offset on the LCD work retroactively on previous mesh data under `BILINEAR` leveling, but this turned out —first— buggy, and in retrospect it opened up a can of worms that the code-base isn't quite ready for.

So for the moment the `bugfix-1.1.x` branch has been reverted so that adjustments to the Z probe Z offset are **not applied** to any previous calibrations, returning to pre-1.1.6 behavior.

Some of the attempted mesh adjustments, and so on, can be facilitated by changing the stored mesh Z values in terms of the nozzle position — putting off subtracting the probe Z offset until leveled moves are made. (Consolidation of leveling movement functions will help to optimize away most of the overhead.) Keeping the probe offset in the mesh data would allow babystepped adjustments to Z probe offset to be applied to mesh data at the same time.

This change, plus any others that purport to apply `M851` to previous probe data, will be on hold until the foundations are laid to facilitate the change.

JupaCreations, 2017-11-24:
Thanks for the explanation Thinkyhead altough I cant follow it. 
But that could be me lacking the technical side of it.
Maybe others can benefit from it running into the same probem.

Although the current version I have created seems to work in my situation, I will not endorse it to group members.
I appriciate all you guys are doing in regards putting time and effort in these versions.

closed 2017-11-24 by JupaCreations