MarlinFirmware/Marlin #8498: SDSUPPORT NOT WORKING MARLIN 2.0 MKS_SBASE_1.3 REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

maar1201gh, 2017-11-19:
[configurtion_marlin_2.0.zip](https://github.com/MarlinFirmware/Marlin/files/1485854/configurtion_marlin_2.0.zip)

In the meantime I send the gcode files through the computer

Thanks

Bob-the-Kuhn, 2017-11-19:
I'm afraid that you're stuck with sending the gcodes through the computer for the time being.  

Making the SD card on the LCD controller functional is still a work in progress.

Bob-the-Kuhn, 2017-11-20:
There is an ugly method of getting access to the LCD's SD card.  It requires making an octopus cable and using some non-standard options.  These non-standard options could get wiped out if you upgrade to a newer version of Marlin 2.0.x.

The octopus cable may not be so bad.  You could use the two cables that came with the RepRap Discount Full Graphics controller and a single [Adafruit 10-pin IDC Socket Rainbow Breakout Cable [1199] ](https://www.adafruit.com/product/1199).

I may not be able to get to the software this week.

Bob-the-Kuhn, 2017-11-20:
The software change turned out to be pretty simple.  Try this:
1. In the directory HAL_LPC1768 replace the file spi_pins.h with this one  [spi_pins.zip](https://github.com/MarlinFirmware/Marlin/files/1486620/spi_pins.zip)
2. Move the wire from EXP2-2 to J8-2
3. Move the wire from EXP2-1 to J8-3
4. Move the wire from EXP2-6 to J8-4

karabas, 2017-11-20:
Already used them for LCD2004 ( reprap discount smart controller) :((

Bob-the-Kuhn, 2017-11-20:
Any available pins will do.  You'll need to change spi_pins.h to match.

Pentelfrost, 2017-11-22:
So I'm using the MKS TFT/Wifi to send my gcode currently. Is there anyway to use the onboard SD for gcode? Using smoothieware the TFT would show the internal card and also an EXT dir which I assume was for the LCD but with Marlin I just get No SD card even though using M50X commands creates an eeprom.dat file on the internal SD.

Bob-the-Kuhn, 2017-11-23:
Marlin can not currently use the main board's on-board SD card for gcode files.  Right now only the LCD's SD card can be used for gcode files.  Unfortunately that isn't officially supported yet.

maar1201gh, 2017-11-28:
@Bob-the-Kuhn 
I did the changes to EXP2 cable and SD CARD Is working now.
Thank you.

vladimirRUS007, 2017-12-01:
Hello, an error occurs when the installation of Marlin 2.0 mks sbase. Board appears in device Manager as" usb device marlin " My steps for installation.

I downloaded the Marlin 2.0
Set Atom
Downloaded PIO
Changed the #define MOTHERBOARD BOARD_MKS_SBASE
IN the lower left corner clicked on PIO Bulid (дсз1768)
Downloaded firmware.bin from .pioenvs/lpc1768 7) zacamil on the stick
connect the charge mks sbase and it appears as :" marlin usb device "
What are your thoughts...
[image]

lkvapil, 2017-12-02:
@Bob-the-Kuhn 

> Move the wire from EXP2-2 to J8-2
Move the wire from EXP2-1 to J8-3
Move the wire from EXP2-6 to J8-4

Do you mean the pins on Sbase labeled as EXP2  and pins like on the image on url ?

 [https://www.google.cz/search?q=pin+header+numbering&prmd=isvn&source=lnms&tbm=isch&sa=X&ved=0ahUKEwidtfeNnezXAhVF6KQKHSnyBIMQ_AUIESgB&biw=360&bih=564#imgrc=VTT_65kYYvdQMM:](url)

Thanks

Bob-the-Kuhn, 2017-12-03:
Yes, the 2x5 header labeled EXP2 on the board.  NOT the one labeled AUX1 on the board..

[image]
[image]

Bob-the-Kuhn, 2017-12-03:
@vladimirRUS007 

Looks like you need to install the correct driver for the virtual disk. See the [Smoothieware drivers page](http://smoothieware.org/windows-drivers) for instructions.

Here's what my Re-ARM card looks like.

[image]

lkvapil, 2017-12-03:
So if the schematic is correct, so it won´t work again.. (and I changed the spi_pins.zip too)

[image]

Bob-the-Kuhn, 2017-12-03:
Don't tie the pins together.  You'll need a custom cable.  In that custom cable you'll need to move the pin/wire that used to go to P0.8 to J8-3, P0.7 to J8-2 and P0.5 to J8-4

MKS & their %^*@##$% backwards LCD connectors!!!

Please see my updated/corrected comment above.

Spawn32, 2017-12-03:
@Bob-the-Kuhn Does a LOL fit here ? 😄

Bob-the-Kuhn, 2017-12-03:
Not much else we can do.  The MKS %^$* pain is here to stay which means constantly having to re-explain the problem & the solution.

Spawn32, 2017-12-03:
Maybe we could add a readme in the SBASE examples directory ?

Bob-the-Kuhn, 2017-12-03:
Great idea!  

I'll add it in a bit.

Spawn32, 2017-12-03:
yes, would probably be the right place to add things like this for all board's and shields...
Would probably offload the ISSUE queue a lot :)

karabas, 2017-12-04:
[image]
This is the mod for LCD2004 non graphic display  rrd smart controller

lkvapil, 2017-12-04:
@Bob-the-Kuhn  Will give a try the new idea..
I had to get working letters to turn 180° the connectors on one side of connectors

Bob-the-Kuhn, 2017-12-04:
@karabas -yes, the RepRap Smart Controller uses those pins.

You can use any free pins for the SD card.  You'll just need to modify the spi_pins.h file accordingly.

lkvapil, 2017-12-04:
Many thanks to everyone, who gave me some advice ! ! !! 
Now I can see :D print from SD card and it works ! ! 
So first of all to get working the LCD to show the letters I had to cut the stumps on 1 side of connectors EXP1 & EXP2 and turn the connectors behind the LCD 180°..
So on MKS Sbase are connectors slipped right direction and on LCD 180°..
Then, as said @Bob-the-Kuhn here : 

> The corrected pin numbers on the MKS Sbase board are:
> 2. Move the wire from EXP2-9 to J8-2
> 3. Move the wire from EXP2-10 to J8-3
> 4. Move the wire from EXP2-5 to J8-4I

I picked up some jumpers and connected wires from EXP2 connector as described above..
And connected obviously the other wires to EXP2 connector on the Sbase...
Last step, I changed the spi_pins.h as uploaded above and rebuilt..

Many thanks to everyone ! !

closed 2017-12-05 by thinkyhead

karabas, 2017-12-15:
On Sbase both cddetects of onboard sdcard and display sdcard join together on one pin!!!!!
So if you change sdcard in display cddetect will NOT do any action. You can check it trying to display several sdcards with different content.

teddybee, 2017-12-15:
Could somebody tell me what is the difference between the reprap full graphics lcd and the Anet A6 12864 LCD pinout?  [https://ae01.alicdn.com/kf/HTB1NqVoQFXXXXXiXpXXq6xXFXXXj/Anet-12864-LCD-Smart-Display-Screen-Controller-Module-with-Cable-3D-Printer-Accessory-for-Reprap-3D.jpg](url)
I don't find so much code for Anet LCD on mks sbase's pins.h so I turned on the reprap lcd in config file. I rotated the lcd's plugs 180° but still just the backlight turned on and keeps beepeing...

lkvapil, 2018-12-26:
Hi,

I moved to SBASE again and now I´m again troubleshooting connection with RepRap Discount Full Graphics controller. 

Lukas