MarlinFirmware/Marlin #8500: Wrong position displayed on repetier...is it normal?

GMagician, 2017-11-19:
Is it normal that after a home sequence LCD quotes are not replicated on repetier ?

eg after home lcd show 195 (safe home is active) and repetier 0...
If X is commanded from repetier (eg move forward 50) X move backwards to 50

sns5400, 2017-11-20:
I have the same problem.
Firmware is last version from marlin repetier as desktop interface on linux and windows.
Afther home Z on controller display is z=5 mm on desktop screen z is 0.
In monitor screen it receive x=0; y=0 z=5

lrpirlet, 2017-11-20:
You need to synchronize Repetier with Marlin... Do issue any Gcode from Repetier that tell Marlin to position the printer...

I use something like (from memory) in my startup code ``` G1 X1 Y1 Z1 ``` to position the hot-end 1 millimeter away from home on each axis...

AnHardt, 2017-11-20:
`M114 - Report current position` used to do the trick.
At least G28 and G29 used to send the position report automatically (all automated moves should do that).

GMagician, 2017-11-20:
Yes I see on repetier log window current position reported correctly. Then or format is not correct or message is unused

AnHardt, 2017-11-20:
Ty
```
void report_current_position() {
  SERIAL_PROTOCOLPGM("X: ");
  SERIAL_PROTOCOL(current_position[X_AXIS]);
  SERIAL_PROTOCOLPGM(" Y: ");
  SERIAL_PROTOCOL(current_position[Y_AXIS]);
  SERIAL_PROTOCOLPGM(" Z: ");
  SERIAL_PROTOCOL(current_position[Z_AXIS]);
  SERIAL_PROTOCOLPGM(" E: ");
  SERIAL_PROTOCOL(current_position[E_AXIS]);
```
instead of (spaces behind the ':')
```
void report_current_position() {
  SERIAL_PROTOCOLPGM("X:");
  SERIAL_PROTOCOL(current_position[X_AXIS]);
  SERIAL_PROTOCOLPGM(" Y:");
  SERIAL_PROTOCOL(current_position[Y_AXIS]);
  SERIAL_PROTOCOLPGM(" Z:");
  SERIAL_PROTOCOL(current_position[Z_AXIS]);
  SERIAL_PROTOCOLPGM(" E:");
  SERIAL_PROTOCOL(current_position[E_AXIS]);
```

GMagician, 2017-11-20:
I'll try this evening

sns5400, 2017-11-20:
Hello,
Thank you for the quick reply but the result is not ok
First i found the code in Main.cpp
I replace and recompile and upload to my geetech bord.
I do the same action and give you a screenshot before and afther
homing.
Problem is the right up coordinates are not synchronised with the info
from the harware bord.
My display on the geetech bord is in sync.
P. Schippers
Belgium

GMagician, 2017-11-20:
Confirm doesn't work

AnHardt, 2017-11-20:
Updating RHs display used a undocumented feature.
May have changed from RH 1.6.2 to 2.0.5

GMagician, 2017-11-20:
I checked repetier source code and it should parse correctly the message:

>             h = extract(res, "X:");
            if (h != null)
            {
                level = 3;
                float.TryParse(h, NumberStyles.Float, GCode.format, out x);
                analyzer.x = x;

>

sns5400, 2017-11-21:
Hello,
Can you confirm you find this in "Repetier-Host version V2.05 for linux
 and or for Windows
is there a difference for x; X; y; Y; z; Z
gr

GMagician, 2017-11-21:
This is present in latest version on github. It check using uppercase but not sure if before it convert all received text in uppercase

AnHardt, 2017-11-21:
As far as i know the latest **open** version is mainly untouched since 4 years. Thats before 1.0

GMagician, 2017-11-21:
lol thats a good news

AnHardt, 2017-11-21:
It's closed source since then. :-(

thinkyhead, 2017-11-24:
Sounds like a host issue. Are you talking about the current Marlin (1.1.6 or `bugfix-1.1.x`)… and seeing the same issue with Repetier on older Marlin (e.g., 1.1.2) too?

GMagician, 2017-11-24:
Marlin 2.0 and host 2.05

sns5400, 2017-11-24:
Hello,
I use Marlin 1.1.5 firmware in Geetech bord and Repetier host desktop
V2.0.5 on Linux Mint and Windows 10
P. Schippers

GMagician, 2017-11-25:
Ok since marlin sends frame with position and format seems to be compatible with old host version.
It's a host bug.

closed 2017-11-25 by GMagician

thinkyhead, 2017-11-26:
Perhaps @repetier can comment on whether this is a bug or a feature.

repetier, 2017-11-26:
In the past host had used xyz responses to get position after homing. Now the response has changed to include/exclude G92 offsets, so we can not trust these information any more since the interpretation depends on marlin version. So now you have to set homing coordinate in host printer config to see them correctly and host relies only on positions it has send it self. During print these coordinates are also useless since there is a divergence between send and interpreted command so host would not know for which part the response is.

Maybe I will ad back interpretation after homing since that also resets G92 making both position responses identical.

pflannery, 2018-01-03:
I also see this in Marlin 1.1.8 and rep host 2.05

thinkyhead, 2018-01-22:
> host had used xyz responses to get position after homing. the response has changed to include/exclude G92 offsets

@repetier 

Any `G92` offsets that may have been set are always cleared by `G28`.

Marlin 1.1 reports the current logical position just as Marlin 1.0 did. So I'm not sure what difference  in output you are seeing.

If you send `G92 X10` then the current X position becomes 10, and `M114` will report X as 10. That is the same in Marlin 1.0 as it is in Marlin 1.1.

repetier, 2018-03-01:
I have reactivated the position parsing for next release. Not sure any more what the problem was. Host will interpret M114 including offsets and add offsets on it's own. So as long as user uses host to set G92 everything will be fine. After all it seems to cause more problems then without it, because users seem never to set homing coordinates in host making it assume wrong wrong coordinates.

thinkyhead, 2018-03-01:
Would it help to have `G92` with no arguments return the current workspace offsets? And, `M206` with no arguments return the current `home_offset`? If these two are subtracted from the position reported by `M114` then you have the "native" position. An extension –additional argument– to `M114` would also be possible (and simpler) returning the native position.

repetier, 2018-03-01:
What is the difference between M206 and G92? As I understand M206 is just offsetting origin so user has origin where wanted, something hosts assume anyway. So I think that would not be needed. Never used this.

G92 returning offsets might be nice, but host will not send G92 after M114. The problem arises mainly if user can change offsets in lcd. The hosts will never know. In repetier-firmware changing offsets returns something like:
12:01:54.149 : X_OFFSET:0.000 Y_OFFSET:0.000 Z_OFFSET:0.000
So as long as firmware returns offsets upon change host could always be up to date.
Sending offsets with no parameter would a good thing in initialize host with correct offsets in case firmware does not reset on connect like due native port.
Alternatively it might be a solution to also display offsets in position reports. Not really sure which would be better. I mean position is only copied when X: Y: or Z: is found in response so if at that time the right offsets are available be it by sending them before or in same line makes not much difference. Only thing is if you change offset and then hosts sends position it would be wrong, so from this point knowing about offset changes should always send a message back.

But that is a just a part of the lcd problem. In general every change that hosts display should emit a notification message. Hosts will not poll for changes so they assume settings set in their interface unless firmware makes a message. Fans are the same. Here repetier returns
12:03:20.783 : Fanspeed0:67
so hosts could know it has changed.

Drawback is here latency and command buffering.
Assume
...moves
M106 S255
1 move
M106 S100

Jost will assume 255 then 100 and then firmware reports 255 setting it 255 and then it sees 100 and goes back to it. Gives some jitter in display, but at the end it will match and normally it is no problem at all.

For temperatures it is more relaxed. Here the current output contains set temperatures anyway so updates are transferred.

This is of course not important for every tiny variable, more the main state visualized in hosts.
- Temperature - already done
- Fans
- Offsets
- Flowrate (FlowMultiply:97)
- feedrate multiplier (SpeedMultiply:110)
- Debug Level: (DebugLevel:7)
- Coordinates if changed by lcd, otherwise it would just put a lot of traffic especially if host is sending the moves and assumes these positions anyway. In fact hosts should only accept coordinates returned if they are not sending anything because the buffering and time of receiving coordinates will not be in sync.

These are the important states normally shown in hosts, and without feedback from firmware the host may show wrong values if changed over lcd.

thinkyhead, 2018-03-01:
I'm about to crash, but I'll address this one:

> The problem arises mainly if user can change offsets in lcd. The hosts will never know.

- It's the firmware's responsibility to let the host know if the position has been changed, even if from LCD. So if that's missing from Marlin it's something we need to fill in.
- Note that Marlin sends the `M114` report in response to `G92` in the current firmware. You can still parse the `G92` going out, but for Marlin it's now redundant.

Ok, more later. Zzzzz…