MarlinFirmware/Marlin #8507: [FR] Multiple Beds Support

dragondgold, 2017-11-20:
### Feature Request

Is it any plan to support multiple heat beds? In bug printers the heat bed is usually split into two or more heaters and thermistors. Being able to have 2, 3 or 4 beds would be great in this cases, right now you have to set the second bed as a second extruder.

psavva, 2017-11-30:
What hardware are you using?  Mega with a Ramps shield?

dragondgold, 2017-11-30:
@psavva Yes, I activate each zone with a GPIO connected to a transistor with a relay.

psavva, 2017-12-06:
I've checked the magnitude of changes needed to support multiple bed support (similar to multi extruder support in Marlin.  It's huge.

I suggest you stick to what you're doing now until someone (maybe myself) finds time to implement this...

thinkyhead, 2018-02-10:
Duplicate of #7710

closed 2018-02-10 by thinkyhead

piolo79, 2019-05-20:
hi, I was also interested in adding a second bed but I had thought of using the pins of the second extruder (E1) since the stepper driver I use as second z, would it be possible?