MarlinFirmware/Marlin #8508: SERVO2_PIN not behaving as expected

Bergerac56, 2017-11-20:
I use the "servo" pins to pilot RVB LEDs. 
If everything is fine for SERVO1 and SERVO3 pins, the behavior of SERVO2_PIN seems not to be right on RE-ARM when PWM is needed.
When issuing a M42 P1.19 S0, the LED connected to it is OFF.
When issuing a M42 P1.19 S1 to S255, the LED is ON (full). PWM  seems not to work on this pin. 

Is it a bug? Or a specificity of the RE_ARM ?

Bob-the-Kuhn, 2017-11-20:
Confirmed - SERVO2 doesn't work on the latest Bugfix-2.0.x.

It won't be until tonight before I have enough time to see what's wrong.

Bob-the-Kuhn, 2017-11-22:
I haven't forgotten about you.  Dr. & Thanksgiving taking a lot of time.

I can see that the PWM ISR run about 10 times and then stop after issuing the Servo2 command.  

I'm going to make the bold assumption that you want it to run a little longer.

Bob-the-Kuhn, 2017-11-22:
Found the problem.  I forgot to re-enable the PWM ISR when it couldn't find a requested pin.

The fix has been merged via PR #8518.

Bergerac56, 2017-11-22:
@Bob-the-Kuhn - Once again... well done. It works nicely now. Thanks a lot.

BTW, I see that the syntax of the M42 gcode changed: no dot anymore in the syntax of the pin: P1.19 or P1_19 seems to be now P119. I suppose that this is the conclusion of the discussion about pin numbering?

In Marlin, it is still today P1_19 (for ex.). Will that change?

Bob-the-Kuhn, 2017-11-22:
The internal syntax is P1_02.  M42 is P102.  

As far as I know that's the way it'll stay.

If you have a customized pins_xxx.h file then you'll need to convert P1_2 to P1_02.

Bergerac56, 2017-11-22:
Nice Thanks.

closed 2017-11-22 by Bergerac56