MarlinFirmware/Marlin #8511: bug? bugfix-1.1x and bugfix-2.0.x G30, G29, G33 probing

cdedwards, 2017-11-21:
so far I've been having no luck with G33, G29 or G30 probing. On my one delta the probe is above the nozzle and currently any probing operation fails. G33 P1, gives me a homing failure along with a delta height of 0.0, G29 just bounces in place at 0,0 with a probe failure and G30 V2, provides a probe failure, lifts off the bed and halts. everything I've looked at points to a probe failure, however the probe does work. M119 will show it triggered at the appropriate time.
My other delta with a probe underneath the nozzle seems to work fine.

[probe_failure.txt](https://github.com/MarlinFirmware/Marlin/files/1489833/probe_failure.txt)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1489836/Marlin.zip)

cdedwards, 2017-11-21:
ok. so using Z_MIN_PROBE_ENDSTOP on a delta is a really bad idea when it defaults to Z_MAX_ENDSTOP. 
Maybe adding a quick sanity check for this for delta printers only would make sense.  On a delta, we will never use Z_MAX_ENDSTOP for the probe pin.

closed 2017-11-21 by cdedwards

thinkyhead, 2017-11-25:
Ok, I've added a sanity check warning and made sure the probe pin setting is `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` in all delta configs.