MarlinFirmware/Marlin #8513: [1.1.6] Billinear bed leveling not precise correcting

Lostish, 2017-11-21:
CR-10S

Bed warped 0.4 as most
Solid bed mounts to get the bed as level as possible
Billinear 7dots, stretching outside of bed
Capacative sensor mounted, with tolerance of 0.002
Cura 3.0.4 Slicer
First layer at 25mm/s

As i have mounted a ABL sensor and activated Billiniear bed leveling hoping i could print my whole bed.
Tested a lot of various things but to no avail. I have about a grid of 100x60mm with good squish.

So i have printed 1 layer test print on the bed, and i have held a finger or 2 on the Z rod, and i actually fel it moves as the nozzle goes around the bed.
Leveled bed at the middle with a superb squish, a 20x20 test cube prints amazingly.
But it is not precise enough. At the edges and corners it is off by 0.1 in that ball park making the corners/edges first layer not good, with gaps and non stick. And if i level the bed to the edges/corners i loose that 90x90 middle bed as a print as it squishes that much so the extruder skips.

So i feel the correction in the rod, but the correction not that precise to fix a pretty even bed.
I dont know how to read the debug data, so feel free to gide me with commands to provide the neccesary debug data. 
```

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4      5      6
Recv:  0 -0.373 -0.250 -0.245 -0.185 -0.187 -0.058 -0.085
Recv:  1 -0.375 -0.297 -0.240 -0.190 -0.137 -0.035 -0.060
Recv:  2 -0.340 -0.252 -0.225 -0.137 -0.130 +0.012 -0.032
Recv:  3 -0.315 -0.232 -0.165 -0.110 -0.055 +0.045 +0.005
Recv:  4 -0.260 -0.167 -0.142 -0.053 -0.045 +0.085 +0.028
Recv:  5 -0.230 -0.162 -0.117 -0.063 -0.025 +0.063 +0.010
Recv:  6 -0.217 -0.123 -0.112 -0.025 -0.028 +0.102 +0.035

Recv: Recv: X:293.00 Y:278.00 Z:3.31 E:0.00 Count X:23440 Y:22240 Z:1340
```

```
Send: M48 Y150 X150 P10
Recv: M48 Z-Probe Repeatability Test

Recv: Finished!
Recv: Mean: -0.059250 Min: -0.063 Max: -0.058 Range: 0.005
Recv: Standard Deviation: 0.001953

Recv: X:213.00 Y:162.00 Z:0.00 E:0.00 Count X:17040 Y:12960 Z:1340
```

```
Send: M420
Recv: echo:Bed Leveling On
Recv: echo:Fade Height Off
```

```
/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 95 }

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_FEEDRATE          { 500, 500, 50, 100 }

/**
 * Default Max Acceleration (change/s) change = mm/s
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_ACCELERATION      { 500, 500, 100, 500 }

/**
 * Default Acceleration (change/s) change = mm/s
 * Override with M204
 *
 *   M204 P    Acceleration
 *   M204 R    Retract Acceleration
 *   M204 T    Travel Acceleration
 */
#define DEFAULT_ACCELERATION          500    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  500    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   500    // X, Y, Z acceleration for travel (non printing) moves

/**
 * Default Jerk (mm/s)
 * Override with M205 X Y Z E
 *
 * "Jerk" specifies the minimum speed change that requires acceleration.
 * When changing speed and direction, if the difference is less than the
 * value set here, it may happen instantaneously.
 */
#define DEFAULT_XJERK                  8.0
#define DEFAULT_YJERK                  8.0
#define DEFAULT_ZJERK                  2.0
#define DEFAULT_EJERK                  3.0
```

Tannoo, 2017-11-21:
All bugfixes for 1.1.x are done in `bugfix-1.1.x`. Try that branch to see if it's still an issue.

Lord-Quake, 2017-11-21:
An issue I posted back in July and still have as I write this: https://github.com/MarlinFirmware/Marlin/issues/7278

Lostish, 2017-11-21:
Thats my issue to, any luck minimizing it?

[image]

Roxy-3D, 2017-11-21:
> An issue I posted back in July and still have as I write this: #7278

Right now...   I'm trying to get the G26-Mesh-Validation-Pattern tool working for Bi-Linear.   If I can get that going...   It will be much easier for people to get their mesh's fine tuned and correct.

Tannoo, 2017-11-21:
Have ya'll tried UBL?

bjarchi, 2017-11-21:
Is it possible this is due to the inherent sensitivity of inductive probes to the density of material they're detecting? The pattern seems right for edge effects - remember that the probe is sensing a volume/area below the probe head, not just the point underneath it. If you probe close enough to an edge (and especially corner) the probe will be seeing a combination of the bed and air. 

This wouldn't show up in the probe repeatability report, necessarily, as the probe could be triggering precisely but not accurately near the edges. It would show up as increasing variance between the actual distance and 'probed' distance as the probe gets closer to the edges / corners.

Reaction-Time, 2017-11-22:
Similar issue here with BLtouch, like in #7278

fweds, 2017-11-22:
Same issue for me with capacitive sensor

Roxy-3D, 2017-11-22:
The capacitive and inductive sensors are the worst for exhibiting the 'Edge' effect.   But even with a micro-switch, there are edge effects on a piece of glass.    You can't see the change in flatness.   But you can certainly measure it.

The point I'm trying to make is that was a big part of the motivation to get the G26 Mesh Validation Tool written and working.   The machine's Z-Probe does a good job of getting you initial numbers.   But those numbers can only be trusted so far.   What really matters is how the printer works with mesh numbers.   And that is what the G26 pattern exposes.

I'm a few days away from having a version of G26 working with the Bi-Linear bed leveling.    If there is interest...   I'll post a link here and we can work through the QA and Testing phase together on it.

fweds, 2017-11-22:
Ok but I've try without glass, same issue :/ Then I don't think this is the glass cause this ;)
I've try Linear and Bilinear and correction is wrong :(

Sorry for my poor English, I'm French

Reaction-Time, 2017-11-22:
Wouldn't your point about edge effect be negated if the glass was rotated 90 degrees and the problem still persisted? I say that because I can rotate my glass piece anyway I want and one side is always higher than the other. I'm still new to this all, so if I'm really off target just ignore me, but thought I'd add to the discussion.

Roxy-3D, 2017-11-22:
I only brought up glass having edge issues because I wanted to point out it isn't just the inductive (and capacitive) probes that are seeing funny stuff at the edge.

Yes...   If rotating the glass doesn't affect anything, it may not be an 'edge effect'.    In a few days...   Let's try G26 and see what the pattern's look like.    And if we use that to edit the mesh points, my guess is you will get a correct first layer across the entire print bed.

Reaction-Time, 2017-11-22:
LCD needed for G26? I assume yes, correct?

Roxy-3D, 2017-11-22:
> LCD needed for G26? I assume yes, correct?

It is best to have an LCD panel because that makes it easier to gracefully abort the operation if the nozzle starts getting to close to the surface of the bed.   And you lose the ability to interactively prime the nozzle.   But it should be helpful to resolve this.

Roxy-3D, 2017-11-22:
If one of you with the Bi-Linear problems cares to try it...  I think I have G26 setup so it will help us see what is going on.    Please cross your settings over to the default Configuration.h file in the \Marlin directory.   (The UBL_G26_MESH_VALIDATION has moved out of the UBL area.)

https://github.com/Roxy-3D/Make_G26_Work_with-Bi-Linear

Please post a picture of the pattern so we can discuss what is happening....

Lostish, 2017-11-22:
I will head over to #8522 

Im in a small print as i write this. But during tomorrow evening i will sure load that firmware test.
If i understand, i just have to run a G26 with UBL inistead of bilinear?

bjarchi, 2017-11-22:
> The capacitive and inductive sensors are the worst for exhibiting the 'Edge' effect. But even with a micro-switch, there are edge effects on a piece of glass. You can't see the change in flatness. But you can certainly measure it.

My point about edge effects is somewhat negated by the fact that people see this with mech/bltouch probes, but what I was describing is different from changes in flatness of the bed - that is something that should be correctly measured and corrected by bed leveling. 

Edge effects WRT capacitative or inductive probes is meant in the physics (E&M) sense - the E/B field lines distort as you approach the edge of a real conductor/plate, which may well throw off the sensor readings. The inductive and capacitative probes were never really meant for this application, and their datasheets describe how triggering distance changes with things like target thickness and etc. If you're probing sufficiently close to the edge of, say, 2mm thick Al then what the sensor 'sees' will be closer to 1mm thick Al - and the trigger distance will change accordingly. 

I don't think it's the root cause of this issue anymore, having read through more reports, but it is something to keep in mind for people using inductive and capacitative probes.

Lostish, 2017-11-22:
I would like to post one thought about this.
I have a capacative sensor and it has a ide offtset of  -63 And if ~8cm probe from an edge can make that kind of difference.
Then from my point of view the same is happening in the middle of the bed aswell.
But that is only for my leveling and i still have that z not correcting eaven with 8cm from an edge.
But if that is the thing, maby ill try a grid of 200x200 and extrapolate for the edges?

Ill help in trying out the G26 as best at i can, brobably an half an hour to an hour a day tops.
Lets make this work as my enthusiasm has gone to bottom and leaving the printer just sit there because of all the work put in it.

Roxy-3D, 2017-11-22:
> I will head over to #8522
> Im in a small print as i write this. But during tomorrow evening i will sure load that firmware test.
If i understand, i just have to run a G26 with UBL instead of bilinear?

You should be able to use the G26 tool with Bi-Linear when we get it going.   It should be close to working.  But I haven't brought up Bi-Linear to test it.

Roxy-3D, 2017-11-22:
@Lostish   When we get to see a G26 pattern...   It may be very clear what is going on.   For example...  If your bed frame has some cut out areas to reduce mass and make it easier for the steppers to push the bed back and forth.  We may see a pattern on the G26 print that shows one thing where there is a lot of bed material...  And another thing where there is very less bed material.    

We shall see....   But either way...  Using the G26 Pattern we should be able to adjust the mesh points so you get good adhesion across the entire bed.

Lostish, 2017-11-23:
I can say the pattern were...
Quite something.
I had a good squish, im printing it as we speak with little less Z offset as the nossle rammed the bed, as i forgot Safe Z homeing.... On the new firmware upload.

I have changed my Bilinear to maby a center grid of 230x230 just to have a square mesh as the probe has a negative x offset by -63

G29 says,

Recv:       0      1      2      3      4
Recv:  0 -0.110 -0.177 -0.220 -0.195 -0.265
Recv:  1 -0.053 -0.120 -0.115 -0.127 -0.210
Recv:  2 +0.015 -0.032 -0.063 -0.063 -0.157
Recv:  3 +0.037 -0.023 -0.020 -0.040 -0.143
Recv:  4 +0.067 -0.005 -0.053 -0.048 -0.150

G26
This is the pattern:
[image]
[image]
[image]

Lostish, 2017-11-23:
So After releveling, well after the heatend crashed in to the bed.
I Leveled the Y axis to be leveled with the bed as possible.
And it is center leveled z offset wise.

This is the result of a tuned print wich is perfect squish middle.

Better leveling then this... dont think i can.

Recv: Bilinear Leveling Grid: ( Extrapolate: On )
Recv:       0      1      2      3      4
Recv:  0 -0.255 -0.252 -0.223 -0.152 -0.157
Recv:  1 -0.217 -0.207 -0.130 -0.080 -0.103
Recv:  2 -0.152 -0.123 -0.077 -0.025 -0.055
Recv:  3 -0.120 -0.110 -0.037 +0.007 -0.037
Recv:  4 -0.085 -0.085 -0.055 +0.000 -0.037

G26,

[image]
[image]
[image]

Tannoo, 2017-11-23:
Why not use UBL?

bjarchi, 2017-11-23:
> Why not use UBL?

Seconded - I recently brought up UBL on my machine after having used bilinear (no LCD - I have a doc update pending that covers this case) and the UBL system really makes it much easier to get a good leveling mesh across the entire bed. 

Looking at that pattern, one thing I would ask is whether the bed was up to temperature *and stable* before you homed and then probed the mesh - and at the same temp that you printed the pattern at? I say stable because I've found that it takes some time for the whole bed to get up to the temp read by the thermistor, and depending on bed material and thickness there can be considerable dimensional changes as it does so.

It won't matter how accurate the probing and computations are if the bed surface is moving around (due to thermal expansion) after you've probed your mesh, or during probing. 

With a bed of that size and no enclosure I'd also look into whether stray air currents or etc. might be affecting things - anything that changes the way heat flows from the heater to the bed top will change the temperature profile throughout the bed.

Roxy-3D, 2017-11-23:
> Why not use UBL?

Yeah...   That is a good question!

In the very last picture....   The Top Left corner and the bottom Left corner look like you have no adhesion.  The nozzle was too far from the bed in those corners, right?   (You don't want to make the mistake of assumming the nozzle was too far when in reality it was pressing so hard against the bed that no plastic was able to come out!!!!)   I would lower those mesh points by .1mm or .15mm and try again.    (And see if the corners start to print OK.)

Lostish, 2017-11-23:
Yes you are correct, the nozzle is shifting away from the bed getting the the layer not to stick.
So yes that leaves me to calibrate those 2 corners.

And to finetune corners like that ( Software, as i have solid bed mounts ), i need to switch to UBL. Right?
And is do someone have a good link to a vera basic UBL mesh correcting tune to start with?

So what you are telling tath there is a lot of math needed to calculte averything for bilinear and linear to have it 100% working, and there is other features that have higher priority ( Or is bilinear as good as it gets )
Just want to know.  To ease my mind. Because i have put tremendus of hours getting the bed level.

Roxy-3D, 2017-11-23:
> Yes you are correct, the nozzle is shifting away from the bed getting the the layer not to stick.
So yes that leaves me to calibrate those 2 corners.

OK...  Good to know.  Those white lines on your bed make it harder to see the white lines from the filament.     

> And to finetune corners like that ( Software, as i have solid bed mounts ), i need to switch to UBL. Right?   And is do someone have a good link to a vera basic UBL mesh correcting tune to start with?

No.   You do not need to switch to UBL.   You can change your mesh points in any of the mesh bed leveling systems with an M421 command.    (It is easier in UBL because you just move the nozzle to the location you want to change, and give it a G29 P4 R and it starts letting you edit the mesh points close to where the nozzle started.)   With Bi-Linear, you will need to enter the coordinates of the mesh points you want to edit and their new value.     But after you do that to the corners...  You should see 'better' results.   And then you can start working on getting all the mesh points dialed in exactly so you get perfect lines.

> So what you are telling tath there is a lot of math needed to calculte averything for bilinear and linear to have it 100% working, and there is other features that have higher priority ( Or is bilinear as good as it gets )

There is a lot of math happening...   But Bi-Linear and UBL will both produce very good results IF (and only IF) you have the mesh accurately mapped to the print bed.     The thing about UBL is the interactive mesh editing tools make it easier to get to that point (but at a cost of 35 KB of program space).

> Just want to know. To ease my mind. Because i have put tremendus of hours getting the bed level.

If you want to stick with Bi-Linear, that is fine.  We can get that tuned perfectly now that we can see the G26 Mesh Validation Pattern.    It looks to me like a number of your mesh points need to be edited.  But the corners are the worst...  So just to learn how to edit a mesh point and to see that it is working, I suggested we just edit those two areas first.    Then once those corners are working correctly...  We can get the whole mesh dialed in.

thinkyhead, 2017-11-24:
An updated `G26` Mesh Validation Tool will be merged soon. After it is, please test out and report if there are any oddities. Bilinear leveling needs an additional patch so it will subdivide its moves more than it currently does, which will help it to follow bed contours better. I will put a PR together for that soon.

Lostish, 2017-11-24:
Lets say i stick with bilinear. There is no way to save the mesh for re use?

And i have switched over to UBL now, little unfamilliar with that routine.
Bat that makes me save the mash when i dialed that in.

I need to read about editing mesh points as thats a weak area for me.
Any tips and hints as like the M421 command that help me gettin the bed to a total "flat" surface ?

Roxy-3D, 2017-11-24:
You should be able to save the mesh with Bi-Linear in the EEPROM.   You would just do an M500 to do that....

You can find good step by step instructions of what to do to bring that up here:   https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

If you are using UBL...  Don't do a M421.    Just move the nozzle to the area of the mesh you want to edit and do a G29 P4 R       And then adjust the mesh point up or down the amount you want on the LCD display.   Press & Hold the encoder wheel to terminate editing that area of the mesh.     Move to the next area, repeat...     And then save the mesh when you are done.

Lostish, 2017-11-25:
The mes thing with 10 dots takes ages...
However, when i have done a first G29 P1 and later do G29 P4 R...

It starts at the -0.08 or what it would be but i have to turn the knob to around -1.0 to get to the 0.2 first layer 

And when i printed G26 i got a nozzle running in to the bed and dragging it.
Trying to fine tune but i dont really know what im doing.
Im trying G29 P6 C command but.. as said, no idea.

Roxy-3D, 2017-11-25:
OK...   It sounds like you should back up a step.   Turn off all bed leveling.   And get your Z_OFFSET_FROM_EXTRUDER set correctly so you can print a small calibration cube at the center of the bed.

If you can print a small object without bed leveling, everything else should go smoother...

Also, consider turning on Z-BabyStepping and the Double Click for Z-BabyStepping.   That is very helpful when it comes time to print.

Lostish, 2017-11-25:
``Send: M420
Recv: echo:Bed Leveling On
Recv: echo:Fade Height 10.00``

Fade Height ?

And yes, i can print a 20x20 test cube with perfect squish ( first layer ) at center where G28 probes.

``Send: G29 W
Recv: Unified Bed Leveling System v1.01 active.
Recv: Mesh 0 Loaded.
Recv: UBL object count: 1
Recv: planner.z_fade_height : 10.0000
Recv: # of samples: 49
Recv: Mean Mesh Height: -0.981123
Recv: Standard Deviation: 0.118111
Recv: zprobe_zoffset: -0.4500000
Recv: MESH_MIN_X  ((((((300) / 2) - (300) / 2) + 15)>(0 + -63)?((((300) / 2) - (300) / 2) + 15):(0 + -63)))=15
Recv: MESH_MIN_Y  ((((((300) / 2) - (300) / 2) + 15)>(0 + -12)?((((300) / 2) - (300) / 2) + 15):(0 + -12)))=15
Recv: MESH_MAX_X  ((((((300) / 2) + (300) / 2) - (15))<(300 + 20 + -63)?((((300) / 2) + (300) / 2) - (15)):(300 + 20 + -63)))=257
Recv: MESH_MAX_Y  ((((((300) / 2) + (300) / 2) - (15))<(300 + -12)?((((300) / 2) + (300) / 2) - (15)):(300 + -12)))=285Recv: GRID_MAX_POINTS_X  7
Recv: GRID_MAX_POINTS_Y  7
Recv: MESH_X_DIST  40.33Recv: MESH_Y_DIST  45.00
Recv: X-Axis Mesh Points at: 15.000  55.333  95.667  136.000  176.333  216.667  257.000
Recv: Y-Axis Mesh Points at: 15.000  60.000  105.000  150.000  195.000  240.000  285.000
Recv: Kill pin on :41  state:1
Recv: Recv: ubl_state_at_invocation :0
Recv: Recv: ubl_state_recursion_chk :0
Recv: 
Recv: Meshes go from 0x02D0 to 0x0F7FRecv: sizeof(ubl) :  1
Recv: Recv: z_value[][] size: 196
Recv: Recv: EEPROM free for UBL: 0x0CAFRecv: EEPROM can hold 16 meshes.
Recv: 
Recv: Unified Bed Leveling sanity checks passed.Recv: ok
``

Roxy-3D, 2017-11-25:
Fade height of 10.0 mm is fine...   That won't cause you any problems.    
Can you post the Topology map you are using?   

It looks like you saved your mesh in slot number 0.  Is that correct?

When you post the G29 T   (Topology map)   can you also indicate where the nozzle was on the bed when it started to dig in?

Lostish, 2017-11-25:
Slot 0, as the default when saving.

As you can see in the pictures this is the first layer, end the right most i aborted just before it completed...
As you can se at X20 Y 70, thats my prime nozzle running. And it started digging as it hits the bed.
So in a G26 it started in the bed printing the mesh and ran for 4 points "in the bed" before i turnd it off.
This has ruined my print mat, but i gor 2 more, so thats fine :)

And when babystepping to 0.8 This is the layer i got.
And without bed leveling the middle cube is pretty much the result after leveling center.

[image]
[image]

```

Recv: Bed Topography Report:
Recv: 
Recv: (0,6)                                           (6,6)
Recv: (15,285)                                        (257,285)
Recv:  -1.000   -0.975   -0.925   -0.920   -0.895   -0.900  [-0.945]
Recv: 
Recv:  -1.005   -0.965   -0.850   -0.840   -0.870   -0.915   -0.970
Recv: 
Recv:  -1.000   -0.960   -0.760   -0.820   -0.755   -0.870   -0.935
Recv: 
Recv:  -1.060   -0.995   -0.775   -0.815   -0.810   -0.845   -0.945
Recv: 
Recv:  -1.175   -1.055   -0.945   -0.975   -0.940   -0.970   -1.055
Recv: 
Recv:  -1.200   -1.095   -1.065   -1.025   -1.060   -1.110   -1.125
Recv: 
Recv:  -1.200   -1.165   -1.065   -1.085   -1.120   -1.120   -1.205
Recv: (15,15)                                            (257,15)
Recv: (0,0)                                             (6,0)
```

Roxy-3D, 2017-11-25:
Your bed is very flat.    That isn't going to cause any problems.   But ideally...  You should have 1/2 of your mesh points above 0.00 mm and half below 0.00 mm.   You don't have any mesh points that are positive at all.    And where you are homing (in the center of the bed) it is claiming the bed is .8 mm. below the 0.00 mm point on the Z-Axis.

Can you please double check your homing?
- Do a G28
- Move the nozzle back to the center of the bed if it isn't there.
- carefully lower it until the nozzle just touches the bed.
If this number is not close to 0.00 mm that is going to cause problems.   (-.8 is not close to zero!)   I'm guessing we need to fix your Z_PROBE_OFFSET_FROM_EXTRUDER and get that set better.   

And if so, we will want to re-probe the mesh.   But this time, before we do another G26, what we can do is use PronterFace (or Repetier Host) to put the nozzle .5 mm above the bed, and go all the way across the bed in mulitiple places in both the X and Y direction.  If the clearance doesn't disappear...   We can try the G26 with more confidence.

Lostish, 2017-11-25:
I have a offset value of -0.45 To have some space with an unleveled bed. Now when i ran a G29 P1 with the capacative sensor with no offset it runs in the bed by top left, so i need a bit of an offset for the probing to work actually.

So what im doing, is i pick a nicke height, around 0.5 mm above the bed and i set the sensor to activate on that "nice for my bed" level. Adjusting Z to G92 Z1 and backing down, subtracting 1 from that and thats my offset value. And then im putting that in to my z offset witch was -0.45 in the previous prints

Roxy-3D, 2017-11-25:
OK...   I understand.    I actually do the same thing.   I home and have the nozzle about .25 mm too high.  I use Z-BabyStepping to lower the nozzle while it is printing the skirt.    I do this because I have a BL-Touch and I don't fully trust it.

BUT...  With that said...   It is important that the mesh reflect reality.    Lets do this:
- Change your clearance numbers so you have plenty of space
--  #define Z_CLEARANCE_DEPLOY_PROBE 
-- #define Z_CLEARANCE_BETWEEN_PROBES
- Set the Z_PROBE_OFFSET_NUMBER correctly.  Probably  that will be in the -1.2 mm range.
- Re-probe the bed and save the mesh.   (G29 P1,  G29 A, G29 S0 )
- Verify the home location has a value close to 0.00 mm  
- Re-Home the machine (G28)
- Load and activate the mesh (G29 L0    G29 A)
- Using PronterFace or Repetier Host to lower the nozzle to .5mm above the bed.
- Move around, from side to side in both directions and make sure you have clearance.  You should see the Z-Axis correcting as you move the nozzle.
- Maybe...  If you are paranoid...  Lower the nozzle to .25mm above the bed and repeat the clearance test.
- Do a G26.      Do not use the 'C' parameter.  The nozzle will only expand out from the starting location so fast...    You should have plenty of time to Stop the print with a Press and Hold of the encoder wheel or a reset if we have miscalculated.
- If the G26 finishes...  Please post the picture.   We will enter an edit cycle and get everything perfect.
- Once we have everything perfect...   We can adjust your Z_PROBE_OFFSET_FROM_EXTRUDER to give you the extra clearance you want.   (Like I say...   I do this also just because I don't fully trust my Z-Probe)

Lostish, 2017-11-25:
I will leave that for another day.
Its 4:20 Am here.

I really appritiate all the effort you put in to this. Thanks, really.

And by that mesh...  What you said..

> One line from the status report is suggestive of a problem:
> Recv: Mean Mesh Height: -0.981123

I did a G29 P4 R On that and adjusted, Babystepped each point down to 0.2 with a steel blade mesure that has a blas of 0.2mm thick.
So that is wat that number came from.

I will redo all the dots above another day.
And when i do G29 P4 R, what is the procedure of that?
As i said, it stepped up from each point babystep displayed on screen, babystepped the height of the NOZZLE that was on to 0.2 mm with a measurement tool. Was that correct ?

Roxy-3D, 2017-11-25:
Part of the problem is adding clearance by using an incorrect Z-Probe-Offset.   The bed leveling system is trying to use the information acquired by probing the bed so it can track the bed as it moves around.  You can get extra clearance if we do it at the end of the process.   But by doing it before we generate the mesh, the bed leveling system is going to 'correct' that error out.

> And when i do G29 P4 R, what is the procedure of that?

Just follow the steps given at https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md    Those are going to work in a reasonable way for you.      

G29 P4 R lets you edit an area of the mesh that doesn't quite look right.  That can happen with inductive and capacitive probes.   If it looks like a line is too squished (not enough clearance between the nozzle and bed as it prints) you can raise that mesh point .1 mm.    If you are getting no adheasion because the nozzle is too high above the bed, you can lower the mesh point (or the nozzle) at that location by .05 or .1 mm.    When the area has been edited...   You press and hold the encoder wheel and the editing for that part of the mesh stops.

Baby Stepping is outside of Marlin's motion control (and bed leveling).    It is very nice to have when you are starting a print.   But right now, you should not be messing with it.   It will introduce errors to the numbers we are trying to acquire to build the mesh.    It is as if you manually twisted the shaft on one of your stepper motors.  Marlin doesn't know you did it.   It will only cause problems if you do Baby Stepping while trying to get the mesh set up.

Lostish, 2017-11-25:
Ive started from the beginning

* G28
* G29 P1
* G26 ( Came out ok, as previous prints edges and corners ar off by 0.1 - 0.2
* G26 P4 T ( Here is whats interesting )

This was what i was trying to tell yestarday. G29 P4 T reads clearence between probes of Z 1 before the mesh.
So lovering Z to 0 and initiating it rises my Z to 1 before initiateing Mesh Editor. Wich means i need to mesh edit down that Z 1 to Z -1 in the mesh editor to get to Z 0
As i want a slight quicker probes i have theese settings

```
#define Z_CLEARANCE_DEPLOY_PROBE   3 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES 1 // Z Clearance between probe points
```

I think am almost there now

Roxy-3D, 2017-11-25:
> So lovering Z to 0 and initiating G29 P4 T/R rises my Z to 1 before initiating Mesh Editor. Wich means i need to mesh edit down that rise of Z 1 to Z -1 in the mesh editor to get to Z 0. 

The nozzle's height during the G29 P4 R should not be used in any way.   You should not lower the nozzle until it touches the bed.   If you do that, you will be getting mesh points that are much too low.    

We should turn off    #define UBL_MESH_EDIT_MOVES_Z     so there is no confusion.

Instead...   When you look at the G26 pattern, you should be able to see "in this area I need the nozzle a little bit lower".   And when you edit the mesh points in that area, in the center you might lower the mesh point .1 mm and on the surrounding mesh points maybe lower them by .05 mm.    

Remember, the reason you are editing the mesh is because it does not correctly map the bed.   Because of this, the nozzle needs to be above the bed so it can move around and not gouge into the bed.   After the initial probing and filling of the unprobed mesh points, most likely no mesh points will need to be moved by more than .15 or .2 mm.    Certainly not by a full 1.00 mm.

> As i browse the menues, Prepare --) UBL Menu --) Step By step --) Validate Mesh --) Valitate PLA
It just spits out filament endlessly where the nozzle currently is not moving and Printing, User priming nossle or something like that in the display. ( Just cosmetic as i use octoprint and run commands throug there)

I typically use G26 P C O2.5       The Prime needs to be told it is done with an encoder wheel click.    Alternatively, you can specify the length of the prime with something like G26 P 5.0 C O2.5

Lostish, 2017-11-26:
Thank you Roxy for all the clarification in using mesh bed leveling and tuning it.
Very helpful and i will take the time neccesary to level the bed with mesh leveling where mesh points needed to level. 3D printing is a steep learning curve and i learn stuff every day.
Thank you for all the effort you put in to getting me to understand the leveling system.

As a feature request, instead of guessing at each and every point that needs to be leveled.
What do you think about doing like the G28 process ?
Getting the z probe height and moving to every probe location and on every probe location getting down to the mesured z height and the final height use a paper to mesh edit it down to the desired squish at that and sourrounding points?
I feel that makes you print G26 a lot less? Rather then guessing and G26 guess again and G26?
G29 P4 R, go through each and every point move head down do measured probe height, get the paper and step down z to squish, save probe point go up to probe z height move to the next lower down to probe height and so on. Saving 10 - 20 G26 prints :)
Just a thougt thoug.

Thanks again for all the help

Roxy-3D, 2017-11-26:
> As a feature request, instead of guessing at each and every point that needs to be leveled.  What do you think about doing like the G28 process ?

I don't know what you mean.  And remember...  We are using several commands.  The first auto-probe with G29 P1.   And then a G26.   And then a G29 P4.    I need some help to understand what you mean.

> Getting the z probe height and moving to every probe location and on every probe location getting down to the measured z height and the final height use a paper to mesh edit it down to the desired squish at that and surrounding points?

Once you have the G26 pattern some what printing on the bed...   (Like you do in yesterday's post)   You can see imperfections in the G26 pattern.    It should take very little effort and time to edit the entire mesh if you choose to do that.   For example...   If you move the nozzle to the lower right of that picture where the bed is marked as an upside down (300,0).   Give it a G29 P4 R.   The nozzle will move to the first location.    

You should be able to turn the Encoder Wheel and lower the first mesh point by .05 mm with just a quarter turn.   Then you click the Encoder Wheel.    The nozzle will move to the next location.  Suppose you decide it needs a little less correction.  You turn the Encoder wheel in the same direction but less than the first point.  Then you click the Encoder Wheel.   The nozzle will jump to the 3rd location.   

You should be able to edit the entire mesh on a point by point basis in under 5 minutes.   There is no paper or micrometer measurements involved.  You are not moving the nozzle close to the bed.  You are simply looking at the shape of the filament on the bed from the previous G26 at that location.   If the filament is not squished, it means the nozzle was too far from the bed.  In that case you can lower that mesh point by .1 mm.   If the filament is too squished or wasn't even able to come out of the nozzle because the nozzle was pressing against the bed...  In that case raise the mesh point by .1 mm.

If you edit every mesh point instead of just editing the 'bad areas' of the mesh....  You should be able to do the entire bed in under 5 minutes.   At each mesh point you decide the status.  If the mesh point is 
- Correct --- Leave it alone.  Just click the encoder wheel
- a little too squished --- raise the mesh point by .05 mm and click the encoder wheel
- too squished or no extrusion possible --- raise the mesh point by .1 mm and click the encoder wheel
- some adhesion but no squish --- lower the mesh point by .05 mm and click the encoder wheel
- no adhesion --- lower the mesh point by .1 mm and click the encoder wheel

If you do two or three G26 / G29 P4 R iterations like that...   The mesh will converge very quickly to perfection.

> I feel that makes you print G26 a lot less? Rather then guessing and G26 guess again and G26?  G29 P4 R, go through each and every point move head down do measured probe height, get the paper and step down z to squish, save probe point go up to probe z height move to the next lower down to probe height and so on. Saving 10 - 20 G26 prints :)  Just a thougt though.

To get a perfect mesh usually requires 3 iterations of G26 / G29 P4 R's.   And the last G26 will be 'correct'.    There should be no micrometer or paper involved.   Just the decision of whether the bad mesh points are too high or too low, and whether it is by a lot or a little bit.

Lord-Quake, 2017-11-29:
May I ask why this has been closed?

Lostish, 2017-11-29:
Still struggeling but im getting there.
How does G28 impact the mesh ? Is there a certain order things has to be done or is excessive use of G28 not good ?
What im asking is i have this in my start gcode, and wondering if i need G28 at the start of every print, or if G28 needs to go after loading of the mesh.

```
G28
G29 L0
G29 A
Prime nozzle commands
wipe nozzle commands
```
One final before this can be closed.
Im very flash crazy as of new features and tweaking by fw rather than LCD to keep the LCD "vanilla"
And also on the bugfix branch.
How does EEPROM saves work?
What i am asking is those preserved between flashes or does the G29 mesh invalidate between every single flash? And how can i save the mesh when i finally get that one perfect?

And i think after some reading with the feature . Is the same as Manual leveling  G29 P2 B T.
But the use of paper/card at every G29 P4 too to faster fine tune every spot rather then guessing.

I need some more training and learning to get the perfect first layer but this is a huge step for me fowrard, thanks.

Roxy-3D, 2017-11-29:
The activation state is preserved across G28's for UBL.   Activation can also be preserved across resets or power ups with UBL if that is desired.   To do that, load or construct a valid mesh.   Do a G29 A to activate the mesh.  Save the mesh to a slot.  And then do a M500.      When the system powers up, it will automatically load that mesh and restore the UBL state to what it was when the M500 was given.    Not everybody wants their printer to behave that way.   But I want my printer to wake up and have the bed leveling active and working.

Your startup gcode is fine.    But because my printer wakes up with UBL active and a valid mesh loaded, I don't even have a G29 in my start up gcode.   

Any time the configuration of the EEPROM changes, you are going to have to reconfigure UBL to know what the active mesh is, and whether or not it should wake up with UBL active.   But the mesh data is saved out of the way (at the very end of the EEPROM) and its format doesn't change.   You should be able to load a previous mesh even if you have re-initialized the EEPROM for a new revision of the firmware.    (By doing a M502, G29 L1,  G29 A, G29 S 1, M500)

But just to be safe, you should also do a G29 S-1   which will print out your mesh in a gcode form.  If you save this to your computer, you can print this file and it will automatically re-load the mesh.  And at that point you can save the mesh to a new slot number.    This is useful for example if you switch Atmega-2560 boards.   You can easily move the mesh to a new board.

> But the use of paper/card at every G29 P4 too to faster fine tune every spot rather then guessing.

Yeah...   But here is the thing...   You can see if the mesh lines have good adhesion or not.   You can do that by just looking.   That is how G29 P4 R was setup.   I haven't found a need to measure anything to get a perfect mesh.   It is more a question of just doing 2 or 3 iterations of G26 and G29 P4 R and the mesh will converge very quickly.

Lostish, 2017-12-02:
Today i got some time to look a bit more on this.

I found G29 P6 C-0.3 Did the trick.
Resulting in like 5 dots needed to be raized 0.5 or 1.
So the whole grid for my printer was off by 0.3 in the whole probing...
While a repetative probe of 10, gives me a fault tolerance of 0.002
Can that G29 P6 C-0.3 correlate to something ?
My Z offset is -0.4 ( Probe stops the nozzle 0.4 above the bed in homeing )

 Just want to say it a great feature and nice codeing.
Keep up the good work

Roxy-3D, 2017-12-02:
> So the whole grid for my printer was off by 0.3 in the whole probing...

It kind of sounds like your Z_PROBE_OFFSET_FROM_EXTRUDER is off .3 mm.   It would be good to home the machine, and slowly move the nozzle down until it just touches the bed.   When you do that, does the printer think the Z-Axis is at 0.000 mm. ?

Lostish, 2017-12-02:
I have a mesurment of 0.2mm
And when homeing G28 i pull Z down to 0 with that 0.2mm mesurment between the bed and nozzle for first layer. So Z 0 is with the 0.2 mesurment between the bed and nozzle for first layer.

Do you men that G29 wants the nozzle homed to bed as 0?
That has been a misunderstanding in that case. And may axplain few things.
As without bed leveling the nozzle is 0 with 0.2mm gap? I i migt have misunderstood that aswell.

Roxy-3D, 2017-12-02:
> Do you mean that G29 wants the nozzle homed to bed as 0?

Yes.  Pretty much the printer should be setup such that the nozzle is just barely touching the bed at the home location when you carefully lower the nozzle so that Z=0.000 mm.    All of the bed leveling systems are built with that assumption.

In a perfect world, that would be true at other locations too.  That is the whole reason different bed leveling systems exist.   It is because that is not true.     But at the home location...   That should be true.

Lostish, 2017-12-02:
Awsome.
Thanks for all the help Roxy!
I was on the edge quitting but now i have the hope's up again.
This can be something great!

After all of this i flashed todays bug fix, my EEPROM seems to missmatch.
Do you know how i can reset eeprom as it's a missmatch now?

```
There was an unhandled error while talking to the printer.
Due to that OctoPrint disconnected.
Error: EEPROM CRC mismatch - (stored) 34597 != 26681 (calculated)!
```

Lostish, 2017-12-05:
The CRC errors wre a bug of octoprint.

Thanks Roxy. Closing this.

The solution were nozzle not zeroed to bed.

closed 2017-12-05 by Lostish