MarlinFirmware/Marlin #8515: [FR] Display other informations when printing from SD/LCD

ianobre, 2017-11-21:
Hi, this request is an continuation of old thread closed because a new firmware was release.

My principal request is:
After printer start using SD card the information in LCD can be changed for one of this options:
File name, ETE (like when using repetier) i knew that this information was send from repetier but maybe a fixed option like TOTAL TIME ESTIMATED: XX:XX:XX, or Total cost, Layer X of Y layers. Some information more useful that a simple message like "heating..."

I questioned for many other friends and all wil be happy if one of this messages can be used.

I upload an image to show ecxacly what i talking about:
https://i.imgur.com/la2YOFE.png

The original issue are located in this link:
https://github.com/MarlinFirmware/Marlin/issues/2936

thinkyhead, 2017-11-23:
Things like this are on the list. No specific display layout satisfies everyone, so whatever options get added for the display layout need to be integrated in a friendly way. This becomes more difficult as the number of potential display customizations grows. Thus we are considering a system where users may specify what to display and where.

As for **estimated time** for the SD card, Marlin can't really do that. Marlin would need to pre-scan the entire file to add up all the moves before a print, guessing the amount of time it would take including acceleration and deceleration. The only "guess" Marlin can make about estimated time is based on how far through the file it has gotten so far, and how much it has left to go. Unfortunately, the length of the file up-to-now is a useless indicator of how much longer the print will take.

Also, Marlin doesn't have a clue about your layer heights (which are allowed to vary throughout a print). Marlin simply moves the Z axis when told.

Host software has more horsepower, so it can bypass these issues, pre-scan G-code, and send whatever it wants to display on the status line using `M117`.

boelle, 2019-02-19:
i think this is impossible to do, but i could be wrong :-)

einencool, 2019-02-24:
Hello,
I'm relativly new to Marlin and 3d Printing, but in my GCode File created with Slic3r there is at the end the printing time shown as:
; estimated printing time (normal mode) = 3h 6m 28s
; estimated printing time (silent mode) = 3h 13m 2s

I changed the values in Slic3r to the same values in the Hardware Config, and the displayed time is the same as the used time.

Is there a possibility to show this time on the display?
Maybe with the M117 command? But how to get the time automatically in the GCode with the M117 command?
I'll try if it works, but I don't know what I'm doing *g*

boelle, 2019-02-24:
@einencool i think what slic3r PE (if that is the one you use) does is it inserts in gcode how far in % the print is, and at least prusa printers use that to update the progress bar on the printer

@thinkyhead have marlin added support for M73 ?

einencool, 2019-02-24:
@boelle 
Yes your right, I'm using Slic3r PE and from the GCode I can see it would not only show the %%, it will also show the remainig Time Rxxx in the Code.
When it would be possible to add this information, I'll think everyone would be happy with this. They will only have to set the right machine values, so that the Program to Slice will calculate the right time...

Since I set the same values for the Hardware Config and the Slic3r PE, the time is still the same. Only the heating time, when not pre-heated will slide the value a few minutes, but on printig jobs from 15 minutes to 4,5h I tested it and it works great...

boelle, 2019-02-24:
what G code is used for displaying time? we know M73 is progress in %

could you paste in a short example of then time is updated and when progress in % is updated

einencool, 2019-02-24:
My printer is busy for the next hour, so I can't check it, but this was found in the internet

**M73 Pxxx Qxxx Rxxx Sxxx**
where P is the percentage of the total time already printed in normal mode, R is the time remaining for normal mode in minutes, Q is the percentage of the total time printed in the stealth mode, and S is the time remaining to the end of print in the stealth mode.

And in my Code from Slic3r it shows The M73 Pxxx and Rxxx because I disabled the "Silent Mode"

boelle, 2019-02-24:
this makes me think, if a user has tmc2130 on this printer, or even tmc2208 could we have a silent mode also?

i guess we can peak at prusa's code and do something similar, 

if not much mistaken all they do is have 2 set's of movment parameters, like acc and feedrates

@thinkyhead would that be a major thing to add? ie option to switch between 2 sets of parameters while printing, would require a display of course

einencool, 2019-02-24:
I've got the TMC2208 on my Ender 3 but in standalone Mode.
In Slic3r there are Options to enter different values:
[image]

boelle, 2019-02-24:
yep, but it could be cool if marlin could support "silent" mode

so far i found where the different values are in their config, its a bit different build but its possible to find it: https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h#L90

einencool, 2019-02-24:
Well, since I havn't activated the M73 Command in the Config, I decided to change the GCode File for testing with the M117 instead the M73 Command, and with this option I'll be very happy, because I see the time from begining and the remainig Time.

So I only have to figure out, how to set this automatically..
I think a little workaround in Notepad++ would be the easiest way, but not very automatic...
[image]

boelle, 2019-02-24:
So the firmware does have the M73 command supported, but does that also make use of the time? or just the %

i know its there but i dont think i have enabled it, will have to take  a look

boelle, 2019-02-24:
so yes

// Add an 'M73' G-code to set the current percentage
//#define LCD_SET_PROGRESS_MANUALLY

#if HAS_CHARACTER_LCD && HAS_PRINT_PROGRESS
  //#define LCD_PROGRESS_BAR              // Show a progress bar on HD44780 LCDs for SD printing
  #if ENABLED(LCD_PROGRESS_BAR)
    #define PROGRESS_BAR_BAR_TIME 2000    // (ms) Amount of time to show the bar
    #define PROGRESS_BAR_MSG_TIME 3000    // (ms) Amount of time to show the status message
    #define PROGRESS_MSG_EXPIRE   0       // (ms) Amount of time to retain the status message (0=forever)
    //#define PROGRESS_MSG_ONCE           // Show the message for MSG_TIME then clear it
    //#define LCD_PROGRESS_BAR_TEST       // Add a menu item to test the progress bar
  #endif
#endif // HAS_PRINT_PROGRESS

and the progress bar is the one under 00:03, 

should the 00:03 number the remain time and just count down?

boelle, 2019-02-24:
my printer is printing for the next few days at least so i cant test if it works on full graph displays

einencool, 2019-02-24:
Well, I didn't activate the M73 command, because I have to update my marlin config before :-)

The 00:03 was the normal time counting up.
So I will test it in a few hours, but what do I have to uncomment in the Config?
Only this one?
#define LCD_SET_PROGRESS_MANUALLY

einencool, 2019-02-24:
I was looking for another workaround, that would definitly OK for me.
I want to add the "Printing Time" with the M117 Command in my Start GCode, so I can see how long it will print and the "normal" Time counting up.

But I didn't get the right placeholder for the "print_time" in Slic3r ???

boelle, 2019-02-24:
i think this one //#define LCD_SET_PROGRESS_MANUALLY and this one //#define LCD_PROGRESS_BAR

but looking at it makes me doubt if full graphics are supported

einencool, 2019-02-24:
You seem to know how to code :-) 
So would you see the possibility to get a variable from Slic3R PE for the "estimated printing time" to add this to the Start-GCode with the M117 comand?

Hmm, my print is finished, but since I have a mesh for the ubl generated, it would be great, not to flash the firmware until it is save for the change :-)

Is there a way to save the mesh from the UBL-Leveling ?

Another question, I've installed a MKS GEN L board, what would be the best Marlin Version for this Board? 1.1.9 or the newer 2.x?

Thanks in advance

boelle, 2019-02-24:
me code? nope

i have just looked at the config 's in prusa before and know how they do silent mode etc
where they actually do it i have no idea about

about UBL @Roxy-3D can answer you but i know yes, just forgot the command to save

always 2.0

closed 2020-08-24 by ellensp