MarlinFirmware/Marlin #8516: Arduino 1.8.5, 1.8.1, 1.0.5 don't compile Marlin 1.1.6

mopication, 2017-11-21:
### Compile Error
In line 253 of endstops.cpp I get the following error:

```
exit status 1
expected primary-expression before ')' token
```

------------------------------------------------------------

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1493096/Marlin.zip)

------------------------------------------------------------

mopication, 2017-11-21:
Oh, the line is 
```cpp
#define UPDATE_ENDSTOP_BIT(AXIS, MINMAX) SET_BIT(current_endstop_bits, _ENDSTOP(AXIS, MINMAX), (READ(_ENDSTOP_PIN(AXIS, MINMAX)) != _ENDSTOP_INVERTING(AXIS, MINMAX)))
```
in 
```cpp
void Endstops::update()
```

mopication, 2017-11-21:
I found some more error lines:
```
C:\Users\..\AppData\Local\Temp\arduino_build_334962\sketch\endstops.cpp: In static member function 'static void Endstops::update()':

endstops.cpp:253: error: expected primary-expression before ')' token

   #define UPDATE_ENDSTOP_BIT(AXIS, MINMAX) SET_BIT(current_endstop_bits, _ENDSTOP(AXIS, MINMAX), (READ(_ENDSTOP_PIN(AXIS, MINMAX)) != _ENDSTOP_INVERTING(AXIS, MINMAX)));

                                                                                                                                                                       ^

C:\Users\..\AppData\Local\Temp\arduino_build_334962\sketch\macros.h:100:38: note: in definition of macro 'SET_BIT'

 #define SET_BIT(n,b,value) (n) ^= ((-value)^(n)) & (_BV(b))

                                      ^

C:\Users\..\AppData\Local\Temp\arduino_build_334962\sketch\endstops.cpp:258:7: note: in expansion of macro 'UPDATE_ENDSTOP_BIT'

       UPDATE_ENDSTOP_BIT(AXIS, MINMAX); \

       ^

C:\Users\..\AppData\Local\Temp\arduino_build_334962\sketch\endstops.cpp:418:11: note: in expansion of macro 'UPDATE_ENDSTOP'

           UPDATE_ENDSTOP(Z, MIN_PROBE);

           ^

Bibliothek LiquidCrystal in Version 1.0.7 im Ordner: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal  wird verwendet
exit status 1
expected primary-expression before ')' token
```

teemuatlut, 2017-11-21:
Remove `#define Z_MIN_PROBE_ENDSTOP_INVERTING` from your `configuration.h` on line 704.

Please remember to close this issue if this solves your problem.

psavva, 2017-11-22:
Did you copy and paste your config file from a previous version?

If so, please don't.  You need to port and reconfigure Marlin with the new config files.

I suggest using a diff tool to help identify the changes

mopication, 2017-11-22:
Thank you @teemuatlut, this worked. I added a "true", because I am using a NO switch as z-probe. Apparently I forgot it..
@psavva I didn't, but I oriented myself at a earlier version.

closed 2017-11-22 by mopication