MarlinFirmware/Marlin #8520: Supported Stepper driver ICs

mamama1, 2017-11-22:
Hi

where do I get information on what stepper driver ICs are supported/tested by the current Marlin AVR 8 Bit branch or which requirements they have to meet to work?

Thanks

psavva, 2017-11-22:
What drivers are you targeting?
Provide the list you're interested in, and we should be able to tell you if that chipset is supported.

I know TMC2130, TMC2100, DRV8825, A4988 are supported.

Also depends on your control board.  Example the TMC2130 is not yet supported on the Re-Arm (LPC) chipset yet...

Provide details on your setup and I'm sure you'll get a more useful response.

mikko70, 2017-11-22:
from code:

  // Init TMC Steppers
  #if ENABLED(HAVE_TMCDRIVER)
    tmc_init();
  #endif

  // Init TMC2130 Steppers
  #if ENABLED(HAVE_TMC2130)
    tmc2130_init();
  #endif

  // Init L6470 Steppers
  #if ENABLED(HAVE_L6470DRIVER)
    L6470_init();

teemuatlut, 2017-11-22:
I guess it's not listed anywhere but the ones that require configuration are listed in `configuration_adv.h`.
I think currently it's L6470, TMC26X, TMC2130. Then on the way are TMC2208, TMC2224, TMC5130 and TMC2660. In addition you could drive anything that uses STEP/DIR pulses (TMC2100, DRV8825, A4988, etc).

mamama1, 2017-11-22:
Thanks! I do not have any preference at all, at the moment.
i wanted to choose the "best" drivers for home budget 3D printers like Anet A8, CR-10, Tevo Tornado, and so on. But I wanted to choose better steppers than the ones which are usually delivered with those printers.

They are intended for 8 Bit AVR so too much microsteps wouldn't make any sense if I recall correctly, because the 8 Bit AVRs don't have enough power to utilize them anyway, correct?

mamama1, 2017-11-22:
asking the other way round... which features does marlin for the 8 bit AVR platform currently support, which make sense to make use of and which steppers would be good to choose therefore?

teemuatlut, 2017-11-22:
I know I _may_ be slightly biased but I'd say the TMC2130.
With them you'll get
* Current adjustment through software. So no more fiddling with the current pots
* Ultra quiet motion with stealthChop
* Stepping interpolation to 256
* and if you need more speed than what is suitable for that mode, you can use the hybrid mode
* Sensorless homing for X and Y. No need for limit switches.
* Z axis calibration similar to how Prusa Research printers do it
* Advanced debugging capabilities

Some of this stuff is not yet pushed to upstream but all of it I've already implemented.

mamama1, 2017-11-23:
i like very much :-) thanks!

psavva, 2017-11-23:
I suggest you get originals from http://www.watterott.com/en/SilentStepStick-TMC2130

mamama1, 2017-11-23:
sure, digikey has them too...

closed 2017-11-27 by thinkyhead