MarlinFirmware/Marlin #8525: Marlin 2.0 is not compiling with BLTouch enabled

FlameFlameFlame, 2017-11-22:
There are the errors I'm getting: 
.pioenvs/LPC1768/src/src/Marlin.o: In function `servo_init()':
C:\Users\flame\Documents\GitHub\Marlin/Marlin\src/Marlin.cpp:273: undefined reference to `servo_probe_init()'
.pioenvs/LPC1768/src/src/Marlin.o: In function `bltouch_init()':
C:\Users\flame\Documents\GitHub\Marlin/Marlin\src\module/probe.h:63: undefined reference to `bltouch_command(int)'
C:\Users\flame\Documents\GitHub\Marlin/Marlin\src\module/probe.h:64: undefined reference to `set_bltouch_deployed(bool)'
C:\Users\flame\Documents\GitHub\Marlin/Marlin\src\module/probe.h:65: undefined reference to `set_bltouch_deployed(bool)'

Are those functions not written yet or I am just building Marling wrong? It compiles fine without BLtouch support.
I want to try writing those function by myself if it so. Could you please tell what servo_probe_init() should do? Functionality of other fuctions is more clear.

thinkyhead, 2017-11-23:
Should be working. Does this appear in your `probe.h` file?
```cpp
#if ENABLED(BLTOUCH)
  void bltouch_command(int angle);
  bool set_bltouch_deployed(const bool deploy);
  FORCE_INLINE void bltouch_init() {
    // Make sure any BLTouch error condition is cleared
    bltouch_command(BLTOUCH_RESET);
    set_bltouch_deployed(true);
    set_bltouch_deployed(false);
  }
#endif
```

FlameFlameFlame, 2017-11-23:
Yes, those are in `probe.h` file and it is where errors occures.

teemuatlut, 2017-11-23:
Disable `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`
Enable `Z_MIN_PROBE_ENDSTOP`
Add `#define Z_MIN_PROBE_PIN    P1_28` in pins_MKS_SBASE.h

tcm0116, 2017-11-25:
@FlameFlameFlame You need to uncomment `USE_ZMIN_PLUG` in your Configuration.h. There's a check in SanityCheck.h that should have thrown an error, but it doesn't seem to be working.

tcm0116, 2017-11-25:
@thinkyhead Upon further investigation, it looks like the `USE_ZMIN_PLUG` check in SanityCheck.h was broken by the following change to line 156 of SanityCheck.h in #8317:

```diff
-#if PROBE_SELECTED
+#if HAS_BED_PROBE
```

By using `HAS_BED_PROBE` instead of `PROBE_SELECTED`, many of the checks in the subsequent section aren't allowed to run in the event of an improper configuration such as in this issue.

FlameFlameFlame, 2017-11-26:
@tcm0116 @thinkyhead Thank you, it have solved the problem with compilation. 
Now I just cannot find where "PWM" Bltouch pin is defined.

Bob-the-Kuhn, 2017-11-26:
On the LPC1768 a PWM is automatically assigned to what ever pin you choose.  

For the least jitter PWM choose SERVO 0 or 1 if using the BLTouch in 3.3V mode.  If 5V mode then best to use SERVO 3.

Bob-the-Kuhn, 2017-11-26:
I don't see any dedicated servo pins on the Sbase.  None of the free pins match up with the servo pins on the Re-ARM card.

By default the BLTouch is assigned to SERVO0.  

The good news is that you can assign SERVO0 to any convenient pin.  My suggestion is to add the following line to your pins_MKS_SBASE.h file:
`#define SERVO0_PIN         P1_23  // J8-3`

The maybe bad news is we have no experience on the Re-ARM with using a general purpose PWM with the BLTouch.  If the BLTouch does strange things when nothing is happening then we'll have to go with a low jitter PWM.  The hopeful part is P1_23 (J8-3) could be assigned to a low jitter PWM with a code change.

Hmmm... I think I'll go ahead and make that code change in the next couple of days.  Might as well keep it as bullet proof as we can.

Bob-the-Kuhn, 2017-11-26:
Since I can't sleep ...

The changes were merged via PR #8570.

closed 2017-11-29 by thinkyhead

FlameFlameFlame, 2017-11-29:
@Bob-the-Kuhn I'll test this tomorrow and write back.

FlameFlameFlame, 2017-11-30:
@Bob-the-Kuhn It works perfectly well, thank you for help.

gtrevi, 2019-01-01:
Hi,
@Bob-the-Kuhn: heads-up: using P1.23 clashes with the mod you suggested to have the REPRAP_DISCOUNT_SMART_CONTROLLER working... (https://github.com/MarlinFirmware/Marlin/issues/8498)

would the jitter from P1.28 be sustainable (by disablin Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and enabling Z_MIN_PROBE_ENDSTOP, so that it'll #define Z_MIN_PROBE_PIN P1_28 in pins_MKS_SBASE.h)?

gtrevi, 2019-01-02:
Hi,
@Bob-the-Kuhn: heads-up: using P1.23 clashes with the mod you suggested to have the REPRAP_DISCOUNT_SMART_CONTROLLER working... (#8498)

Would the pwm jitter from P4.28 be sustainable (by disabling Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN, and enabling Z_MIN_PROBE_ENDSTOP, so that it'll #define Z_MIN_PROBE_PIN P4_28 in pins_MKS_SBASE.h)?

Thanks