MarlinFirmware/Marlin #8529: EEPROM not working correctly in last builds on MKS BASE

mexusbg, 2017-11-23:
### Bug Report
- Description: EEPROM values are not read by Repetier Host or Octoprint. Changes made to the EEPROM are not saved. I've reset the EEPROM to default values. But it still doesn't work.
EEPROM works perfect on Build 1.1.1 but not on Build 1.1.6.
Steps per mm and other values are 0 when read from EEPROM. Setting them to the correct value, storing the settings, reading them back and they are still 0-s

My configurations:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1498693/Marlin.zip)

closed 2017-11-24 by mexusbg

AnHardt, 2017-11-24:
It's always nice to know what went wrong.

mexusbg, 2017-11-24:
AnHardt. I don't know. There was a wrong board selected in configuration.h - but this was not the problem.
Tried the RC-Fix version and it worked.
Then started working on 1.1.6 again and made few changes at a time and cannot pinpoint which one got the eeprom settings working properly. I don't have time to reproduce the problem.
I'll post my working configs just in case if it helps.

[Working configs.zip](https://github.com/MarlinFirmware/Marlin/files/1501297/Working.configs.zip)

mexusbg, 2017-11-24:
@AnHardt Investigated a bit more:
one of the following options brakes EEPROM/Settings saving (reported values are zeroes and cannot be changed)
#define INCH_MODE_SUPPORT
or
#define TEMPERATURE_UNITS_SUPPORT