MarlinFirmware/Marlin #8533: BL Touch causing printer to always have STOPPED condition MKS SBASE

icthus86, 2017-11-23:
I have installed the BL Touch clone on the MKS SBASE using Z min x stop (Black and white cables) and x max stop (Orange brown red (SGV) cables) 

I have used the attached configuration files 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1500215/Configuration.zip)

I get these errors when trying to compile 

Marlin\src\module\configuration_store.cpp: In static member function 'static bool MarlinSettings::save()':
Marlin\src\module\configuration_store.cpp:667:11: error: 'ubl' was not declared in this scope
if (ubl.storage_slot >= 0)
^~~
Marlin\src\module\configuration_store.cpp:668:36: error: 'store_mesh' was not declared in this scope
store_mesh(ubl.storage_slot);
^
*** [.pioenvs\LPC1768\src\src\module\configuration_store.o] Error 1

If I comment out the Eeprom settings it compiles fine. The probe tests itself twice but then the printer sits in the STOPPED condition and I can't get the probe to do anything. I'm sure I've just missing something.

Any suggestions as to where I've gone wrong?

thinkyhead, 2017-11-26:
Try the latest `bugfix-1.1.x` from today. Presumably these lines in `configuration_store.cpp` will ensure that `ubl` is declared.
```cpp
#if HAS_LEVELING
  #include "../feature/bedlevel/bedlevel.h"
#endif
```

icthus86, 2017-11-29:
I've downloaded a copy of the new build and I'm still not getting it to compile with the following error

`
Marlin\src\module\configuration_store.cpp:667:11: error: 'ubl' was not declared in this scope
if (ubl.storage_slot >= 0)
^~~
Marlin\src\module\configuration_store.cpp:668:36: error: 'store_mesh' was not declared in this scope
store_mesh(ubl.storage_slot);
^
*** [.pioenvs\LPC1768\src\src\module\configuration_store.o] Error 1
 [ERROR] Took 41.90 seconds`

Any suggestions?

Roxy-3D, 2017-11-29:
One idea might be to use the bugfix_v1.1.x branch.   The functionality is identical if you don't care about 32-bit platforms.

Bob-the-Kuhn, 2017-11-29:
You have an error in the Configuration.h file.   Replace all instances of **//#elif** with **#elif**. 

It compiles with this change.

icthus86, 2017-11-29:
Fantastic bob-the-kuhn! That is now compiled and appears to allow me to home but instead of the touch servo pin going down and up the z-axis moves up twice?

Roxy-3D, 2017-11-29:
The Z-Axis does need to move up to provide clearance to do X and Y moves.  And then it needs to go up again (depending upon your settings) to provide clearance to deploy the Z-Probe.   It maybe those moves you are seeing.

icthus86, 2017-11-29:
The Z-probe never extends other than self test when the machine is turned on. M401 or M402 does not work on the probe for some reason.

Bob-the-Kuhn, 2017-11-29:
I forgot to mention that you have two LCDs enabled.

closed 2018-02-09 by thinkyhead