MarlinFirmware/Marlin #8537: BTN_ENC broken in bugfix-1.1.x

karabas, 2017-11-24:
RRD Smart Controller + arduino 2560./ ramps1.4
I am able to change frame rate on display rotating encoder. But encoder button does not work.
1.1.6  - works!
I tried to replace pins_ramps.h to file from 1.1.6 - nothing changes.

benlye, 2017-11-24:
Related to #8522?

pmbroth, 2017-11-24:
Sorry, I did not see this error already opened.  But I will still leave mine opened.  :-)

Bob-the-Kuhn, 2017-11-24:
Try it with the latest 1.1.x

It worked for me with a failing config from another issue.   I did the download about 15 minutes ago.

thinkyhead, 2017-11-24:
I'm glad to hear the encoder works properly. I noticed that the condition in `lcd_update` for the user-click debouncer was reversed. I just now patched it in https://github.com/thinkyhead/Marlin/commit/59a699063a49c1384c184c22ddccc12aa9feb538. So, the right-now `bugfix-1.1.x` should be in best shape.

commit 59a69906 (2017-11-24): Fix #8537
closed 2017-11-24 by thinkyhead