MarlinFirmware/Marlin #8539: Marlin reboots on Anet A8 every time I press a button

ituri, 2017-11-24:
I have this display connected to my ANET A8: https://de.aliexpress.com/item/1Pcs-Anet-3D-printer-Part-controller-RAMPS-1-4-LCD-12864-control-panel-blue-screen-For/32723690510.html?spm=2114.010208.3.43.m3Km0e&ws_ab_test=searchweb0_0,searchweb201602_2_10065_10068_10501_10000074_10503_10000032_119_10000025_10000029_430_10000028_10060_10000067_10062_10056_10055_10000062_10054_10059_10099_10000022_10000013_10103_10102_10000016_10096_10000018_10000019_10000056_10000059_10052_10053_10107_10050_10106_10051_10000053_10000007_10000050_10084_10083_10000047_10080_10082_10081_10110_10111_10112_10113_10114_10115_10000041_10000044_10078_10079_10000038_10073_10000035_10121-10051_10503_10501_10112,searchweb201603_3,afswitch_1,single_sort_3_default&btsid=77767f84-cdfb-4d05-b889-743c8e024855

I then downloaded Marlin 1.1, copied the Configuration.h for the Anet A8 and uncommented this line:
`#define ANET_FULL_GRAPHICS_LCD`

The flashing went fine, but each time I try to press a button on the display, the firmware reboots. What can I do to make Marlin work properly?
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1501960/Configuration.zip)

benlye, 2017-11-24:
The black button is a reset button, that's what it's supposed to do.

To perform an action you press on the encoder wheel.

ituri, 2017-11-24:
Oops. That's embarrassing. Thanks though!

closed 2017-11-24 by ituri