MarlinFirmware/Marlin #8540: Probe does report correct height

LVD-AC, 2017-11-24:
### Bug Report
- Description: Running G33 with different probe grids G3 G4 G5 G6...
- Expected behaviour: probing the same point would give the same reported height
- Actual behaviour: when the 1st probe point is in the centre (0,0) all seems to go well, however with probe grids that start off centre (P3, P6, P9) the reported height at the radius points e.g. X-tower (and this is always the same coordinates in all grids) is way off (5-7mm)
- Steps to reproduce:
  1) Run G33 V2 P2, P4, P5, P7, P8, P10 and check calibration (should be very close to each other)
  2) Run G33 V2 P3, P6 or P9 and check calibration (is way off and unusable for printing)

at 1) the effector moves horizontally from the centre to X-tower at 2) it almost digs into the bed doing the exact same move.

Results of G33 P3: .Height:257.99  **Ex:-5.76  Ey:-5.70**  Ez:+0.00  Radius:100.60 
Results of G33 P6: .Height:259.88  Ex:-0.03  **Ey:-5.76**  Ez:+0.00  Radius:100.55
Results of G33 P9: .Height:258.89  Ex:+0.00  **Ey:-6.50  Ez:-2.23**  Radius:100.48
any other G33 Pn: .Height:261.65  Ex:+0.00  Ey:-0.00  Ez:-0.06  Radius:100.56

**Bugfix-1.1.x only**

I couldn't reproduce this error in bugfix-2.0.x or by copy pasting the bugfix-1.1.x G33 code into 1.1.6 so it is not in the G33 code of bugfix-1.1.x

LVD-AC, 2017-11-24:
see https://github.com/LVD-AC/Marlin-AC/tree/1.1.6d-AC for Marlin 1.1.6 with the bugfix-1.1.x G33 code

LVD-AC, 2017-11-24:
The error seems to be introduced by the bugfix changes to the `Probe_pt` routine since 1.1.6.

LVD-AC, 2017-11-24:
See #8505 for fix

LVD-AC, 2017-11-24:
The bug also affects G29 on delta printers; levelling data will always be off if you have DELTA_HOME_TO_SAFE_ZONE disabled

thinkyhead, 2017-11-24:
Looking at `probe_pt` now… If it works in 2.0.x then something bad fell into 1.1.x, possibly a regression from an errant merge.

---
I've posted a PR a change to make the 1.1.x version of `probe_pt` match 2.0.x, which also involves `DELTA_HOME_TO_SAFE_ZONE`, and probably fixes a feedrate issue. Definitely looks like a merge glitch. Let's hope the undo takes care of it.

tcm0116, 2017-11-25:
I was doing some unrelated testing on my cartesian machine, and I noticed that my Z offset in bugfix-1.1.x has to be set to roughly half of what it was in 1.1.6. In my case, it was around 1.5mm in 1.1.6 and 0.75mm in bugfix-1.1.x. I think it should be closer to 1.5mm, so perhaps my issue and this issue are related.

PR #8550 (2017-11-24): Fix #8540
Does this fix #8540? This brings parity between 1.1.x and 2.0.x versions of `probe_pt`.

This should also fix #8549.
commit adee1784 (2017-11-24): Fix #8540

Does it?
commit d38aaa0c (2017-11-24): Merge pull request #8550 from thinkyhead/bf1_fix_probe_pt

Fix #8540
closed 2017-11-25 by LVD-AC