MarlinFirmware/Marlin #8541: Can't compile bugfix-2.0.x with G26_MESH_VALIDATION enabled

benlye, 2017-11-24:
In bedlevel.cpp on line 44 - should this `#if G26_MESH_VALIDATION` actually be this `#if ENABLED (G26_MESH_VALIDATION)`?

And after changing that line, compiling in either Arduino IDE or PlatformIO gives this output:
```
In file included from C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\../../inc/MarlinConfigPre.h:27:0,

                 from C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\../../inc/MarlinConfig.h:26,

                 from C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\G26.cpp:27:

C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\G26.cpp: In function 'mesh_index_pair find_closest_circle_to_print(const float&, const float&)':

C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\G26.cpp:206:20: error: 'g26_x_pos' was not declared in this scope

         f += HYPOT(g26_x_pos - mx, g26_y_pos - my) / 15.0;

                    ^

C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\../../inc/../core/macros.h:212:26: note: in definition of macro 'SQRT'

 #define SQRT(x)     sqrt(x)

                          ^

C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\../../inc/../core/macros.h:112:22: note: in expansion of macro 'sq'

 #define HYPOT2(x,y) (sq(x)+sq(y))

                      ^

C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\../../inc/../core/macros.h:217:26: note: in expansion of macro 'HYPOT2'

 #define HYPOT(x,y)  SQRT(HYPOT2(x,y))

                          ^

C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\G26.cpp:206:14: note: in expansion of macro 'HYPOT'

         f += HYPOT(g26_x_pos - mx, g26_y_pos - my) / 15.0;

              ^

C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\G26.cpp:206:36: error: 'g26_y_pos' was not declared in this scope

         f += HYPOT(g26_x_pos - mx, g26_y_pos - my) / 15.0;

                                    ^

C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\../../inc/../core/macros.h:212:26: note: in definition of macro 'SQRT'

 #define SQRT(x)     sqrt(x)

                          ^

C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\../../inc/../core/macros.h:112:28: note: in expansion of macro 'sq'

 #define HYPOT2(x,y) (sq(x)+sq(y))

                            ^

C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\../../inc/../core/macros.h:217:26: note: in expansion of macro 'HYPOT2'

 #define HYPOT(x,y)  SQRT(HYPOT2(x,y))

                          ^

C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\G26.cpp:206:14: note: in expansion of macro 'HYPOT'

         f += HYPOT(g26_x_pos - mx, g26_y_pos - my) / 15.0;

              ^

C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\G26.cpp: In function 'bool look_for_lines_to_connect()':

C:\Users\ben\AppData\Local\Temp\arduino_build_262903\sketch\src\gcode\bedlevel\G26.cpp:406:82: error: 'debug_current_and_destination' was not declared in this scope

                   debug_current_and_destination(PSTR("Connecting vertical line."));

                                                                                  ^
```

benlye, 2017-11-24:
Probably related to #8535.

Roxy-3D, 2017-11-24:
I'm not able to duplicate the compile time errors....    And you attach
your configuration.h files?

benlye, 2017-11-24:
Sure, sorry - here you go.  I tried with bilinear levelling and UBL, and got the same result.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1502094/config.zip)

Roxy-3D, 2017-11-24:
I can duplicate what you are seeing...   I'll try to get this fixed ASAP.

Thank You for the heads up on the problem!!!!

benlye, 2017-11-24:
Adding declarations for g26_x_pos and g26_y_pos seems to make that error go away.  At line 144 in G26.cpp I changed this:
```
float g26_extrusion_multiplier,
      g26_retraction_multiplier,
      g26_layer_height,
      g26_prime_length;
```
To this:
```
float g26_extrusion_multiplier,
      g26_retraction_multiplier,
      g26_layer_height,
      g26_prime_length,
      g26_x_pos,
      g26_y_pos;
```

Now I get this error:
```
sketch\src\gcode\bedlevel\G26.cpp: In function 'bool look_for_lines_to_connect()':

sketch\src\gcode\bedlevel\G26.cpp:408:82: error: 'debug_current_and_destination' was not declared in this scope

                   debug_current_and_destination(PSTR("Connecting vertical line."));

                                                                                  ^
```

benlye, 2017-11-24:
In other places in G26.cpp lines starting `debug_current_and_destination` are simply commented out.  I did the same in this instance and everything compiles now.

Roxy-3D, 2017-11-24:
Yes...   I have a patch very similar to what you just described being
checked by Travis.    I'll merge in a minute or two when Travis says "OK!"

benlye, 2017-11-24:
Thanks. I had to step out, but I'll try again with patch as soon as I get home. I'm sure that will sort it :-)

Roxy-3D, 2017-11-24:
This should be resolved  now...

benlye, 2017-11-24:
All set.  Thanks, Roxy.

closed 2017-11-24 by benlye