MarlinFirmware/Marlin #8542: ADVANCED PAUSE FEATURE not behaving properly.

Submariner75, 2017-11-24:
### Bug Report
- Description: I have installed a filament runout sensor and activated the ADVANCED PAUSE FEATURE. Detecting runout and unloading works as expected.
- Expected behaviour: when I relaod filament and press the controller box button, the filament should load all the way to the extruder, following the instructions given in these two lines:

 #define FILAMENT_CHANGE_LOAD_FEEDRATE 500     // Load filament feedrate in mm/s - filament loading into the bowden tube can be fast
  #define FILAMENT_CHANGE_LOAD_LENGTH 660       // Load filament length over hotend in mm

- Actual behaviour: the two lines above are skipped and when I press the button, these two other lines are executed once:

#define ADVANCED_PAUSE_EXTRUDE_FEEDRATE 10   // Extrude filament feedrate in mm/s - must be slower than load feedrate
  #define ADVANCED_PAUSE_EXTRUDE_LENGTH 50   // Extrude filament length in mm after filament is loaded over the hotend,

So, the filament moves only very slowly 50 mm at a time towards the extruder and it takes multiple executions of "Extrude more" to completely fill the bowden tube and the nozzle.

- Steps to reproduce:
  - Do this : trigger the filament runout sensor. Watch the filament unload. Reload filament. Press the button on the controller, watch the filament moving only 50 mm very slowly.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1502044/Config.zip)

mjc619, 2017-11-24:
I know this, it is the prevent lengthy extrude option, it says to set it to  the maximum length you would push, but the best way to load the filament before the slow feed is to just disable, setting it to my load length still didn't work,

tcm0116, 2017-11-25:
Are you using the 1.1.6 release? If so, can you give the bugfix-1.1.x branch a try?

Submariner75, 2017-11-25:
I am using the Bugfix release that I downloaded mid-september 2017, which displays bugfix-1.1.x at startup. So I guess this is the one you are referring to. And this one does not work.

Submariner75, 2017-11-25:
Temporarily fixed: cause is bad value in variable extrude_length in main.cpp

Fix is: in Marlin_Main.cpp amend line 6405 which was
      float extrude_length = initial_extrude_length;
to read
      float extrude_length = ADVANCED_PAUSE_EXTRUDE_LENGTH;

It works in my case but I did not yet do any regression test.

Submariner75, 2017-11-25:
Change in previous post !

Fix is: in Marlin_Main.cpp amend line 6405 which was
float extrude_length = initial_extrude_length;
to read
float extrude_length = FILAMENT_CHANGE_LOAD_LENGTH;

I tried with a 0 in the value ADVANCED_PAUSE_EXTRUDE_LENGTH, and obtained the desired results.

Open for comments ! Thanks

tcm0116, 2017-11-25:
@Submariner75 The change you're proposing is not correct. Loading is done in two phases. The first phase is supposed to load `FILAMENT_CHANGE_LOAD_LENGTH` mm of filament, and the second phase then loads `ADVANCED_PAUSE_EXTRUDE_LENGTH` mm of filament. The LCD will then give you the option to resume the print or load an additional `ADVANCED_PAUSE_EXTRUDE_LENGTH` mm of filament.

I would suggest you update to a current recent version of the bugfix-1.1.x branch as the version you're using is older than Marlin 1.1.6 (which was released on October 10 from bugfix-1.1.x). We are constantly making changes to that branch, so it's possible that this issue has already been fixed.

Submariner75, 2017-11-25:
Will do, thanks.

mjc619, 2017-11-25:
>Are you using the 1.1.6 release? If so, can you give the bugfix-1.1.x branch a try?
was using that, I had moved from the manufacturer marlin to marlinkimbra when I made it corexy, currently on 2.0 bugfix but had the errors on both, will try 1.1 bugfix, already have to change some features and will see if it hangs.

Submariner75, 2017-11-25:
@tcm0116

Thanks for your fast review and comments, appreciated. I indeed understood that the loading should happen in two phases and this was precisely the issue I was trying to describe. The phase 1, where we are supposed to load FILAMENT_CHANGE_LOAD_LENGTH mm of filament, does not happen on my machine, which is going to straight to phase 2.

I just tried 1.1.6. and observed the same behavior (no phase 1). Unless I am mistaken, this is how I read the current code:

In Marlin_Main.cpp, line 6373, the variable **initial_extrude_length** is initialized and set to zero (**const float &initial_extrude_length = 0**)

In line 6422, if we are in auto mode (ADVANCED_PAUSE_EXTRUDE_LENGTH > 0), the variable **extrude_length** is set as equal to **initial_extrude_length**, i.e. set to zero (**float extrude_length = initial_extrude_length;**)

In lines 6430 to 6432, the code executes phase 1, the automatic extrusion of a given length of filament, set by the variable **extrude_length**, which is currently equal to zero.

Starting line 6435, the code starts the second phase, which is manual and governed by the "extrude more" routine.

The overall effect is that no extrusion takes place during phase 1.

If in line 6422, the variable **extrude_length** is set to **load_length** (instead of **initial_extrude_length**), the phase 1 extrusion does take place as expected. 

As a summary, the only change is to amend line 6422 to read:

**float extrude_length = load_length;**

The variable **initial_extrude_length** seems useless in the code. I might be wrong somewhere, but this modification seems to do the job on my machine. Thanks for your time!

tcm0116, 2017-11-25:
If you take a look at the call to `resume_print` in `gcode_M600()`, you'll see that the `load_length` parameter is set to `FILAMENT_CHANGE_LOAD_LENGTH`, unless the L parameter is passed to M600. 

Another consideration is that there are different feed rates used in the different loading phases. By initializing `extrude_length` to `load_length`, the initial phase will be done at the `ADVANCED_PAUSE_EXTRUDE_FEEDRATE`, and not at the `FILAMENT_CHANGE_LOAD_FEEDRATE`. Furthermore, the LCD screens will be out of sync with what's happening, and if the initial loading is done as it should be, you'll end up loading twice as much filament as desired. We need to determine why the filament loading in lines 6415-6417 are not occurring. 

Please pull down the bugfix-1.1.x branch and give that a try. There have been some changes to the M600 code since 1.1.6, so your problem may have already been fixed.

thinkyhead, 2017-11-26:
> In Marlin_Main.cpp, line 6373, the variable initial_extrude_length is initialized and set to zero `(const float &initial_extrude_length = 0)`

The initializer only sets the `initial_extrude_length` to 0 when no argument is passed to the function.

> In line 6422, if we are in auto mode (`ADVANCED_PAUSE_EXTRUDE_LENGTH > 0`), the variable `extrude_length` is set as equal to `initial_extrude_length`, i.e. set to zero (`float extrude_length = initial_extrude_length;`)

Again, not set to zero. It is set to the value of the passed `initial_extrude_length` parameter.

Submariner75, 2017-11-26:
No time today for further investigation, and I fully admit that I do not have a clue about all intricacies of this complex code. But fact of life, 1.1.6 does not behave the expected way on my machine, when it does with my mod. Thanks for your time. Maybe I'll come back later.

closed 2017-11-26 by Submariner75

thinkyhead, 2018-01-26:
We've been working on improving filament change, so please let us know if it's working better in the nightly (`bugfix-1.1.x`) branch. We're aiming to release 1.1.9 soon and have been doing a lot of patches to bugs along the way.

Submariner75, 2018-01-31:
Will try today.

Submariner75, 2018-01-31:
Sorry, I cannot test it right now, I run into tons of issues. The main one
is an EEPROM version mismatch. Will come back to you.

tcm0116, 2018-01-31:
The EEPROM version mismatch means that the structure of the EEPROM data has
been updated in the code since you last saved the EEPROM. Just run M500 to
save to the EEPROM.

Submariner75, 2018-01-31:
Thanks, will do and come back to you.

Submariner75, 2018-02-01:
Dear developers,

I tested the new firmware (bugfix 1.1.x, downloaded yesterday morning). The
filament change function works as expected for me in this new
configuration. One small thing : after the initial retraction, there is
some kind of 2 seconds latency time while the filament change initializes.
During this time, nothing shows on the display, driving immediate anxiety
into the user. Maybe display the message, THEN initialize.

Good job, guys. Thanks for what you are doing for the community.

Denis

Submariner75, 2018-02-01:
This is how I configured the function on my TEVO Taratula. It works.
Looking more closely at the code in configuration.adv, I have a few
comments :

#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define PAUSE_PARK_RETRACT_FEEDRATE 500      // (mm/s) Initial retract
feedrate.
  #define PAUSE_PARK_RETRACT_LENGTH 15         // (mm) Initial retract.

                                              // This short retract is done
immediately, before parking the nozzle.
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE 500  // (mm/s) Unload filament
feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_UNLOAD_LENGTH 700   // (mm) The length of
filament for a complete unload.
                                              //   For Bowden, the full
length of the tube and nozzle.
                                              //   For direct drive, the
full length of the nozzle.
                                              //   Set to 0 for manual
unloading.
  #define FILAMENT_CHANGE_LOAD_FEEDRATE 500     // (mm/s) Load filament
feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_LOAD_LENGTH 680       // (mm) Load length of
filament, from extruder gear to nozzle.
                                              //   For Bowden, the full
length of the tube and nozzle.
                                              //   For direct drive, the
full length of the nozzle.
  #define ADVANCED_PAUSE_EXTRUDE_FEEDRATE 10   // (mm/s) Extrude feedrate
(after loading). Should be slower than load feedrate.
  #define ADVANCED_PAUSE_EXTRUDE_LENGTH 10    // (mm) Length to extrude
after loading.
                                              //   Set to 0 for manual
extrusion.
                                              //   Filament can be extruded
repeatedly from the Filament Change menu
                                              //   until extrusion is
consistent, and to purge old filament.

                                              // Filament Unload does a
Retract, Delay, and Purge first:
                         For non native speakers, what does "purge" mean in
this context ?
  #define FILAMENT_UNLOAD_RETRACT_LENGTH 20   // (mm) Unload initial
retract length.
  #define FILAMENT_UNLOAD_DELAY 2000          // (ms) Delay for the
filament to cool after retract.
  This is when nothing shows on the display. 5 seconds is long. It works
fine with 2 s

                                                      on my system. There
should be an on-screen message for the peace of mind of the

                                                      user
  #define FILAMENT_UNLOAD_PURGE_LENGTH 8      // (mm) An unretract is done,
then this length is purged.                            What does this mean
(unretract = extrude?) ? What is the purpose ?

  #define PAUSE_PARK_NOZZLE_TIMEOUT 60        // (seconds) Time limit
before the nozzle is turned off for safety.                    45 s is
short when you fumble with the filament.
  #define FILAMENT_CHANGE_ALERT_BEEPS 10      // Number of alert beeps to
play when a response is needed.
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT       // Enable for XYZ steppers to
stay powered on during filament change.

  #define PARK_HEAD_ON_PAUSE                // Park the nozzle during pause
and filament change.
  #define HOME_BEFORE_FILAMENT_CHANGE       // Ensure homing has been
completed prior to parking for filament change.

Hope it helps, thanks !

Denis

tcm0116, 2018-02-01:
`FILAMENT_UNLOAD_RETRACT_LENGTH`,  `FILAMENT_UNLOAD_DELAY`, and `FILAMENT_UNLOAD_PURGE_LENGTH` work together in an attempt to give a clean filament pull when unloading without pulling a long string through the feeder. The idea is that is retracts the filament out of the hot section, gives the blob on the end some time to cool, and then rams it back into the nozzle to force out any molten material still present (the purge). It then immediately unloads the filament. 

These parameters are intended to be tweaked to provide a clean filament pull on a user's specific printer. If you have values that are working well for you, then you've got it dialed in.