MarlinFirmware/Marlin #8545: First layer

joao-righetti, 2017-11-24:
I'm having some problems with my first layer.
In some impressions i've seen, when I looked at first layer it is really visible each "line" that was printed. That is not my case. my first layer is completaly flat, like if each of these lines have merged together.
I will leave some informations about my first layer that I beleave that are relevant for this issue (I'm using slic3r by the way):

- First layer height: 0.2mm

- First layer speed: 20mm/s

- Extrusion width for first layer: 200%

- First layer temperature (ºC): 210/70

I'm pretty sure the nozzle distance from the bed is well set when the print starts, so I really don't beleave that is the case.
If there is any other information that you think is important, please let me know!
If there is any problem understanding what my problem is I can post some pictures.

Thanks for the help any way!

Roxy-3D, 2017-11-24:
What Bed Leveling system are you using (if any) ???   If it is one of the mesh bed leveling systems, my recommendation is you do a G26 and see what the mesh validation pattern looks like.

Also...  Another good thing to try is turn on the Double Click for Z-BabyStepping.    When the first layer is printing, you might try using that to move the nozzle up and down and see what happens.   If the nozzle is too close to the bed, you may not be getting any extrusion.   If moving the nozzle up a little bit lets plastic to start coming out, you will know you have a Z-Homing issue and can set that a little bit higher.

thinkyhead, 2017-11-24:
Try dialing down the **extrusion width for the first layer** from 200% down to 110% or less. The reason that slicers set extra extrusion for the first layer is to allow for imperfection in the Z endstop, minor imperfections in the bed, etc. With bed leveling turned on you should be able to use something close to 100% (normal) extrusion width on the first layer. Try some tests in that range and see whether it makes a difference.

closed 2019-07-21 by boelle