MarlinFirmware/Marlin #8546: Bugfix 1.1X,  REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER    Dial not work on controller.

pmbroth, 2017-11-24:
Hello,
I downloaded Bugfix 1.1 today, made my configuration changes as I always do, uploaded and all works well except on the controller dial does nothing. Clicking it, or tuning it - no affect.  I have ensured the controller is enable, and looked at all my changes as I always use winmerge to do so.   I rolled back to my distribution from 10/25 and all works.  So something  had to change some where.  Thank you.

benlye, 2017-11-24:
Same as #8537?  

Does it work if you disable G26_MESH_VALIDATION?

pmbroth, 2017-11-24:
Trying it now, I will let you know

pmbroth, 2017-11-24:
With G26 Mesh Validation disabled - still does not work..

pmbroth, 2017-11-24:
I have attached configs from 10/25 that work with my printer
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1502442/config.zip)

pmbroth, 2017-11-24:
[config 1124.zip](https://github.com/MarlinFirmware/Marlin/files/1502471/config.1124.zip)
config for my printer that do not work, based on changed config compared to my previous config that work 10/25, posted in last post

Bob-the-Kuhn, 2017-11-24:
Maybe the bug has been fixed.  They've been working on a couple of issues today.

I downloaded bugfix-1.1.x about 10 minutes ago, copied over the two config.1124.zip files, compiled with Arduino 1.8.4 and the encoder works on the main/info screen and on the submenus.

Bob-the-Kuhn, 2017-11-24:
Per the notes in Issue #8537, 1.1.x was changed about 20 minutes ago to fix this issue.

pmbroth, 2017-11-25:
Hello, I just downloaded, and the fix seems good, all is working!  I will test more tomorrow, but the dial works great!  Thank you

closed 2017-11-25 by pmbroth