MarlinFirmware/Marlin #8549: Z axis using X and Y feed rate when traveling

wdoler, 2017-11-24:
### Bug Report
- Description: When issuing a travel command that contains xy and z-axis the z-axis is trying to use the X and Y axis feed rate example command G0 F3000 X100 Y100 Z10 the Z-axis will try traveling at F2400 (50mm/s) instead of being limited by the defaults set by M203 or DEFAULT_MAX_FEEDRATE.
- Expected behaviour: The z-axis will limit travel speed to a max set by command M203 
- Actual behaviour: The Z-axis travels at whatever speed that the G0 command gives.
- Steps to reproduce:
  - Home all axis of the printer
  - Send M203 Z3 (limit feed rate to 3 mm/s)
  - Send G0 F3000 X100 Y100 Z10 ( send head to x 100, y 100, z 10 at a rate of 50mm/s which should be way too fast for z-axis)
  - X and Y axis should move and Z axis should bind.
  - Re-home all axis
  - Issue a G0 F180 Z10
  - axis should move at a rate of 3mm/s
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1502502/Configuration.zip)

thinkyhead, 2017-11-25:
What version of Marlin are you using?

wdoler, 2017-11-26:
Marlin Version 1.1.6

thinkyhead, 2017-11-26:
Please test `bugfix-1.1.x` to see if the issue still exists. If it does, then we will look into it.

wdoler, 2017-11-26:
I was checking out the branch when you wrote that. Using branch bugfix-1.1.x I am still having the buzzing with the z-axis when running G0 F3000 X100 Y100 Z10. Thank you again for looking into this.

thinkyhead, 2017-11-26:
What is the complete output of `M503`? The Z acceleration and jerk values can also affect Z movement starts.

wdoler, 2017-11-26:
Thanks again for your help the M503 command made me realize that I had a testing parameter in my M203 setting, once adjusted to the correct Z speed (and having the bugfix firmware) I was able to run G0 commands with no issue. This is fixed for me. Thanks again for your help!

closed 2017-11-26 by wdoler

forum999, 2019-01-28:
i am having this same issue can you give me step by step how you fixed it please wdoler? its driving me nuts

wdoler, 2019-01-28:
@forum999 This was on a version of marlin that was from 2017, and probably doesnt apply to current versions. My problem was that I was using too high of a value in 

[image]

Again this is purely from memory since i have upgraded my firmware many times since this issue and no longer have this problem.

forum999, 2019-01-28:
Thanks for the reply, what speed were you using and what did you change it to? At the moment im having to use the start and end scripts in s3d to slow the z its a nightmare and nothing seems to solve it

wdoler, 2019-01-28:
Sorry I cant be of much help, but I do not have that information anymore.

PR #8550 (2017-11-24): Fix #8540
Does this fix #8540? This brings parity between 1.1.x and 2.0.x versions of `probe_pt`.

This should also fix #8549.