MarlinFirmware/Marlin #8562: Travis CI ImportError "No module named backports.functools_lru_cache"

thinkyhead, 2017-11-25:
What could be causing this in our Travis CI testing?
```
Traceback (most recent call last):
  File "/home/travis/virtualenv/python2.7.13/bin/platformio", line 7, in <module>
    from platformio.__main__ import main
  File "/home/travis/virtualenv/python2.7.13/lib/python2.7/site-packages/platformio/__main__.py", line 23, in <module>
    from platformio import __version__, exception, maintenance
  File "/home/travis/virtualenv/python2.7.13/lib/python2.7/site-packages/platformio/maintenance.py", line 25, in <module>
    from platformio.commands.lib import lib_update as cmd_lib_update
  File "/home/travis/virtualenv/python2.7.13/lib/python2.7/site-packages/platformio/commands/lib.py", line 22, in <module>
    import arrow
  File "/home/travis/virtualenv/python2.7.13/lib/python2.7/site-packages/arrow/__init__.py", line 3, in <module>
    from .arrow import Arrow
  File "/home/travis/virtualenv/python2.7.13/lib/python2.7/site-packages/arrow/arrow.py", line 19, in <module>
    from arrow import util, locales, parser, formatter
  File "/home/travis/virtualenv/python2.7.13/lib/python2.7/site-packages/arrow/parser.py", line 12, in <module>
    from backports.functools_lru_cache import lru_cache  # pragma: no cover
ImportError: No module named backports.functools_lru_cache
```

Roxy-3D, 2017-11-25:
My guess is Travis is broken...   And this problem will go away when they fix it.    But that is just a guess.

thinkyhead, 2017-11-25:
#8563 seems to fix it.

closed 2017-11-25 by thinkyhead

ivankravets, 2017-11-26:
Fixed in https://github.com/crsmithdev/arrow/issues/495

Just restart a job.

PR #8563 (2017-11-25): Fix #8561. Fix #8562
Fix #8561. Fix #8562.

Do a manual install of `backports.functools_lru_cache` to work around a problem with pip.
commit d3ed0f53 (2017-11-25): Merge pull request #8563 from thinkyhead/bf2_functools_lru_cache

Fix #8561. Fix #8562