MarlinFirmware/Marlin #8566: Issue with recent updates to G26_Mesh_Validation_tool.cpp [bugfix-1.1.x]

autonumous, 2017-11-26:
getting following errors unless i specify -fpermissive 

```
In file included from /home/pi/oprint/lib/python2.7/site-packages/octoprint_bigboxfirmware/lib/arduino-1.6.8/hardware/arduino/avr/cores/arduino/Arduino.h:24:0,
                 from MarlinConfig.h:37,
                 from G26_Mesh_Validation_Tool.cpp:27:
G26_Mesh_Validation_Tool.cpp: In function ‘void gcode_G26()’:
G26_Mesh_Validation_Tool.cpp:58:19: error: return-statement with a value, in function returning 'void' [-fpermissive]
   #define G26_ERR true
...
```

suspect it’s due to return_type of function   gcode_G26()

`  void gcode_G26() {`

 despite multiple `return G26_ERR ` statements throughout

if the return_type is changed, the following would be problematic.
 
 line 572:  ` if (axis_unhomed_error()) return;`

thinkyhead, 2017-11-26:
Fixed in latest `bugfix-1.1.x`. Please give it a test!

closed 2017-11-26 by thinkyhead

Roxy-3D, 2017-11-26:
I have a Pull Request doing a Travis check for bugfix-v2.0.0 for this right now.   It will be cleaned up there in a few minutes.